micropython/micropython #8028: esp32: psram invalid pin on original pins if remaped

petrkr, 2021-11-24:
I make own build with PSRAM remaped to PINs 32,33 (because original setting has colision with pin 17 used for LAN CLK-IN).

But compared to version 1.14 witih SDK 3.3.2 of uPy I can not use those 16 and 17 pins now, because it claim them as "invalid pin"

diff to SDKconfig
```
+CONFIG_D0WD_PSRAM_CLK_IO=33
+CONFIG_D0WD_PSRAM_CS_IO=32
```

```
>>> machine.Pin(16)
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
ValueError: invalid pin

```

but RAM itself works fine

```
>>> gc.mem_free()
4096416
```

of course, if I will use Pin.(33) or (32) program will crash due overlap with RAM

```
>>> machine.Pin(32,machine.Pin.OUT)
Pin(32)
>>>  
>>> 
>>> ����������������
Guru Meditation Error: Core  0 panic'ed (LoadProhibited). Exception was unhandled.
```

But I can not find part of code which should mark those pins (32,33) as invalid instead of original (16,17)

Do not know if that is BUG of uPy 1.17 or ESP_IDF SDK 4.3
Both works fine in uPy 1.14 with SDK 3.3.2

UnexpectedMaker, 2021-11-27:
IO 16 & 17 are reserved as they are used by internal flash if you have internal flash in your ESP32 chip, like a PICO-D4 or PICO-V3 etc. PSRAM also leverages those so they don't waste extra IO unnecessarily.

petrkr, 2021-11-27:
Sorry, I did not mention ESP32 original, legacy, or how to say

There pins 16,17 are not connected to flash at all.. Check schematics.

Pins 16.17 are UART2 as default pins.

for me I can not use those pins for RAM as I want use pin 17 for LAN (mentioned before)

wangshujun-tj, 2021-11-30:
The most common modules of esp32 spiram use io16 and 17, and these two pins are not necessary for LAN. As for UART, the IO matrix of esp32 is almost unlimited

petrkr, 2021-11-30:
PIN  17 **is** necessary for LAN. I it used for PHY's CLOCK OUT. 2nd pin which supports that is GPIO 16 where is this signal inverted.

[image]

wangshujun-tj, 2021-11-30:
https://sautter.com/blog/ethernet-on-esp32-using-lan8720/
I did it according to this page, very smoothly

petrkr, 2021-11-30:
You can not use gpio0 Because you have half-half chance to stay in
bootloader, also you need external clock source, which is extra part (which
cost another money). Yed I know I can turn off oscillator and turn it after
boot, but for that you need another GPIO which I do not have, since I have
other peripheries.

But this issue is about bad claimn pins as used if they are not. So it is
bug.

petrkr, 2021-12-01:
Well I found it.. It is caused by this commit https://github.com/micropython/micropython/commit/a39a596b791f14e277ba526f9b6bd8ce2da36550 which wrongly suggest SPI RAM is fixed on 16 and 17 pin. Found it in issue https://github.com/micropython/micropython/issues/7819

Since ESP32 SDK allow remap those pins, this code must read real pin instead of suggest something "because most of users use it like that".

Also why it does not block pins 6 to 11, which are connected to internal Flash memory and program will crash as-well if those pins are used?

For now I know where to comment/uncomment to make it works to me, but I suggest (or I will try to think) about some more systematic solution. Because of course block those "used" pins is good idea and it will help to solve lot of novice problems with "why my program crashes" issues.