micropython/micropython #8034: I ask for help with compilation for FreeDos

robmaster2016, 2021-11-26:
I was very pleased to read (changelog) that micropython in version 1.17 works again for Freedos. Unfortunately I can't get any further right now.  I built djgpp for Linux and could also compile the old version of micropython 1.8.5. Now I want to build 1.17 and have selected freedos as target in the ports/unix directory in the Makefile and then run a make axtls && make and unfortunately get the error message that sched.h is missing.
Previously I ran export PATH=/usr/local/djgpp/bin:$PATH and export GCC_EXEC_PREFIX=/usr/local/djgpp/lib/gcc/.
If someone could explain to me how to build the new Freedos version under Linux, I would be very happy. 
I would also like to thank all Micropython developers.

jonnor, 2024-09-08:
Were you able to fix the issue?

Unfortunately, I do not think there are many that use the FreeDOS version, so it might be a bit tricky to get much help.

robmaster2016, 2024-10-07:
Unfortunately that didn't work, but that doesn't matter because a minimal
busybox environment with Micropython is now running solidly on the server.
So I have replaced FreeDos. Best regards Rob.

jonnor, 2024-10-07:
Thank you for reporting back! Then I think we will close the issue for now. Others experiencing problems on FreeDOS with the latest MicroPython are encouraged to file a new issue.

closed 2024-10-07 by jonnor