micropython/micropython #8035: ESP32-S2 2nd usb-serial port

emard, 2021-11-26:
HI

At micropython for ESP32-S2, I'd like to have
secondary USB-serial port (like /dev/ttyACM1 on linux)
which directly maps to board's hardware serial port
RX/TX pins and if possible, few RS232 signaling lines.

Can this (easily) be done?

UnexpectedMaker, 2021-11-27:
What do you mean by second USB Serial port?
S2 has 1 native USB, and 3 HW UARTs. Native USB uses UART1 by default (not the RX/TX pins).
You can map any available IO to be any hardware UART. 

You can already use native USB and connect 2 other UARTs (via any IO) to any external Serial2UART chip (or onboard if your board has a Serial2UART chip), connected to a USB connector to effectively have 3 USB connectors already without MP needing to change in any way.

Can you be more specific about how you want your setup to be?

emard, 2021-11-27:
This is what I need:

for example now ESP32-S2 firmware binary from
https://www.wemos.cc/en/latest/tutorials/s2/get_started_with_micropython_s2.html

when plugged to PC USB, ESP32-S2 enumerates
a single USB-serial port which is used for micropython prompt

I would like it to enumerate 2 USB-serial ports:
1st USB-serial should remain for micropython prompt
2nd USB-serial should be transparently routed to
ESP32-S2 hardware UART (TX/RX pins)

Somehow I "think" that could possible but I'm not handy
with recompilation and config of the firmware

UnexpectedMaker, 2021-11-27:
There can only be 1 natively enumerated USB port on the S2. If you want more enumerated ports  you need to add more Serial2UART chips. RX/TX are not USB ports to enumerate. They are just UART pins. You need physical hardware sitting between RX/TX and a USB connector to have it enumerate. Like a CP2104 or CH340 or FTDI chip.

This is a hardware thing not a MP thing. In fact, you don't need any firmware on your S2 at all for external Serial2UART chips to enumerate.

UnexpectedMaker, 2021-11-27:
And when using native USB on the S2, UART0 is already active on RX/TX - plug an external Serial2UART adapter to those pins and see output from it from the IDF. MP already has this.

emard, 2021-11-27:
circuitpython on ESP32-S2 can create multiple
USB devices, multiple USB-serial ports, virtual microphones
keyboards etc. No chips are needed, this is done from
a single USB cable by specially crafted USB descriptor
and supporting code in ESP32-S2

UnexpectedMaker, 2021-11-27:
1. That is not what you asked for above...   you were talking about different UART pins for second enumeration. That is not what CP does.

```I would like it to enumerate 2 USB-serial ports:
1st USB-serial should remain for micropython prompt
2nd USB-serial should be transparently routed to
ESP32-S2 hardware UART (TX/RX pins)```

2. CPY handles USB totally differently from MP.  MP only offers the USB as a single serial device.

emard, 2021-11-27:
OK I didn't know MP can only make a single USB serial
device. I would like to have multiple USB serial devices.

Real use is chip and PCB space saving, to not have
onboard  ESP32-S2, FT232R/CP2102 and USB hub,
I'd like everything be done only by ESP32-S2 and firmware.

UnexpectedMaker, 2021-11-27:
Sure, understood. Less requires hardware is always better :)
Any reason you cant just use CP?

emard, 2021-11-27:
I have project for small PCB that needs a sort of multithreading for example uftpd.py should be
always running plus some additional user-written
theading handler like disk emulation or whatever
he needs

project is this
https://github.com/emard/esp32ecp5

I have ported it to CP but without thread support
it's just not good enough for my use case.

UnexpectedMaker, 2021-11-27:
Right, ok, well you can add proper USB support yourself for MP. TinyUSB is there in the IDF, though I am not sure how much it differs from the mainstream TinyUSB that Adafruit support and use. There ist no reason why you could not hack MSC and more into MP, but I don't think you'll ever see this stuff included by default in MP as not all MCUs that MP supports can do native USB  (The reason CP is not available on a pre S2 ESP32) and the general mindset of who and what MP is targeting is quite different to that of CP.

CP now has asyncio (from 7.1 I believe) so though it's not threading, maybe some of your stuff could leverage async instead? Or maybe not.

emard, 2021-11-27:
When compared both MP and CP;
MP overall suits my project better. 
For CP code is more "clumsy", I
need to upload and keep fresh
multiple additional libraries
while MP has all I need included by
"factory default".

Ok, I see, 2nd USB-serial port won't get into mainstream
MP but some compile option or howto get it would be nice.
I can try to pull sources and recompile somehow
to see what can be done.

emard, 2021-11-27:
Tell me can I do something like this on current
ESP32-S2 MP, I need a simple command that
will convert ESP32-S2 to something like FT232R
chip replacement.

with this command I want to loose micropython prompt response and have USB-serial bidirectional traffic
appear at some hardware pins, for example

>>> usbserial(rx=Pin(18), tx=Pin(19), rts=Pin(10), cts=Pin(11), baudrate=115200)