micropython/micropython #8038: ESP32: socket.getaddrinfo() returns inconsistent socket type - DNS vs. mDNS ?

karfas, 2021-11-27:
Thread in the forum:
https://forum.micropython.org/viewtopic.php?f=2&t=11498

The thread opener there used "1.17 released version for ESP32 generic with SPIRAM" (I assume a downloaded image).

I tried this myself, with v1.17 compiled with ESP-IDF v4.3:
```
>>> getaddrinfo('eris', 80)
[(2, 1, 0, 'eris', ('192.168.8.60', 80))]     # <===== return SOCK_STREAM as type
>>> getaddrinfo('eris.local', 80)
[(2, 0, 0, 'eris.local', ('192.168.8.60', 80))]   # <===== returns SOCK_DGRAM as type
```
Both names resolve to the same address, so I don' see a reason why the call with ".local" should return a different socket type.
I assume this has something to do with different methods of name lookup.

The socket type returned by getaddrinfo() is used e.g. in the the current urequest source (https://github.com/micropython/micropython-lib/blob/master/python-ecosys/urequests/urequests.py), where the socket gets created as:
```
    ai = usocket.getaddrinfo(host, port, 0, usocket.SOCK_STREAM)
    ai = ai[0]

    s = usocket.socket(ai[0], ai[1], ai[2])
    try:
        s.connect(ai[-1])
```

massimosala, 2023-05-25:
On esp8266 is even worst...

```
>>> from socket import *
>>> from network import *
>>> hostname("eris.local")
>>> getaddrinfo('eris.local', 80)
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
OSError: -2

```

samskiter, 2024-01-08:
Hrmm randomly started facing the same issue - has anyone resolved?

projectgus, 2024-01-10:
@samskiter Have opened a PR for this issue just now.

PR #13398 (2024-01-17): esp32: Use all supplied arguments to socket.getaddrinfo().
- Completes a longstanding TODO in the code, to not ignore the optional family, type, proto and flags arguments to socket.getaddrinfo().

- Note that passing family=socket.AF_INET6 will now cause queries to fail (OSError -202). Previously this argument was ignored so IPV4 results were returned instead.

- Optional 'type' argument is now always copied into the result. If not set, results have type SOCK_STREAM.

- Fixes inconsistency where previously querying mDNS local suffix (.local) hostnames returned results with socket type 0 (inval [... cut]
closed 2024-01-17 by dpgeorge