micropython/micropython #8039: ESP32: getaddrinfo() does not conform to documentation

karfas, 2021-11-27:
Micropython v1.17, compiled with ESP-IDF v4.3

While trying #8038, I read the documentation of getaddrinfo() at https://docs.micropython.org/en/latest/library/socket.html?highlight=socket#socket.getaddrinfo

There I find
``` 
socket.getaddrinfo(host, port, af=0, type=0, proto=0, flags=0, /)
```
However, trying this at home gives me
```
>>> getaddrinfo('eris', 80, type=SOCK_DGRAM)
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
TypeError: function doesn't take keyword arguments
```

peterhinch, 2021-11-27:
There are a number of MicroPython functions which only accept positional args.

karfas, 2021-11-27:
@peterhinch: Maybe, but the documentation should at least mention that. 
This is either wrong documented or wrong implemented (for me, it's a Bug in both cases).

dpgeorge, 2024-01-17:
The documentation is correct.  The `/` at the end of the argument list indicates that it accepts only positional arguments.

See https://peps.python.org/pep-0570/

closed 2024-01-17 by dpgeorge