micropython/micropython #8041: ESP32: "user C modules" should be able to add components to the build

aguaviva, 2021-11-28:
Right now the list of components is hard coded in 

https://github.com/micropython/micropython/blob/97a7cc243b028833bdcb8ce0bc19b2bce7545851/ports/esp32/main/CMakeLists.txt#L96

USer C modules should be able to add new components without having to modify the above file

My current use case is a user module that adds support for the esp32-camera

mocleiri, 2022-01-15:
@aguaviva if you have control over the board directory then its possible with the current structure to include additional idf components in the build.

This is the reference for the custom board config:
https://github.com/micropython/micropython-example-boards/blob/main/boards/CUSTOM_ESP32

There are two steps:
1. Define the **EXTRA_COMPONENT_DIRS** variable to point at the directory containing the component.  This makes additional idf components available to include in the build.
2. Reference the extra components on the **COMPONENTS** variable.   According to the [documentation](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-guides/build-system.html?highlight=components#optional-project-variables) there are certain default components that build and then any **REQUIRES** components from the **idf_component_register** are also included.

Its sort of a hack but the esp32 build only defines the components to build through the **REQUIRES**.  That is where the **IDF_COMPONENTS** come in,  So the **COMPONENTS** variable is available and when you set it that triggers building the additional components.

I tried to set both via my user moduleś micropython.cmake file but it didn´t work. I was able to set the **EXTRA_COMPONENT_DIRS** in the board makefile:
https://github.com/mocleiri/tensorflow-micropython-examples/blob/80a43ee1cf9ec24a7ebc0e1b7bb30ad5f461c01a/boards/esp32/MICROLITE/CMakeLists.txt#L6

Then I set the components to build in the user module:
https://github.com/mocleiri/tensorflow-micropython-examples/blob/80a43ee1cf9ec24a7ebc0e1b7bb30ad5f461c01a/micropython-modules/microlite/micropython.cmake#L52

This process works, I can see the custom component build and the firmware links properly.  For my project the firmware when flashed is buggy but I think that´s most likely unrelated to this approach.

The downside is that it doesn´t work if you don´t control the board.

mortbauer, 2023-09-09:
Probably this should be solved with the switch to esp-idf v5, where `list(APPEND IDF_COMPONENTS` is used, however I still do not manage make this work, the result is different if I have  user_c_module which appends a module or if I directly add it to the variable in `esp32_common.cmake`. The latter case works and the first produces errors when linking, I do not know what could be possibly be going on.

LnnrtS, 2023-12-18:
This is (at least one of) the reason why it doesn't work

https://github.com/micropython/micropython/blob/3270d856fda58585d44dd05aefa7a95551fa76cc/ports/esp32/esp32_common.cmake#L14-L17

Just always including generic `usermod.cmake` is not working since it calls `add_library`. Instead it would need some port-specific machinery to directly include the user module cmake file in the  `CMAKE_BUILD_EARLY_EXPANSION`-phase. But then that flag needs to be checked there (to not call `add_library`). Plus you will probably need `PARENT_SCOPE` to write to the top level `IDF_COMPONENTS` variable.

All in all, adding a port-specific make flag for additional components required my the c modules could be a cleaner compromise.