micropython/micropython #8042: Assertion fail in mpz.c

rajeee, 2021-11-28:
I am running into assertion error at mpz.c here: 

https://github.com/micropython/micropython/blob/97a7cc243b028833bdcb8ce0bc19b2bce7545851/py/mpz.c#L322

The error message I get is:
`assertion "carryi == 0" failed: file "/home/micropython/micropython-autobuild/py/mpz.c", line 322, function: mpn_or_neg`

This occurs when I am calling the [vocalgorithm_process ](https://github.com/adafruit/Adafruit_CircuitPython_SGP40/blob/754bf92d9e8fbe9740daca8164d4eebcc7505206/adafruit_sgp40/voc_algorithm.py#L302)function.
It might be because the voc_algorithm.py was written for circuit python, but, I think assertion fail is a pretty serious issue. Likely, it's a bug.

dpgeorge, 2021-11-28:
What platform/board are you running on, and what version of MicroPython (what does it say in the banner at start up)?  Also, can you provide a complete example that shows the bug?

rajeee, 2021-11-29:
@dpgeorge  
Here is the platform info:
(sysname='esp32', nodename='esp32', release='1.17.0', version='v1.17 on 2021-09-02', machine='ESP32 module with ESP32')

Here is a complete example:

```
from voc_algorithm import VOCAlgorithm
vals = [28559,28563,28566,28551,28554,28551,28551,28546,28543,28542,28539,28519,28514,28515,28526,28531,28519,28522,28531,28526,28531,28543,28546,28535,28534,28547,28551,28538,28527,28530,28523,28519,28530,28535,28538,28539,28543,28542,28535,28530,28531,28531,28526,28531,28538,28527,28535,28542,28535,28538,28527,28527,28518,28511,28502,28519,28519,28518,28519,28518,28531,28531,28550,28523]
alg = VOCAlgorithm()
alg.vocalgorithm_init()
for val in vals:
    val = 28551
    print(alg.vocalgorithm_process(val))
```

dpgeorge, 2021-11-29:
Thanks for the code.  I can reproduce the issue.

The problem comes about because of a negative zero.  Eg the following gives the wrong result:
```
>>> (-((1<<100)-(1<<100)))|1
-65535
```
and the following triggers the assertion:
```
(-((1<<100)-(1<<100)))|0
```

rajeee, 2021-11-29:
Interesting. Thanks for tracking down the root cause.

I don't quite understand it though. What is special about 1<<100? I played around and the problem starts from 1<<30 and larger; but  still, -(x-x) gives negative zero but only if x is larger than a certain number. Why? And why does negative zero cause problem?

Could you suggest any workarounds until this gets fixed?

dpgeorge, 2021-11-29:
> What is special about 1<<100?

It's too big to fit in a 64-bit small-integer, so forces creating a big-int on all platforms (eg to test on the unix port).

> I played around and the problem starts from 1<<30 and larger

On a 32-bit arch (like esp32) a big-int will start at 1<<30.

> And why does negative zero cause problem?

Negative zero bit-int is not handled correctly in some bit-wise operations.

dpgeorge, 2021-11-29:
> Could you suggest any workarounds until this gets fixed?

That VOC algorithm looks like it uses 16.16 fixed point arithmetic which is translated from C code, which was probably needed for MCUs that did not have float capabilities.  It would be much faster to just use float in Python.  To do that, replace the first 8 methods of `VOCAlgorithm` class with (I did test this code with the example above):
```python
class VOCAlgorithm:
    def __init__(self):
        self.params = DFRobot_vocalgorithmParams()

    def _f16(self, x):
        return x

    def _fix16_from_int(self, a):
        return a

    def _fix16_cast_to_int(self, a):
        return int(a)

    def _fix16_mul(self, inarg0, inarg1):
        return inarg0 * inarg1

    def _fix16_div(self, a, b):
        return a / b
        
    def _fix16_sqrt(self, x):
        return pow(x, 0.5)
            
    def _fix16_exp(self, x):
        import math
        return math.exp(x)
```

------------

Alternatively, you can workaround the mpz bug by changing like 125 to at `or 0` at the end:
```python
            product_hi = product_hi + 1 or 0
```

dpgeorge, 2021-11-29:
See #8046 for a proposed fix.

dpgeorge, 2021-12-21:
Should be fixed by 2c139bbf4e5724ab253b5b034ce925e04267a9c4

closed 2021-12-21 by dpgeorge