micropython/micropython #8047: ESP32-S3 Board and SPIRAM board wont't build

TGiles1998, 2021-11-29:
Neither of these boards will build. ninja failed with exit code 1.

[3/230] Generating ../../genhdr/qstr.i.last
FAILED: genhdr/qstr.i.last

UnexpectedMaker, 2021-11-30:
Might want to include more info...   as they both build fine.

What IDF version are you using? Did you compile mpy-cross? Are you sure you downloaded the MP and IDF submodules? 

This will be a configuration/path issue on your machine, not an Ninja or board specific issue.

TGiles1998, 2021-11-30:
Thank you for your response. On further inspection it is an issue with submodues:

make submodules
git submodule update --init ../../lib/berkeley-db-1.xx
fatal: not a git repository (or any of the parent directories): .git
make: *** [submodules] Error 128

I have tried to update them using:

git submodule update --init
fatal: not a git repository (or any of the parent directories): .git

Where can I find these submodules?

Thanks a lot!

TGiles1998, 2021-11-30:
I have now got this to work. Ive forked the repository and it seems to be okay now. Thanks a lot for your help!

closed 2021-11-30 by TGiles1998

mikecando, 2022-01-06:
I struggled with this bad and I am using Ubuntu 20. when running the export for idf, I saw this

Using Python interpreter in /home/xxxxx/.espressif/python_env/idf5.0_py3.8_env/bin/python
Notice the idf5.0. I do not know where that came from????? but it does not work and I get the same error as above

I was able to resolve by uninstalling (deleting and re-installing with the following command
git clone -b release/v4.4 --recursive https://github.com/espressif/esp-idf.git

Note the v4.4 in the command which gave me 4.4

I did not even know there was a 5.0 so I was surprised