micropython/micropython #8048: ESP32 S2 or S3 , can not from machine import TouchPad

Wind-stormger, 2021-11-30:
I built the firmware under the following conditions:
ESP-IDF v5.0-dev-517-g68cf4ef2be
MicroPython v1.17-212-g23a150789-dirty

I tested on esp32, esp32s2 and esp32s3 chips，Touchpad module is only available on esp32.

When I enter this code in repl:
```python
from machine import TouchPad
```

```
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
ImportError: can't import name TouchPad
````

So I further confirm whether it exists:

```
>>> import machine
>>> machine.
ADC             DAC             DEEPSLEEP       DEEPSLEEP_RESET
EXT0_WAKE       EXT1_WAKE       HARD_RESET      I2C
I2S             PIN_WAKE        PWM             PWRON_RESET
Pin             RTC             SLEEP           SOFT_RESET
SPI             Signal          SoftI2C         SoftSPI
TIMER_WAKE      TOUCHPAD_WAKE   Timer           UART
ULP_WAKE        WDT             WDT_RESET       bitstream
deepsleep       disable_irq     enable_irq      freq
idle            lightsleep      mem16           mem32
mem8            reset           reset_cause     sleep
soft_reset      time_pulse_us   unique_id       wake_reason
>>> machine.
```

What is hard to understand is , TOUCHPAD_WAKE is still there, but the TouchPad is gone?!

I reconfirmed that touchpad has been supported on esp32s2 and s3

https://github.com/micropython/micropython/blob/23a150789df1029def2a26cb757e7d6703520159/ports/esp32/machine_touchpad.c#L59-L62

I'm not sure if there's anything I haven't checked.

Hope to get some inspiration or help.

dpgeorge, 2021-11-30:
It looks like the touch pad interface has been significantly changed/expanded on the S2 and S3.  So the MicroPython `machine.TouchPad` interface needs to be updated for that to work.

Wind-stormger, 2022-02-21:
ESP-IDF 4.4
MicroPython v1.18-105
Still unable to work，just marking ，not push.

lyzp, 2022-04-23:
### Modify the following files to make the touchpad available:

1. ports/esp32/modmachine.c

Add class'TouchPad' to 'machine'
`#if CONFIG_IDF_TARGET_ESP32------>#if CONFIG_IDF_TARGET_ESP32 || CONFIG_IDF_TARGET_ESP32S2 || CONFIG_IDF_TARGET_ESP32S3`

2. ports/esp32/machine_touchpad.c 

Add 'touch_pad_fsm_start()' to function 'mtp_make_new'
Refer to[touch read example](https://github.com/espressif/esp-idf/blob/079b5b1857242d6288478c62dc36a843718882e9/examples/peripherals/touch_sensor/touch_sensor_v2/touch_pad_read/main/tp_read_main.c)
The modified file is as follows
```
#include "py/runtime.h"
#include "py/mphal.h"
#include "modmachine.h"

#include "driver/gpio.h"
#if CONFIG_IDF_TARGET_ESP32
#include "driver/touch_pad.h"
#elif CONFIG_IDF_TARGET_ESP32S2 || CONFIG_IDF_TARGET_ESP32S3
#include "driver/touch_sensor.h"
#endif

typedef struct _mtp_obj_t {
    mp_obj_base_t base;
    gpio_num_t gpio_id;
    touch_pad_t touchpad_id;
} mtp_obj_t;

#if CONFIG_IDF_TARGET_ESP32
STATIC const mtp_obj_t touchpad_obj[] = {
    {{&machine_touchpad_type}, GPIO_NUM_4, TOUCH_PAD_NUM0},
    {{&machine_touchpad_type}, GPIO_NUM_0, TOUCH_PAD_NUM1},
    {{&machine_touchpad_type}, GPIO_NUM_2, TOUCH_PAD_NUM2},
    {{&machine_touchpad_type}, GPIO_NUM_15, TOUCH_PAD_NUM3},
    {{&machine_touchpad_type}, GPIO_NUM_13, TOUCH_PAD_NUM4},
    {{&machine_touchpad_type}, GPIO_NUM_12, TOUCH_PAD_NUM5},
    {{&machine_touchpad_type}, GPIO_NUM_14, TOUCH_PAD_NUM6},
    {{&machine_touchpad_type}, GPIO_NUM_27, TOUCH_PAD_NUM7},
    {{&machine_touchpad_type}, GPIO_NUM_33, TOUCH_PAD_NUM8},
    {{&machine_touchpad_type}, GPIO_NUM_32, TOUCH_PAD_NUM9},
};

STATIC mp_obj_t mtp_make_new(const mp_obj_type_t *type, size_t n_args, size_t n_kw,
    const mp_obj_t *args) {

    mp_arg_check_num(n_args, n_kw, 1, 1, true);
    gpio_num_t pin_id = machine_pin_get_id(args[0]);
    const mtp_obj_t *self = NULL;
    for (int i = 0; i < MP_ARRAY_SIZE(touchpad_obj); i++) {
        if (pin_id == touchpad_obj[i].gpio_id) {
            self = &touchpad_obj[i];
            break;
        }
    }
    if (!self) {
        mp_raise_ValueError(MP_ERROR_TEXT("invalid pin for touchpad"));
    }

    static int initialized = 0;
    if (!initialized) {
        touch_pad_init();
        touch_pad_set_fsm_mode(TOUCH_FSM_MODE_TIMER);
        initialized = 1;
    }
    esp_err_t err = touch_pad_config(self->touchpad_id, 0);
    if (err == ESP_OK) {
        return MP_OBJ_FROM_PTR(self);
    }
[... 100 lines cut ...]
    esp_err_t err = touch_pad_read_raw_data(self->touchpad_id, &value);
    if (err == ESP_OK) {
        return MP_OBJ_NEW_SMALL_INT(value);
    }
    mp_raise_ValueError(MP_ERROR_TEXT("Touch pad error"));
}
MP_DEFINE_CONST_FUN_OBJ_1(mtp_read_obj, mtp_read);

STATIC const mp_rom_map_elem_t mtp_locals_dict_table[] = {
    // instance methods
    { MP_ROM_QSTR(MP_QSTR_config), MP_ROM_PTR(&mtp_config_obj) },
    { MP_ROM_QSTR(MP_QSTR_read), MP_ROM_PTR(&mtp_read_obj) },
};

STATIC MP_DEFINE_CONST_DICT(mtp_locals_dict, mtp_locals_dict_table);

const mp_obj_type_t machine_touchpad_type = {
    { &mp_type_type },
    .name = MP_QSTR_TouchPad,
    .make_new = mtp_make_new,
    .locals_dict = (mp_obj_t)&mtp_locals_dict,
};

#endif
```

Wind-stormger, 2022-04-29:
@lyzp I tested it on an esp32s2 board.

```python
>>> import machine
>>> machine.TouchPad()
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
AttributeError: 'module' object has no attribute 'TouchPad'
```

what type of chip did you pass the test on?

Wind-stormger, 2022-05-23:
@lyzp  Sorry for delaying this suggestion for too long.

Your code works very well, now both ESP32S2 and ESP32S3 are tested.

Perhaps a pull request should be made.

closed 2022-05-23 by Wind-stormger

peterhinch, 2022-07-22:
It would be good to see this implemented. 

@lyzp Do you plan to raise a PR?

ruihua-code, 2024-03-02:
esp32c3 also has this problem, when can this bug be fixed @peterhinch

mattytrentini, 2024-03-02:
> There is also this problem on `esp32c3`, the firmware I use is the latest version `v1.22.x`, when can this bug be fixed @peterhinch

The [ESP32-C3 doesn't have touch sensor hardware](https://docs.espressif.com/projects/esp-idf/en/v4.3/esp32c3/hw-reference/chip-series-comparison.html).