micropython/micropython #8052: freezing of network_wlan.c broken.

envirocoder, 2021-11-30:
The following code doesn't work in _boot.py but does when run from the repl.
```
import network
ap = network.WLAN(network.AP_IF)
print(ap.config('mac'))
print(ap.config('dhcp_hostname'))
```
resulting in the error:

```
Traceback (most recent call last):
  File "_boot.py", line 4, in <module>
ValueError: unknown config param
```

I think this is related to string interning. Looking in corresponding generated frozen_content.c:

```
    0x16, MP_QSTR_ap & 0xff, MP_QSTR_ap >> 8, 
    0x11, MP_QSTR_print & 0xff, MP_QSTR_print >> 8, 
    0x11, MP_QSTR_ap & 0xff, MP_QSTR_ap >> 8, 
    0x14, MP_QSTR_config & 0xff, MP_QSTR_config >> 8, 
    0x10, MP_QSTR_mac & 0xff, MP_QSTR_mac >> 8, 
    0x36, 0x01, 
    0x34, 0x01, 
    0x59, 
    0x11, MP_QSTR_print & 0xff, MP_QSTR_print >> 8, 
    0x11, MP_QSTR_ap & 0xff, MP_QSTR_ap >> 8, 
    0x14, MP_QSTR_config & 0xff, MP_QSTR_config >> 8, 
    0x23, 0x00, 
    0x36, 0x01, 
    0x34, 0x01, 
    0x59, 
    0x51, 
    0x63, 
};
STATIC const mp_obj_str_t const_obj__boot__lt_module_gt__0 = {{&mp_type_str}, 18850, 13, (const byte*)"\x64\x68\x63\x70\x5f\x68\x6f\x73\x74\x6e\x61\x6d\x65"};
```
It appears  `MP_QSTR_mac` is treated differently than `\x64\x68\x63\x70\x5f\x68\x6f\x73\x74\x6e\x61\x6d\x65` (dhcp_hostname)

I think this means that the reference to `case QS(MP_QSTR_dhcp_hostname):` on https://github.com/micropython/micropython/blob/de7e3cd792d1d4cf6c4b3895d431fc8bc516b51b/ports/esp32/network_wlan.c#L527 doesn't match.

I found this when adding my own code to the network_wlan as I initially thought it was my bug (see https://forum.micropython.org/viewtopic.php?f=3&t=11519) but I now think its something wider.

dpgeorge, 2021-12-10:
Thanks for the report.  This was previously an issue on esp8266 and was fixed on that port, see https://github.com/micropython/micropython/issues/3442

dpgeorge, 2021-12-10:
See #8077 for a fix.

dpgeorge, 2021-12-14:
Fixed by 5adb1fa40e95d938a713a3ed9a3b80615edf90fa

closed 2021-12-14 by dpgeorge