micropython/micropython #8054: stm32 NUCLEO WB55 : issue on embedded user LEDs numbers

YannickMarietti, 2021-12-01:
Hello

On firmware 1.17 with board NUCLEO-WB55 (stm32), the numbers attributed to the 3 embedded LEDs are inverted versus PCB printings:

**User LED* *               **Printing on boad PCB**      **Example µPython call**
Blue LED (right)           LED1                                   pyb.LED(3).toggle()
Green LED (center)      LED2                                   pyb.LED(2).toggle() (matching for this one !)
Red LED (left)              LED3                                   pyb.LED(1).toggle()

**Is it possible to get Blue LED referenced as LED(1) and red LED referenced as LED(3) on future firmwares ?**
 
Best regards

Yannick

jimmo, 2021-12-01:
Thanks -- just checked and you're right. It's confusing that it's RGB from left to right, but the labelling is 3 2 1. See #8055 for a fix.

YannickMarietti, 2021-12-01:
Thanks Jiimmo :)

dpgeorge, 2021-12-09:
Fixed by 92f54fe8d9618dd5e6101ac5c5f83ffc53b1bc57

closed 2021-12-09 by dpgeorge