micropython/micropython #8056: stm32 - NUCLEO-WB55 - firmware 1.17  - Modules lcd160cr and lcd160cr_test ?

YannickMarietti, 2021-12-01:
Hello

Modules _lcd160cr_ and _lcd160cr_test_ are embedded inside firmware 1.17 for stm32 NUCLEO-WB55. 

But they are related to a Pyboard-custom LCD shield, which is not compatible with Arduino or Morpho connectors available on NUCLEO-WB55.

**Maybe it would make sense to remove them from default firmware built ?**

NB: I'm speaking about NUCLEO-WB55 but the point is maybe relevant for all other stm32 boards (?).

Best regards

Yannick

jimmo, 2021-12-01:
It's approximate 9kiB... I think you're probably right that these should probably only be included by default on the GR (i.e. pybv1.x and pybd) boards. I've updated the issue title.

The other frozen modules (onewire and dht) are probably worth keeping. (But should eventually be controlled perhaps by `MICROPY_CONFIG_ROM_LEVEL`)

peterhinch, 2021-12-01:
A couple of points.

Firstly the LCD160CR has been out of stock for about a year - discontinued? This is a shame as it's an excellent board.

Secondly it's easy to use with platforms other than Pyboards: I had no problems running with ESP32. I had planned on documenting how to do this, but there seems little point as it seems to be obsolete.

YannickMarietti, 2021-12-01:
Hello

Thanks for yous answers !

I confirm, **specific drivers like OneWire and DHT are mandatory for my teaching usage !**

Going that way, full embedding of Neopixel drivers on STM32 boards, like on Pyboard,  would be cool also.
But it works as this, adding manually *neopixel.py*, so no trouble. 

No trouble too about LCD160CR modules, I don't use them, but they are not causing any trouble being there anyway.

Best regards

Yannick

dpgeorge, 2021-12-10:
I agree it'd make sense to remove the LCD160 drivers from all stm32 boards and just leave them for PYB and PYBD's.

> Firstly the LCD160CR has been out of stock for about a year - discontinued?

There are plans to make a new version with more resolution but sourcing parts has been very hard so this was put on hold.

dpgeorge, 2022-09-06:
See #9218 for a PR to address this.

dpgeorge, 2022-09-06:
Fixed by 4e4c28bf27ff4a9e1a2ce5b6fffa641d9a332507

closed 2022-09-06 by dpgeorge