micropython/micropython #8058: mpremote: btree database won't work

chrisdecker1201, 2021-12-02:
Hello,

I really like micropython and really like mpremote :).

I started now using `btree` to store some date in combination with mpremote. But I'm having problems, due to a wrong btree-database.

## Steps to reproduce

1. create btree.py (that's just the example from the documentation)

~~~python
import btree

# First, we need to open a stream which holds a database
# This is usually a file, but can be in-memory database
# using io.BytesIO, a raw flash partition, etc.
# Oftentimes, you want to create a database file if it doesn't
# exist and open if it exists. Idiom below takes care of this.
# DO NOT open database with "a+b" access mode.
try:
    f = open("mydb", "r+b")
except OSError:
    f = open("mydb", "w+b")

# Now open a database itself
db = btree.open(f)

# The keys you add will be sorted internally in the database
db[b"3"] = b"three"
db[b"1"] = b"one"
db[b"2"] = b"two"

# Assume that any changes are cached in memory unless
# explicitly flushed (or database closed). Flush database
# at the end of each "transaction".
db.flush()

# Prints b'two'
print(db[b"2"])

# Iterate over sorted keys in the database, starting from b"2"
# until the end of the database, returning only values.
# Mind that arguments passed to values() method are *key* values.
# Prints:
#   b'two'
#   b'three'
for word in db.values(b"2"):
    print(word)

del db[b"2"]

# No longer true, prints False
print(b"2" in db)

# Prints:
#  b"1"
#  b"3"
for key in db:
    print(key)

db.close()

# Don't forget to close the underlying stream!
f.close()
~~~

2. install mpremote

~~~bash
pip3 install mpremote==0.0.6
~~~

3. execute btree.py with mounting

~~~bash
mpremote u0 mount . run btree.py
~~~

output:

~~~log
b'two'
b'two'
b'three'
False
b'1'
b'3'
~~~

4. execute btree.py a second time

~~~bash
mpremote u0 mount . run btree.py
~~~

output:

~~~log
Local directory . is mounted at /remote
Traceback (most recent call last):
  File "<stdin>", line 15, in <module>
OSError: 79
~~~

## Expected behaviour:

Same behaviour like when I'm copying the btree.py file to the board and executing it:

~~~bash
 mpremote u0 cp btree.py :
 mpremote u0 run btree.py
 mpremote u0 run btree.py
 mpremote u0 cp :mydb mydb
~~~

output:

~~~log
# first run
b'two'
b'two'
b'three'
False
b'1'
b'3'
# second run
b'two'
b'two'
b'three'
False
b'1'
b'3'
~~~

## Some notes

* I found out, when I execute the steps from the expected behaviour (copy the `mydb` from the board on my computer) and then execute `mpremote u0 mount . run btree.py` it works.

dpgeorge, 2021-12-04:
Thanks for the report, and the concise instructions to reproduce the issue.

I could reproduce it and the problem is that mpremote's mount doesn't implement seek or flush via the ioctl method.  That should be fixed by #8061.

chrisdecker1201, 2021-12-05:
@dpgeorge Thank you for the quick fix. It works now.

dpgeorge, 2021-12-10:
Fixed by 67f66795c05f729ff1ca39f1d41461807c14c824

closed 2021-12-10 by dpgeorge