micropython/micropython #8059: ImportError: can't import name Servo

iwordly, 2021-12-03:
`from pyb import Servo
`
`Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
ImportError: can't import name Servo`
using STM32F4DISC file but it can't import Servo ,PWM……
Even your have this issue?
I want get a answer and thinks you support.

dpgeorge, 2021-12-04:
`pyb.Servo` is not enabled on the STM32F4DISC board, although I guess it could be.  You can try building with:
```
make BOARD=STM32F4DISC CFLAG_EXTRA=-DMICROPY_HW_ENABLE_SERVO=1
```

iwordly, 2021-12-04:
Thank your answer, unfortunately, it has the mistakes aslo.

dpgeorge, 2021-12-10:
`pyb.Servo` has been enabled on various STM32F4/F7 boards in 0892ebe09105185a2e9bc97a7047187b4d141270

closed 2021-12-10 by dpgeorge

iwordly, 2021-12-10:
Thank you for your patience to answer, adding this line of code, `#define MICROPY_HW_ENABLE_SERVO     (1)`
there is no error, if I want to add the encoder device, can I also join it?

dpgeorge, 2021-12-14:
> if I want to add the encoder device, can I also join it?

There is no encoder.  But you can use `pyb.Timer`.

iwordly, 2021-12-15:
okay thanks you support!

robert-hh, 2021-12-15:
There are two implementations for an encoder/counter class in progress. One by @IhorNehrutsa for ESP32, and one by me for MIMXRT. There also a PR from @jonathanhogg for a slightly different implementation. We three had a discussion phase to get a common basic API for the class, taking into account on how a implementation for STM32 could look like.

iwordly, 2021-12-16:
okay thanks.
I will join it later.