micropython/micropython #8060: chain of commands to connect to port and execute script does not work in mpremote

generator87, 2021-12-04:


generator87, 2021-12-04:
Trying to run this chain and getting nothing in reply:
**mpremote connect COM2 exec "import os; os.listdir()"**
but when I add mount to this after changing setup chain then it works:  **mpremote connect COM2 mount . exec "import os; os.listdir()"**
also tried to first run  this: **mpremote connect COM2**
and received: 
_Connected to MicroPython at COM2
Use Ctrl-] to exit this shell_
next I ran the second half : **import os; os.listdir()**
and got expected responce:
['abc', 'def']
>>>

so it seems like combining connection to a port with executing a script not on mounted location does not work

dpgeorge, 2021-12-04:
You may have to explicitly print the listdir output to get it to show:
```
mpremote c2 exec "import os; print(os.listdir())"
```
(note that "c2" is a shortcut for "connect COM2")

generator87, 2021-12-05:
> You may have to explicitly print the listdir output to get it to show:
> 
> ```
> mpremote c2 exec "import os; print(os.listdir())"
> ```
> 
> (note that "c2" is a shortcut for "connect COM2")

Thank you, my issue is fixed. It would be great if mpremote website would have this in the list of their examples.

dpgeorge, 2021-12-29:
mpremote has documentation at http://docs.micropython.org/en/latest/reference/mpremote.html, and a help command was added in 74e995dfd26c7c9304392b2073ec4558173b699c

closed 2021-12-29 by dpgeorge