micropython/micropython #8065: sort is not stable

stinos, 2021-12-06:
As mentioned in #338 already the builtin sort isn't stable even though it should be:

> One thing remaining -- which is a separate issue -- is that our list.sort is not stable, whereas the Python specs say it should be. We should probably care about this since it could create subtle bugs in code ported from CPython to uPy or vice-versa. See eg http://h2database.googlecode.com/svn/trunk/h2/src/tools/org/h2/dev/sort/InPlaceStableQuicksort.java for an in-place stable quicksort; or maybe we use a different sorting algorithm altogether.

Had exactly this problem today, quite subtle indeed :) Using separate issue to track this.

dpgeorge, 2021-12-10:
I thought a lot about this at one point.  It's really hard to find an algorithm that is stable and O(N log N) and inplace.

stinos, 2021-12-10:
Is the O(N log N) a requirement?

dpgeorge, 2021-12-10:
I would like O(N log N) to be a requirement.  I know the existing quicksort is not that, so perhaps a stable version of quicksort would be a good improvement?