micropython/micropython #8069: fatal error: 'asm/types.h' file not found

leapofazzam123, 2021-12-08:
`make V=1`: https://gist.github.com/e01e33df9b4d62802f5ada736b0b0084
Makefile:
```make
include external/micropython/py/mkenv.mk
MICROPY_ROM_TEXT_COMPRESSION ?= 0
include $(TOP)/py/py.mk

INC += -I.
INC += -Isrc
INC += -Iexternal/stivale
INC += -I$(TOP)
INC += -I$(BUILD)

CC = clang
LD = ld.lld

CFLAGS = -target x86_64-none-elf $(INC) -nostdlib -ffreestanding -fno-stack-protector -std=c99 -msoft-float
LDFLAGS = -nostdlib -Tsrc/linker.ld -z max-page-size=0x1000
QSTR_GEN_FLAGS = $(INC)
QSTR_GEN_CFLAGS = $(INC)
QSTR_GEN_CXXFLAGS = $(INC)
CXXFLAGS += $(filter-out -std=c99,$(CFLAGS))
CXXFLAGS += $(CXXFLAGS_MOD)
CFLAGS += $(CFLAGS_MOD)
LDFLAGS += $(LDFLAGS_MOD)

SRC_C += $(wildcard src/*.c src/*/*.c) shared/libc/printf.c shared/readline/readline.c shared/runtime/pyexec.c shared/runtime/stdout_helpers.c shared/libc/string0.c
SRC_C += $(SRC_MOD)
SRC_CXX += $(SRC_MOD_CXX)
SRC_QSTR += $(SRC_MOD) $(SRC_MOD_CXX)

OBJ += $(PY_CORE_O) $(addprefix $(BUILD)/, $(SRC_C:.c=.o))
OBJ += $(addprefix $(BUILD)/, $(SRC_CXX:.cpp=.o))

all: $(shell mkdir -p $(BUILD)) $(BUILD)/kernel.elf

$(BUILD)/kernel.elf: $(OBJ)
	$(ECHO) "LINK $@"
	$(Q)$(LD) $(LDFLAGS) $^ $(LIBS) -o $@
	$(Q)$(SIZE) $@

run: $(BUILD)/kernel.elf
	qemu-system-x86_64 -m 2G -kernel $< -debugcon stdio

include $(TOP)/py/mkrules.mk
```

dpgeorge, 2021-12-08:
This looks like a problem with the toolchain/clang, not being able to find standard C headers.  Try building a simple test C program, and/or install newlib, and/or or use gcc,

leapofazzam123, 2021-12-10:
No

leapofazzam123, 2021-12-10:
The problem is with the CFLAGS are being passed

dpgeorge, 2021-12-10:
> The problem is with the CFLAGS are being passed

Can you elaborate?  What CFLAGS should be passed?

leapofazzam123, 2021-12-11:
The CFLAGS are for my target and it being passed to the qstr generation

leapofazzam123, 2021-12-11:
It shouldn't be

leapofazzam123, 2021-12-11:
How can I make qstr to not use the CFLAGS for my target

dpgeorge, 2021-12-14:
> How can I make qstr to not use the CFLAGS for my target

In general the target CFLAGS must be used during qstr generation so that the correct qstrs are included (because qstr generation uses the pre-processor and #if's will be relevant).

leapofazzam123, 2022-02-28:
Is there a way to use my own CFLAGS thst doesn't include `-target` so clang won't freak out

dpgeorge, 2022-03-01:
> clang will freak out if its passed

AFAIK clang is OK.  The javascript port builds fine with Emscripten, and the CI builds the unix port using clang.  There's also a CI to build on mac.

> Is there a way to use my own CFLAGS thst doesn't include `-target` so clang won't freak out

See how the javascript port Makefile works, it may give you some ideas.