micropython/micropython #8073: rp2: BLE not working after update to pico-sdk 1.3.0

dpgeorge, 2021-12-09:
From https://github.com/micropython/micropython/pull/8031#issuecomment-990381299:

> There seems to be regression after the pico-sdk 1.3.0 update, RP2 Bluetooth is no longer working, it seems to get stuck right after UART is initialized (so possibly UART driver in general is broken). I think I've tracked it down to this commit: raspberrypi/pico-sdk@2622e9b it seems to enable an extra IRQ that needs to be handled in uart_service_interrupt

iabdalkader, 2021-12-10:
I sent a PR #8074 that fixes the issue, I tested the bluetooth temperature example (and a blinky), using the old and aioble API and both seem to be working fine. Although I'm not really sure why the board gets stuck if that IRQ is not handled and then eventually just crashes.

dpgeorge, 2021-12-14:
> Although I'm not really sure why the board gets stuck if that IRQ is not handled and then eventually just crashes.

Maybe because the UART IRQ handler function gets called over and over again because the flag is not cleared.  Then the main thread is starved and USB stops working.

Anyway, fixed by f21c5655835ba3703deeb1f0fa375d6a706b03cc

closed 2021-12-14 by dpgeorge