micropython/micropython #8083: RFC: change WLAN keywords authmode->security, password->key

dpgeorge, 2021-12-14:
As proposed by @iabdalkader in https://github.com/micropython/micropython/pull/8022#issuecomment-979232486:

> I think auth, authmode and password should all be deprecated in favor of security and key, as referred to in the 802.11 IEEE standards, but also logically authentication is only a part of the security protocol.

Eg `wlan.config(password="abc")` would become `wlan.config(key="abc")`, among others.

The change would be made in a backwards compatible way, such that the existing keys would be supported as well.

Please comment if you think this is a good or bad idea.

dpgeorge, 2021-12-14:
Also could consider renaming `essid` to `ssid`, since the latter seems more widely used.

dpgeorge, 2022-04-21:
I think this is a good change to make, so let's move forward with this.

iabdalkader, 2022-05-11:
On my todo list.

iabdalkader, 2022-05-28:
I checked the WLAN keyword args across ports and in the docs, and it seems no port/module shares the same args names with another (except for esp ports):

network_cyw43: `essid`, `auth`, `password`
network_ninaw10: `essid`, `security`, `key`
esp based ports: `essid`,  `authmode`, `password`
modncc3k.c: `ssid`, `security`, `key`
cc3200 port: `ssid, and an `auth` tuple
wipy: `ssid` and an `auth` tuple.

I suggest a breaking change from `essid` -> `ssid`  since it's usually a "required" arg and usually passed as a positional arg from scripts, and for the `auth` and `password` these can be changed in a backwards compatible way.

dpgeorge, 2022-06-02:
> I suggest a breaking change from `essid` -> `ssid` since it's usually a "required" arg and usually passed as a positional arg from scripts

Yes that makes sense.

But note that `essid` is also used in `.config(essid=...)`, and that can be changed in a backwards compatible way.

iabdalkader, 2022-06-03:
> But note that `essid` is also used in `.config(essid=...)`, and that can be changed in a backwards compatible way

Yes that's easy to fix.

iabdalkader, 2022-06-04:
https://github.com/micropython/micropython/pull/8725

dpgeorge, 2022-06-17:
Done, see 6e868d47dc5dd7a90444f1173118b7d57e1e4df7

closed 2022-06-17 by dpgeorge