micropython/micropython #8084: ESP32C3 Timer period doubled

Walkline80, 2021-12-14:
Run the test code

```
from machine import Timer, RTC

def timer_cb(_):
    print(rtc.datetime()[4:7])

rtc = RTC()
rtc.init((2000, 1, 1, 0, 0, 0, 0, 8))

t = Timer(0)
t.init(
    mode=Timer.PERIODIC,
    period=1000,
    callback=timer_cb
)
```

Result on ESP32C3 (IDF 4.3.1) board

```
>>> (0, 0, 2)
>>> (0, 0, 4)
>>> (0, 0, 6)
>>> (0, 0, 8)
>>> (0, 0, 10)
```

Result on ESP32 (IDF 4.3.1) board

```
>>> (0, 0, 1)
>>> (0, 0, 2)
>>> (0, 0, 3)
>>> (0, 0, 4)
>>> (0, 0, 5)
```

dpgeorge, 2021-12-15:
Thanks for the report, I can confirm the issue (using IDF 4.4 at v4.4-dev-3513-g142bb32c50).

It looks like the `machine_timer.c` code is correctly calculating the period based on `TIMER_BASE_CLK`.  So maybe this is a bug in the IDF?  I'm not sure.

dpgeorge, 2021-12-15:
On both ESP32 and ESP32C3 I get the following values:
- TIMER_DIVIDER = 8
- TIMER_SCALE = 10000000
- TIMER_BASE_CLK = 80000000

Walkline80, 2021-12-16:
Here is the introduction of General Purpose Timer

```docs
The ESP32-C3 chip contains two hardware timer groups. Each group has one general-purpose hardware timer(s). They are all 54-bit generic timers based on 16-bit pre-scalers and 54-bit up / down counters which are capable of being auto-reloaded.
```

```doc
The ESP32 chip contains two hardware timer groups. Each group has two general-purpose hardware timer(s). They are all 64-bit generic timers based on 16-bit pre-scalers and 64-bit up / down counters which are capable of being auto-reloaded.
```

I have no idea if there is a matter between the auto-reloaded `54-bit up / down counters` and `64-bit up / down counters`

dpgeorge, 2021-12-16:
It would be good to try an IDF example that uses the timer, to see how that behaves on ESP32 vs C3.

xpader, 2022-01-24:
And Timer ID is double too.
0 is ok, 1 is error, 2 is ok, 3 is error, 4 is ok, 5 is error.

sugarpines, 2022-05-13:
Per https://docs.espressif.com/projects/esp-idf/en/v4.3/esp32c3/api-reference/peripherals/timer.html the ESP32-C3 has 2 timer groups, each with 2 timers. I understand that to be 4 timers. I can confirm that timer id's 0 and 2 work as expected. Other id's result in: OSError: (-258, 'ESP_ERR_INVALID_ARG') using MicroPython v1.18 on 2022-05-06; ESP32C3 module with ESP32C3.

dpgeorge, 2022-11-23:
Thanks @fivdi for investigating this!

commit 7fe7c55b (2022-11-23): esp32/machine_timer: Fix ESP32C3 timer period doubling.

The original ESP32 only supports timer source clock APB so it doesn't need
and doesn't have a clk_src field.

The ESP32C3 supports timer source clock APB and XTAL so it does have a
clk_src field, and this needs to be configured to get the correct period.

Fixes #8084.
closed 2022-11-23 by dpgeorge