micropython/micropython #8086: Viper: problem updating nonlocal variables

peterhinch, 2021-12-15:
I am trying to write a Viper function which retains state between calls, preferably using a closure. The following produces an unexpected result (all testing on a Pyboard 1.1 running V1.17):
```python
def foo():
    x : int = 0
    @micropython.viper
    def inner() -> int:
        nonlocal x
        q : int = int(x)
        q += 1
        x = q  # x never increments
        return int(x)
    return inner

bar = foo()
bar()  # 0
bar()  # 0
```
Oddly similar code using a global works:
```python
x = 0
@micropython.viper
def foo() -> int:
    global x
    q : int = int(x)
    q += 1
    x = q
    return int(x)

foo()  # 1
foo()  # 2
foo()  # 3
```
Attempting this with a class produces a type error
```python
class Foo:
    def __init__(self):
        self.x : int = 0

    @micropython.viper
    def bar(self) -> int:
        q : int = int(self.x)
        self.x += q  # ViperTypeError: can't do binary op between 'object' and 'int'
        return int(self.x)

foo = Foo()
```
To date my "best" solution is to use a closure with an integer array to contain state:
```python
from array import array
def foo():
    x = array('i', (0,))
    @micropython.viper
    def inner() -> int:
        i = ptr32(x)
        r = i[0]
        i[0] += 1
        return r
    return inner

bar = foo()
bar()  # 0
bar()  # 1
bar()  # 2
```
It would be good if `nonlocal` worked as expected and if there were a solution to the problem of accessing integer bound variables.

Incidentally the speedup using Viper in my actual application is **phenomenal** :)

dpgeorge, 2021-12-16:
Thanks for the report.  I can reproduce the issue.

Closures are almost working in viper.  But it's storing a raw integer value instead of an integer object when assigning to the closed-over variable.

As a workaround you can explicitly create the integer-object (a small int) via:
```python
def foo():
    x : int = 0
    @micropython.viper
    def inner() -> int:
        nonlocal x
        q : int = int(x)
        q += 1
        x = q << 1 | 1  # <- workaround!
        return int(x)
    return inner

bar = foo()
bar()  # 1
bar()  # 2
```

rkompass, 2021-12-17:
As I see it, this would make it possible to program generators with @micropython.viper.
Could someone please give a very short explanation, why/how this workaround works?

dpgeorge, 2021-12-17:
> Could someone please give a very short explanation, why/how this workaround works?

In Python everything is an object, and the system needs to know how to extract the object type.  For integer objects the low bit is set.  So, the integer value `x` is packed into an object via the formula `x << 1 | 1`.

Viper works with raw integer values, not integer objects, so to communicate with Python code it must convert between objects and raw values.  The bug here is that it's not done automatically.

peterhinch, 2021-12-17:
It's perhaps worth noting that if the bug is fixed the workround will produce unexpected results. I opted for the integer array approach to avoid this hazard. Array indexing is fast and there doesn't seem to be a performance penalty.

bixb922, 2024-04-14:
I know this is an old issue, but there is still another solution which I think is a bit more readable:
```
import builtins
def foo():
    x = 0
    @micropython.viper
    def inner() -> int:
        nonlocal x
        x = builtins.int( int(x)+1 )
        return int(x)
    return inner
```
Tested on MicroPython v1.23.0-preview.47.g16c6bc47c on 2024-01-17; Generic ESP32S3 module with Octal-SPIRAM with ESP32S3.

rkompass, 2024-04-16:
Another idea does not work but yields a new error:
The idea was: make the enclosing function `@micropython.viper`
 too, then its variables, used for the closure already have the `C` format.
```
@micropython.viper
def foo():
    x : int = 0         # <-- ViperTypeError: local 'x' used before type known
    @micropython.viper
    def inner() -> int:
        nonlocal x
        q : int = x
        q += 1
        x = q  # x never increments
        return int(x)
    return inner

bar = foo()
print(bar())  # 0
print(bar())  # 0
```

peterhinch, 2024-04-16:
@bixb922 Please can you explain _how_ this works? :)

peterhinch, 2024-04-17:
Thanks for that, very informative.
> Note that 0x3fffffff is the maximum value

This is the [MicroPython object model](https://docs.micropython.org/en/latest/develop/memorymgt.html#the-object-model) - I suggest you read this.

bixb922, 2024-04-17:
>This is the MicroPython object model - I suggest you read this.

Thanks!!

bixb922, 2024-04-17:
Here is still another twist. And yes, there is a problem somewhere, but a bit different from the OP:
```py
def nonlocal_test():
    x = 1
    y = None
    z = None
    @micropython.viper
    def internal_viper():
        nonlocal x, y, z
        # Operation of viper int with nonlocal builtins.int needs 
        # converting all terms to builtins.int, if not we get a ViperTypeError
        x = x + builtins.int(1)
        viperx:int = 111
        # The next statement does *not* convert from viper int to builtins.int
        y = viperx
        # This works:
        z = builtins.int(viperx*2)
    internal_viper()
    print(f"nonlocal test {x=}, expected 2")
    print(f"nonlocal test {y=}, expected 222")
    print(f"nonlocal test {z=}, expected 111")

nonlocal_test()
```
The output is:
```
nonlocal test x=2, expected 2
nonlocal test y=55, expected 222
nonlocal test z=222, expected 111
```
With the way objects are stored now in mind, it seems that `y` gets the viper format `int`, which is interpreted as half the value.

So nonlocal int operations only work if the nonlocal was previously created as int. Other data types seem not to be affected.

rkompass, 2024-04-17:
As this is very interesting, moving back to discussions probably is best here, as new posts in issues are mailed to all core developers and thus make a lot of noise. I appreciate the discussion and think it would be great if you/we could copy your/our contribution over to the original discussion too, for others searching viper. No offence meant, just my feeling:-)