micropython/micropython #8088: On the windows platform, a unicode error occurred in bytes.decode(b'\xec')

china-zhangdj, 2021-12-15:
I am developing a socket module for the windows platform (ported from unix), but when I verify through requests, mine got a unicode error, it seems that bytes does not support the decoding of data such as b'\xec'. How can I solve this problem .

the code:
```python
while 1:
            recv = s.recv(1)
            print("recv = {}".format(recv))
            if len(recv) == 0: break
            print(type(recv))
            self.text += recv.decode()
        s.close()
        self._parse_result()
```
the error:
```python
Traceback (most recent call last):
  File "test.py", line 2, in <module>
  File "Z:/home/protons/build/micropython_with_lvgl/lv_micropython_win_need_ffi_only/ports/windows/requests.py", line 108, in get
  File "Z:/home/protons/build/micropython_with_lvgl/lv_micropython_win_need_ffi_only/ports/windows/requests.py", line 128, in urlopen
  File "Z:/home/protons/build/micropython_with_lvgl/lv_micropython_win_need_ffi_only/ports/windows/requests.py", line 62, in __init__
UnicodeError: 
```

stinos, 2021-12-15:
Can you post the output of those `print` statements as well?

china-zhangdj, 2021-12-19:
It seems that the default encoding is UTF-8, which does not support decoding of characters such as "\xec".

dpgeorge, 2021-12-19:
> It seems that the default encoding is UTF-8

Yes, MicroPython only supports ASCII and UTF-8 encodings.  This is not restricted to only the Windows port, so the same limitation would be present on all ports.

china-zhangdj, 2021-12-20:
OK, thanks.

closed 2021-12-20 by china-zhangdj