micropython/micropython #8095: ESP8266 - conflicting types for 'ets_delay_us'

pacmac, 2021-12-17:
Is there a solution to this issue for ESP8266 GENERIC build ?

I saw a similar error in micropython forum, but no solution ?

```
In file included from ./esp_mphal.h:80:0,
                 from ../../py/mphal.h:33,
                 from ../../py/mpprint.c:33:
./etshal.h:9:6: error: conflicting types for 'ets_delay_us'
 void ets_delay_us(uint16_t us);
      ^
In file included from /usr/share/upydev/esp8266/esp-open-sdk/xtensa-lx106-elf/xtensa-lx106-elf/sysroot/usr/include/arch/cc.h:40:0,
                 from /usr/share/upydev/esp8266/esp-open-sdk/xtensa-lx106-elf/xtensa-lx106-elf/sysroot/usr/include/lwip/arch.h:43,
                 from /usr/share/upydev/esp8266/esp-open-sdk/xtensa-lx106-elf/xtensa-lx106-elf/sysroot/usr/include/lwip/debug.h:35,
                 from /usr/share/upydev/esp8266/esp-open-sdk/xtensa-lx106-elf/xtensa-lx106-elf/sysroot/usr/include/lwip/opt.h:46,
                 from /usr/share/upydev/esp8266/esp-open-sdk/xtensa-lx106-elf/xtensa-lx106-elf/sysroot/usr/include/lwip/ip_addr.h:35,
                 from /usr/share/upydev/esp8266/esp-open-sdk/xtensa-lx106-elf/xtensa-lx106-elf/sysroot/usr/include/user_interface.h:30,
                 from ./esp_mphal.h:27,
                 from ../../py/mphal.h:33,
                 from ../../py/mpprint.c:33:
/usr/share/upydev/esp8266/esp-open-sdk/xtensa-lx106-elf/xtensa-lx106-elf/sysroot/usr/include/osapi.h:33:6: note: previous declaration of 'ets_delay_us' was here
 void ets_delay_us(uint32_t us);
      ^
CC ../../py/formatfloat.c
CC ../../py/parsenumbase.c
make: *** [build-GENERIC/py/mpprint.o] Error 1
../../py/mkrules.mk:77: recipe for target 'build-GENERIC/py/mpprint.o' failed
make: *** Waiting for unfinished jobs....
```

mcauser, 2021-12-20:
In SDK 2.1.x the `ets_delay_us` method us argument was unsigned 16bit, but in SDK 2.2.x it was changed to unsigned 32bit.

Try changing the us data type from `uint16_t` to `uint32_t` in:

https://github.com/micropython/micropython/blob/master/ports/esp8266/etshal.h#L9
```
// void ets_delay_us(uint16_t us);
void ets_delay_us(uint32_t us);
```

mcauser, 2021-12-20:
There was another change to the `esp_init_data_default.bin` file, mentioned here:
https://github.com/micropython/micropython/issues/6359

And further attempts at getting SDK 3.0.x working here:
https://github.com/micropython/micropython/pull/4318

dpgeorge, 2021-12-20:
It sounds like we should just remove that declaration in `etshal.h`.

dpgeorge, 2021-12-20:
See #8101.

pacmac, 2021-12-20:
Mike, Damien thanks so much.

dpgeorge, 2021-12-21:
Fixed by 05bea70979232629e059a7453fb7965545113d9f

closed 2021-12-21 by dpgeorge