micropython/micropython #8099: ESP8266 Memory problems

chrisdecker1201, 2021-12-18:
Hello,

I'm not quite sure what is happening, but I try to summarize as far as I understand my issue.

I write a micropython application which is starting to grow, and mainly I added `urequest` to make some requests to a https server.
Furthermore, I understand that https will need some memory, but how much and how much is available? (And yes, I tried using mem_free and also gc.collect(), but maybe I'm using it wrong)

I'm using:

* micropython v1.17
* micropython-urequests 0.9.1 
* ESP8266
* mpremote

When I execute the part of my request separate it works fine, but when I'm starting to use it in the "big" program I'm getting the following output

~~~log
Local directory . is mounted at /remote

 ets Jan  8 2013,rst cause:2, boot mode:(3,0)

load 0x40100000, len 30720, room 16 
tail 0
chksum 0x76
load 0x3ffe8000, len 1012, room 8 
tail 12
chksum 0x1d
ho 0 tail 12 room 4
load 0x3ffe8400, len 1080, room 12 
tail 12
chksum 0x9a
csum 0x9a
����n�r��n|�
            l
             lll`b�
                   �|r�l�n�
                           �n�
                              l`��r�l�l�

                                        l`��r�l�l�

                                                  l`��r�l
                                                         ���

                                                            ll`rl��rl���b
                                                                         ��b|
                                                                             lb��b|����
                                                                                       llb��n�nn�
                                                                                                 �ln�
                                                                                                     �
                                                                                                      �
                                                                                                       l�ll�
                                                                                                            ��

                                                                                                              l`�n����b
                                                                                                                       ll쌎
                                                                                                                           b
                                                                                                                            ���b
                                                                                                                                l
                                                                                                                                 �brlrlr�n���
                                                                                                                                             �bpbl��l��

                                                                                                                                                       �l�b��b�b��n����bpbl��l��

                                                                                                                                                                                �l�b���
~~~

I've a class to make the request like this

~~~python
class Handler():
    def __init__(self, server_url, username, password):
        self.server_url = server_url
        self.route = self.server_url + "api/v1/endpoint"
        if username is None or not username or username.isspace():
            raise ValueError

        if password is None or not password or password.isspace():
            raise ValueError
        auth = "%s:%s" % (username, password)
        basic_auth = "Basic %s" % b2a_base64(auth).decode("utf-8")
        self.headers = {
            "Authorization": basic_auth,
            "Content-Type": "application/json"
        }

    def get(self):
        json = {
            "filter_datetime": "2021-12-13 00:00:00.000000"
        }

        response = request("GET", self.route, json=json, headers=self.headers)
~~~

The output appears during the call of `ussl.wrap_socket` inside the *urequest* package.

I thought, that maybe my files are too big, so I used `myp-cross` to create *.mpy-files, but that didn't help.
I also build my own micropython without dependencies like upip, neopixel, webrepl, dht and onewire to get more memory, but also without success. And I think these dependencies are bigger than my whole program, so I thought that it will work.

Has someone an idea? And can tell me perhaps what at least the strange output means?

Kind regards,
Christian

jsimmonds2, 2021-12-20:
> When I execute the part of my request separate it works fine, but when I'm starting to use it in the "big" program I'm getting the following output

@chrisdecker1201 I think you need to provide a simple **reproducer** example here :wink: 

[ b.t.w. I'm happy to go out and buy some hardware and attempt to reproduce/diagnose this issue if your example compels  ]

chrisdecker1201, 2021-12-20:
@jsimmonds2 Hey :).

I think the easiest thing to do, is to use the real project. I tried to reproduce it with a little example but wasn't successful.

~~~bash
 git clone https://gitlab.com/lookslikematrix/timeslime-mpy
cd timeslime-mpy/
git checkout feature/timeslime_server
cp timeslime_mpy/timeslime.config timeslime_mpy/timeslime.config.local
micropython setup.py
~~~

Set in *timeslime.config.local* your SSID, PSK to your WIFI and use *https://api.timeslime.com/* as server. As I described, I built my own *micropython* here https://gitlab.com/lookslikematrix/timeslime-system/.

~~~bash
esptool.py --port /dev/ttyUSB0 erase_flash
wget https://gitlab.com/lookslikematrix/timeslime-system/-/jobs/1897435582/artifacts/raw/timeslime-system_20211217_826c3dca.bin
esptool.py --port /dev/ttyUSB0 --baud 460800 write_flash --flash_size=detect 0 timeslime-system_20211217_826c3dca.bin
cd timeslime_mpy/
mpremote u0 mount . run main.py
~~~

Now you should get the output I described. I'm glad to help with more informations.

chrisdecker1201, 2021-12-21:
I investigated the issue a little further and found out, if I copy my files to the ESP8266 and execute them just by powering the ESP8266 it works. The `main.py` script is then executed automatically.

Maybe it's a mpremote issue?

chrisdecker1201, 2021-12-22:
I enabled all self.log_cmd in mpremote and get now the following output. It might be related to the `[seek 0 0]`?

~~~log
...
[close 0]
[open .//timeslime.config.local r]
[read 0 -1 -> 337]
[close 0]
#7 ets_task(4020f560, 28, 3fff9ee8, 10)
[open .//timespan.db r+b]
[read 0 24 -> 0]
[seek 0 0]

 ets Jan  8 2013,rst cause:2, boot mode:(3,6)

load 0x40100000, len 30720, room 16 
tail 0
chksum 0x76
load 0x3ffe8000, len 1012, room 8 
tail 12
chksum 0x1d
ho 0 tail 12 room 4
load 0x3ffe8400, len 1080, room 12 
tail 12
chksum 0x9a
csum 0x9a
����n�r��n|�
            l
             lll`b�
                   �|r�l�n�
                           �n�
                              l`��r�l�l�

                                        l`��r�l�l�
...
~~~

chrisdecker1201, 2021-12-22:
I changed my program to use a ByteIO instead of a File where no `seek` is needed and it crashes as before.

~~~log
...
[close 0]
[open .//timeslime.config.local r]
[read 0 -1 -> 337]
[close 0]

 ets Jan  8 2013,rst cause:2, boot mode:(3,6)

load 0x40100000, len 30720, room 16 
tail 0
chksum 0x76
load 0x3ffe8000, len 1012, room 8 
tail 12
chksum 0x1d
ho 0 tail 12 room 4
load 0x3ffe8400, len 1080, room 12 
tail 12
chksum 0x9a
csum 0x9a
����n�r��n|�
            l
             lll`b�
                   �|r�l�n�
                           �n�
                              l`��r�l�l�

                                        l`��r�l�l�
...
~~~

chrisdecker1201, 2021-12-22:
I narrowed it down to the `read_until`-method in the *pyboard.py* file. It's the first time this method is called with a `data_consumer`-callback.

The callback is therefore executed `data_consumer(data)`, but maybe I'm on the wrong track.

chrisdecker1201, 2022-01-04:
@jsimmonds2 @dpgeorge: I've now a small script to reproduce the error. I think I've really do have not enough memory, but there probably should be an `MemoryError`.

~~~python
from urequests import request

class MemoryUse:
    number = 0

array = []
for x in range(10000):
    array.append(MemoryUse())

request("GET", "https://micropython.org/")
 ~~~

dpgeorge, 2022-01-06:
Thanks for providing a full reproduction.  I didn't yet try it but I can say a few things:
1. running `mpremote mount .` consumes about 5000 bytes of RAM on the device, so this is very likely the problem, that the esp8266 just doesn't have enough RAM left to do that and also run your code
2. the fact that the esp8266 crashes (that's what the strange characters are) means there is a bug somewhere, most likely with the remote mounting code, but that shouldn't affect your application (when run just by itself)
3. to workaround the issue, don't use `mpremote mount .` for large applications when there's not enough RAM

chrisdecker1201, 2022-01-08:
@dpgeorge Thank you for your help :smile:. It explains a lot. I tried to use:

~~~bash
mpremote u0 cp -r . :
mpremote u0
~~~

and

 ~~~python
exec(open("main.py").read())
~~~

but then I get a similar issue and `mem_free` gives me not really more free memory.

~~~log
 ets Jan  8 2013,rst cause:2, boot mode:(3,6)

load 0x40100000, len 30720, room 16 
tail 0
chksum 0x76
load 0x3ffe8000, len 1012, room 8 
tail 12
chksum 0x1d
ho 0 tail 12 room 4
load 0x3ffe8400, len 1080, room 12 
tail 12
chksum 0x9a
csum 0x9a
[8c][e2][02][ec][12][82]n[ec][92]r[82][f2]n|[ec][0c]l[0c]lll`[02][1c]b[92][12][02][0c][9e]|[02]r[92][02]l[9c]n[e0][0c][82]n[e2][00][0c]l`[02][8c][e2]r[92]l[8c]l[12][8c][0c][0c][0c]l`[02][8c][e2]r[92]l[ec]l[12][8c][0c][0c][0c]l`[02][8c][e2]r[92]l[0c][8e][9e][00][8c][0c][0c]ll`[02]rl[8e][82]rl[8c][9e][e2][8c]b[0c][8c][8c]b[1c]|[0c]lb[9c][e2][9c][02]b|[90][8c][8c][8c][0c]llb[ec][f2]n[ee][80][12]nn[e2][10][02][0c][02][8c]ln[9e][0c][ec][0c][8c][0c]l[ec][8e]ll[8c][0c][8e][8c][0c][0c]l`[02][80][02]n[fc][00][8c][9e][e2][8c]b[0c]ll[ec][8c][8e][12][0c]b[0c][02][02][8c][9e][e2][8c]b[0c]l[0c][8c]b[1c]rlrlr[f2]n[9c][e2][00][8c][0c][ec][82][12][02]bp[0e]bl[8e][8c]l[9c][1c][ec][0c][0c][8c]l[ec]b[8c][92][12]b[ec]b[ec][f2]n[9c][9e][80][80][02]bp[12]bl[8e][8c]l[9c][1c][ec][0c][0c][8c]l[ec]b[8c][92][ff]------------------------------------------------------
~~~

Do `mpremote` in general use about 5000 bytes?

If someone can tell me if there is somehow a tutorial to debug issues like that, I can also look into it. I would be really glad to help more.

chrisdecker1201, 2022-02-03:
I'm using now the ESP32 board. So at least for me, I don't need a fix anymore. But of course it would be nice for the future to handle the out of memory problem probably and throw an error.

mattytrentini, 2024-07-21:
I think it best to close this issue for now as this is quite old and not a problem for the original developer. If anyone is using an ESP8266 and can reproduce the issue then please reopen or create a new ticket.

closed 2024-07-21 by mattytrentini