micropython/micropython #8102: Unable to build rp2

pacmac, 2021-12-20:
See below, I have tried with various versions of cmake, any idea of cause/ solution ?

```
[ 43%] Building C object CMakeFiles/firmware.dir/usr/share/upydev/work/micropython/lib/tinyusb/src/device/usbd.c.obj
/usr/share/upydev/work/micropython/lib/pico-sdk/src/rp2_common/pico_standard_link/new_delete.cpp:22:1: error: identifier 'noexcept' is a keyword in C++11 [-Werror=c++0x-compat]
 void operator delete[](void *p) noexcept { std::free(p); }
 ^
/usr/share/upydev/work/micropython/lib/pico-sdk/src/rp2_common/pico_standard_link/new_delete.cpp:22:33: error: expected initializer before 'noexcept'
 void operator delete[](void *p) noexcept { std::free(p); }
                                 ^
cc1plus: all warnings being treated as errors
CMakeFiles/firmware.dir/build.make:3980: recipe for target 'CMakeFiles/firmware.dir/usr/share/upydev/work/micropython/lib/pico-sdk/src/rp2_common/pico_standard_link/new_delete.cpp.obj' failed
make[3]: *** [CMakeFiles/firmware.dir/usr/share/upydev/work/micropython/lib/pico-sdk/src/rp2_common/pico_standard_link/new_delete.cpp.obj] Error 1
make[3]: *** Waiting for unfinished jobs....
make[2]: *** [CMakeFiles/firmware.dir/all] Error 2
CMakeFiles/Makefile2:1356: recipe for target 'CMakeFiles/firmware.dir/all' failed
make[1]: *** [all] Error 2
Makefile:90: recipe for target 'all' failed
Makefile:18: recipe for target 'all' failed
make: *** [all] Error 2
```

pacmac, 2021-12-21:
Hmm, tried downloading and installing older versions (now using 3.22.1) and get same error no matter what the version.

Is this a error that is actually just a warning, and if so can it be suppressed so that the build can complete ?

dpgeorge, 2021-12-21:
It builds ok for me with `arm-none-eabi-gcc` v11.2.0.  It may be that you have an old compiler that does not support this new C++ syntax.

> Is this a error that is actually just a warning,

It looks like a syntax error for you.

pacmac, 2021-12-21:
Damien;

Looks like I am using 5.4.1, Will see if I can dowload and compile from source rather than using distro version.

```
arm-none-eabi-gcc --version
arm-none-eabi-gcc (15:5.4.1+svn241155-1) 5.4.1 20160919
Copyright (C) 2015 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.

root@7dbad9ba26cb:/usr/share/upydev/work#
```

pacmac, 2021-12-21:
I can confirm that installing version 10.3 solves this issue.

Thanks so much.

closed 2021-12-21 by pacmac