micropython/micropython #8103: WPA3 support

lbhm, 2021-12-20:
I'm currently planning a small sensor project with an ESP8266 MCU and considered to use Micropython for the implementation since I already use a lot of Python for other projects. Unfortunately, my WiFi at home is running WPA3 and the Micropython docs ([[1]](https://docs.micropython.org/en/latest/library/network.WLAN.html)) indicate that WPA3 is currently not supported.

Is it planned to add WPA3 support to Micropython in the nearer future? Since WPA3 is the new and more secure standard, this would be a great feature to have!

sanderjo, 2025-03-24:
> WPA3 is supported on ESP32.

Yes. But is it supported in microPython? Because https://docs.micropython.org/en/latest/library/network.WLAN.html#:~:text=There%20are%20five%20values%20for%20security%3A says:

```
There are five values for security:

0 – open
1 – WEP
2 – WPA-PSK
3 – WPA2-PSK
4 – WPA/WPA2-PSK
```

... so no option for WPA3?

Reason I'm asking: my ESP32-S3 with micropython could not connect to my WPA3 Wifi. After hard setting "WPA2 personal" on the router/ap, I could connect.

sanderjo, 2025-03-24:
FWIW searching for "WPA3" on github does reveal some: https://github.com/search?q=repo%3Amicropython%2Fmicropython+wpa3&type=code