micropython/micropython #8104: ESP32: Setting Bluetooth TX power

ctschach, 2021-12-20:
I had trouble connecting to a Bluetooth master running on an ESP32. After about  4m I couldn't get a reliable connection anymore. 

After trying out different ESP32's and external antennas, I run across the following forum post:

https://forum.micropython.org/viewtopic.php?t=7733#p62651

He is increasing the Bluetooth power directly in the init function with the following two commands:

```
    esp_ble_tx_power_set(ESP_BLE_PWR_TYPE_DEFAULT, ESP_PWR_LVL_P9);
    esp_ble_tx_power_set(ESP_BLE_PWR_TYPE_ADV, ESP_PWR_LVL_P9);
```

And that actually solved the problem. Not sure what the default Bluetooth TX power is, but would there be a chance that you can set the TX power from the bluetooth init-function inside of MicroPython?

stephanelsmith, 2022-01-08:
Thank you for sharing your findings.  Working on BLE as well in my application.  A bit on hold as I'm on S3...

crackwitz, 2026-01-13:
Sadly, the Micropython build for C3 does not offer `esp32.esp_ble_tx_power_set`.