micropython/micropython #8106: Changing  Build Path Breaks ESP8266 build.

pacmac, 2021-12-23:
My apologies for all the recent issue, but I am building a new docker container for building ALL micropython ports with an easy to use interface, and so I need to change some default paths and use different O/S than may have been used when for example the esp8266 open sdk was last updated.

Anyway, I have made good progress and most is working, however I have just changed the BUILD PATH from the standard:

```$BUILD-build``` relative path

To an absolute path in my working folders i.e. ```\usr\share\pydev\work\boards\$PORT\$BOARD\build```.

This new env var $BUILD_PATH then gets set in the make command:

```make .....other-args BUILD=$BUILD_PATH```

And on building the esp8266 it results in this:

```
xtensa-lx106-elf-ld: address 0x400081e8 of /build/firmware.elf section `.bss' is not within region `dram0_0_seg'
xtensa-lx106-elf-ld: address 0x400081e8 of /build/firmware.elf section `.bss' is not within region `dram0_0_seg'
```

As soon as I restore my custom build path override to the default, everything works again.

Any idea on the cause and a fix ?

dpgeorge, 2021-12-23:
You can try this patch.  It builds but I didn't test it on hardware:
```diff
--- a/ports/esp8266/boards/esp8266_common.ld
+++ b/ports/esp8266/boards/esp8266_common.ld
@@ -144,7 +144,7 @@ SECTIONS
         *shared/timeutils/*.o*(.literal*, .text*)
         *drivers/bus/*.o(.literal* .text*)
 
-        build-*/main.o(.literal* .text*)
+        *main.o(.literal* .text*)
         *fatfs_port.o(.literal* .text*)
         *gccollect.o(.literal* .text*)
         *gchelper.o(.literal* .text*)
@@ -182,7 +182,7 @@ SECTIONS
         */frozen.o(.rodata.mp_frozen_content) /* frozen modules */
 
         /* for -mforce-l32 */
-        build-*/*.o(.rodata*)
+        *.o(.rodata*)
 
         _irom0_text_end = ABSOLUTE(.);
     } >irom0_0_seg :irom0_0_phdr
```

pacmac, 2021-12-23:
Thanks Damien, but I am rather embarrassed to tell you that I have never applied and patch and don't know how :-(

I tried saving it to a file and running it with:

```patch < mypatch.patch```

But that prompts me for a file name and I thought the file name is contained in the file ??

dpgeorge, 2021-12-23:
You can do it by hand, just remove the `build-*` parts on those 2 lines.  The final lines in your file should look like the green lines above.

pacmac, 2021-12-23:
Thanks, yes but I have a LOT of scripts to build a docker image and really need to automate everything.

Build of esp32 also just failed, I switched back to to using the default build path and again, problem went away.

Chances are there will be other ports that fail as well, I guess there's no global fix for this ??

dpgeorge, 2021-12-23:
All ports except esp8266 should build in any directory.  This is already all automated by the `tools/autobuild/*` scripts.

dpgeorge, 2021-12-23:
See #8107 for a fix for this issue.

pacmac, 2021-12-23:
Hmmm rp2 fails for me when using a custom path, when I use the default path it succeeds:

```
# make BOARD=PICO BUILD=/usr/share/upydev/work/boards/rp2/PICO/build clean
rm -f -rf /usr/share/upydev/work/boards/rp2/PICO/build
root@dd7f00d8dc1a:/usr/share/upydev/work/micropython/ports/rp2# make BOARD=PICO BUILD=/usr/share/upydev/work/boards/rp2/PICO/build
[ -d /usr/share/upydev/work/boards/rp2/PICO/build ] || cmake -S . -B /usr/share/upydev/work/boards/rp2/PICO/build -DPICO_BUILD_DOCS=0 -DMICROPY_BOARD=PICO
PICO_SDK_PATH is /usr/share/upydev/work/micropython/lib/pico-sdk
Defaulting PICO_PLATFORM to rp2040 since not specified.
Defaulting PICO platform compiler to pico_arm_gcc since not specified.
PICO compiler is pico_arm_gcc
-- The C compiler identification is GNU 10.3.1
-- The CXX compiler identification is GNU 10.3.1
-- The ASM compiler identification is GNU
-- Found assembler: /usr/bin/arm-none-eabi-gcc
PICO target board is pico.
Using board configuration from /usr/share/upydev/work/micropython/lib/pico-sdk/src/boards/include/boards/pico.h
-- Found Python3: /usr/bin/python3.5 (found version "3.5.3") found components: Interpreter
TinyUSB available at /usr/share/upydev/work/micropython/lib/tinyusb/src/portable/raspberrypi/rp2040; enabling build support for USB.
Found User C Module(s):
-- Configuring done
-- Generating done
-- Build files have been written to: /usr/share/upydev/work/boards/rp2/PICO/build
make -s -C /usr/share/upydev/work/boards/rp2/PICO/build
[  0%] Building C object CMakeFiles/micropy_extmod_btree.dir/usr/share/upydev/work/micropython/lib/berkeley-db-1.xx/btree/bt_close.c.obj
[  1%] Building C object CMakeFiles/micropy_extmod_btree.dir/usr/share/upydev/work/micropython/lib/berkeley-db-1.xx/btree/bt_conv.c.obj
[  1%] Building C object CMakeFiles/micropy_extmod_btree.dir/usr/share/upydev/work/micropython/lib/berkeley-db-1.xx/btree/bt_debug.c.obj
[  1%] Building C object CMakeFiles/micropy_extmod_btree.dir/usr/share/upydev/work/micropython/lib/berkeley-db-1.xx/btree/bt_delete.c.obj
[  2%] Building C object CMakeFiles/micropy_extmod_btree.dir/usr/share/upydev/work/micropython/lib/berkeley-db-1.xx/btree/bt_get.c.obj
[  2%] Building C object CMakeFiles/micropy_extmod_btree.dir/usr/share/upydev/work/micropython/lib/berkeley-db-1.xx/btree/bt_open.c.obj
[  2%] Building C object CMakeFiles/micropy_extmod_btree.dir/usr/share/upydev/work/micropython/lib/berkeley-db-1.xx/btree/bt_overflow.c.obj
[  3%] Building C object CMakeFiles/micropy_extmod_btree.dir/usr/share/upydev/work/micropython/lib/berkeley-db-1.xx/btree/bt_page.c.obj
[  3%] Building C object CMakeFiles/micropy_extmod_btree.dir/usr/share/upydev/work/micropython/lib/berkeley-db-1.xx/btree/bt_put.c.obj
[  3%] Building C object CMakeFiles/micropy_extmod_btree.dir/usr/share/upydev/work/micropython/lib/berkeley-db-1.xx/btree/bt_search.c.obj
[  4%] Building C object CMakeFiles/micropy_extmod_btree.dir/usr/share/upydev/work/micropython/lib/berkeley-db-1.xx/btree/bt_seq.c.obj
[  4%] Building C object CMakeFiles/micropy_extmod_btree.dir/usr/share/upydev/work/micropython/lib/berkeley-db-1.xx/btree/bt_split.c.obj
[  4%] Building C object CMakeFiles/micropy_extmod_btree.dir/usr/share/upydev/work/micropython/lib/berkeley-db-1.xx/btree/bt_utils.c.obj
[  5%] Building C object CMakeFiles/micropy_extmod_btree.dir/usr/share/upydev/work/micropython/lib/berkeley-db-1.xx/mpool/mpool.c.obj
[  5%] Built target micropy_extmod_btree
Scanning dependencies of target bs2_default
[  5%] Building ASM object pico-sdk/src/rp2_common/boot_stage2/CMakeFiles/bs2_default.dir/compile_time_choice.S.obj
[  6%] Linking ASM executable bs2_default.elf
[  6%] Built target bs2_default
[  7%] Generating bs2_default.bin
[  7%] Generating bs2_default_padded_checksummed.S
[  7%] Built target bs2_default_padded_checksummed_asm
[  7%] Creating directories for 'ELF2UF2Build'
[  8%] No download step for 'ELF2UF2Build'
[  8%] No update step for 'ELF2UF2Build'
[  9%] No patch step for 'ELF2UF2Build'
[  9%] Performing configure step for 'ELF2UF2Build'
-- The C compiler identification is GNU 6.3.0
-- The CXX compiler identification is GNU 6.3.0
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: /usr/bin/cc - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: /usr/bin/c++ - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Configuring done
-- Generating done
-- Build files have been written to: /usr/share/upydev/work/boards/rp2/PICO/build/elf2uf2
[  9%] Performing build step for 'ELF2UF2Build'
[ 50%] Building CXX object CMakeFiles/elf2uf2.dir/main.cpp.o
[100%] Linking CXX executable elf2uf2
[100%] Built target elf2uf2
[  9%] No install step for 'ELF2UF2Build'
[  9%] Completed 'ELF2UF2Build'
[  9%] Built target ELF2UF2Build
[  9%] Generating genhdr/mpversion.h
Traceback (most recent call last):
  File "/usr/share/upydev/work/micropython/py/makeversionhdr.py", line 117, in <module>
    make_version_header(sys.argv[1])
  File "/usr/share/upydev/work/micropython/py/makeversionhdr.py", line 79, in make_version_header
    info = get_version_info_from_docs_conf()
  File "/usr/share/upydev/work/micropython/py/makeversionhdr.py", line 67, in get_version_info_from_docs_conf
    for line in f:
  File "/usr/lib/python3.5/encodings/ascii.py", line 26, in decode
[... cut ...]
UnicodeDecodeError: 'ascii' codec can't decode byte 0xc2 in position 2154: ordinal not in range(128)
CMakeFiles/firmware.dir/build.make:73: recipe for target 'genhdr/mpversion.h' failed
make[3]: *** [genhdr/mpversion.h] Error 1
CMakeFiles/Makefile2:1356: recipe for target 'CMakeFiles/firmware.dir/all' failed
make[2]: *** [CMakeFiles/firmware.dir/all] Error 2
Makefile:90: recipe for target 'all' failed
make[1]: *** [all] Error 2
Makefile:18: recipe for target 'all' failed
make: *** [all] Error 2
```

pacmac, 2021-12-23:
Strange, the version of gcc-arm-none-eabi I have installed is 10.3.1, and when I build using a custom path:

```
-- The C compiler identification is GNU 6.3.0
-- The CXX compiler identification is GNU 6.3.0
```

But when I build using the default path:

```
-- The C compiler identification is GNU 10.3.1
-- The CXX compiler identification is GNU 10.3.1
```

pacmac, 2021-12-23:
2 Builds from same folder:

```
###############
#### FAILS ####

# make BOARD=PICO BUILD=/usr/share/upydev/work/boards/rp2/PICO/build
[ -d /usr/share/upydev/work/boards/rp2/PICO/build ] || cmake -S . -B /usr/share/upydev/work/boards/rp2/PICO/build -DPICO_BUILD_DOCS=0 -DMICROPY_BOARD=PICO
PICO_SDK_PATH is /usr/share/upydev/work/micropython/lib/pico-sdk
Defaulting PICO_PLATFORM to rp2040 since not specified.
Defaulting PICO platform compiler to pico_arm_gcc since not specified.
PICO compiler is pico_arm_gcc
-- The C compiler identification is GNU 10.3.1
-- The CXX compiler identification is GNU 10.3.1
-- The ASM compiler identification is GNU
-- Found assembler: /usr/bin/arm-none-eabi-gcc
PICO target board is pico.

...
more output...
...

[  9%] Built target ELF2UF2Build
[  9%] Generating genhdr/mpversion.h
Traceback (most recent call last):
  File "/usr/share/upydev/work/micropython/py/makeversionhdr.py", line 117, in <module>
    make_version_header(sys.argv[1])
  File "/usr/share/upydev/work/micropython/py/makeversionhdr.py", line 79, in make_version_header
    info = get_version_info_from_docs_conf()
  File "/usr/share/upydev/work/micropython/py/makeversionhdr.py", line 67, in get_version_info_from_docs_conf
    for line in f:
  File "/usr/lib/python3.5/encodings/ascii.py", line 26, in decode
    return codecs.ascii_decode(input, self.errors)[0]
UnicodeDecodeError: 'ascii' codec can't decode byte 0xc2 in position 2154: ordinal not in range(128)
CMakeFiles/firmware.dir/build.make:73: recipe for target 'genhdr/mpversion.h' failed
make[3]: *** [genhdr/mpversion.h] Error 1
CMakeFiles/Makefile2:1356: recipe for target 'CMakeFiles/firmware.dir/all' failed
make[2]: *** [CMakeFiles/firmware.dir/all] Error 2
Makefile:90: recipe for target 'all' failed
make[1]: *** [all] Error 2
Makefile:18: recipe for target 'all' failed
make: *** [all] Error 2

### FAILS ####
##############
```

```
############
#### OK ####

# make BOARD=PICO BUILD=build-PICO
[ -d build-PICO ] || cmake -S . -B build-PICO -DPICO_BUILD_DOCS=0 -DMICROPY_BOARD=PICO
make -s -C build-PICO
[  5%] Built target micropy_extmod_btree
[  6%] Built target bs2_default
[  7%] Built target bs2_default_padded_checksummed_asm
[  7%] Performing build step for 'ELF2UF2Build'
[100%] Built target elf2uf2
[  7%] No install step for 'ELF2UF2Build'
[  7%] Completed 'ELF2UF2Build'
[  9%] Built target ELF2UF2Build
[  9%] Generating genhdr/mpversion.h
[ 10%] Generating genhdr/qstrdefs.collected.h
QSTR not updated
[ 11%] Generating frozen_content.c
Consolidate compiler generated dependencies of target firmware
[ 11%] Generating genhdr/mpversion.h
[ 11%] Generating genhdr/qstrdefs.collected.h
QSTR not updated
[ 12%] Building C object CMakeFiles/firmware.dir/main.c.obj
[ 13%] Building C object CMakeFiles/firmware.dir/moduos.c.obj
[ 13%] Generating frozen_content.c
[ 13%] Building C object CMakeFiles/firmware.dir/frozen_content.c.obj
[ 13%] Linking CXX executable firmware.elf
   text	   data	    bss	    dec	    hex	filename
 289608	     92	 201816	 491516	  77ffc	/usr/share/upydev/work/micropython/ports/rp2/build-PICO/firmware.elf
[100%] Built target firmware
root@dd7f00d8dc1a:/usr/share/upydev/work/micropython/ports/rp2# make BOARD=PICO BUILD=build-PICO clean
rm -f -rf build-PICO
root@dd7f00d8dc1a:/usr/share/upydev/work/micropython/ports/rp2# make BOARD=PICO BUILD=build-PICO
[ -d build-PICO ] || cmake -S . -B build-PICO -DPICO_BUILD_DOCS=0 -DMICROPY_BOARD=PICO
PICO_SDK_PATH is /usr/share/upydev/work/micropython/lib/pico-sdk
Defaulting PICO_PLATFORM to rp2040 since not specified.
Defaulting PICO platform compiler to pico_arm_gcc since not specified.
PICO compiler is pico_arm_gcc
-- The C compiler identification is GNU 10.3.1
-- The CXX compiler identification is GNU 10.3.1
-- The ASM compiler identification is GNU
-- Found assembler: /usr/bin/arm-none-eabi-gcc
PICO target board is pico.

...
more output...
...

[ 99%] Building C object CMakeFiles/firmware.dir/usr/share/upydev/work/micropython/lib/tinyusb/src/class/vendor/vendor_device.c.obj
[ 99%] Building C object CMakeFiles/firmware.dir/usr/share/upydev/work/micropython/lib/tinyusb/src/class/video/video_device.c.obj
[100%] Building C object CMakeFiles/firmware.dir/usr/share/upydev/work/micropython/lib/pico-sdk/src/rp2_common/pico_fix/rp2040_usb_device_enumeration/rp2040_usb_device_enumeration.c.obj
[100%] Linking CXX executable firmware.elf
   text	   data	    bss	    dec	    hex	filename
 289608	     92	 201816	 491516	  77ffc	/usr/share/upydev/work/micropython/ports/rp2/build-PICO/firmware.elf
[100%] Built target firmware

#### OK ####
############
```

pacmac, 2021-12-24:
The RPT problem appears to be caused by locales, not sure why it's only a problem when using a custom build path, but installing and setting locales solved this problem.

```
## Locales
RUN apt update
RUN apt -y install locales
RUN sed -i 's/^# *\(en_US.UTF-8\)/\1/' /etc/locale.gen && locale-gen
ENV LC_ALL=en_US.UTF-8
ENV LANG=en_US.UTF-8
ENV LANGUAGE=en_US.UTF-8
```

PR #8107 (2022-01-04): esp8266: Allow building a board to any dest directory.
Fixes #8106.
closed 2022-01-04 by dpgeorge