micropython/micropython #8108: PWM, Modes, and channels issue

kthod861, 2021-12-23:
Hello,

While trying to get the same behavior as "ledc"+"ledcWriteTone" using micropython, i ended up stuck with no solution to get the 8 PWM output with different frequencies .

Goal: get 8 frequencies outputs (fixed duty) that can be modified using python as the ESP32 is able to provide this ( tested using ledc on a bare new ESP32 board).
But i probably have missed something here.

For example, using "esp32-20210902-v1.17.bin", and by default testing the example provided here is a fail:
- https://docs.micropython.org/en/latest/esp32/tutorial/pwm.html

i say fail as the output of it i absolutly not :

- `PWM(Pin(15), freq=100, duty=64, resolution=10, mode=0, channel=0, timer=0)`

but rather:

- `PWM(Pin(15), freq=100, duty=64 )`

Did also a ton of test by myself without understanding the logic behind mode and channel selection when creating a PWM object.
It sounds like it dynamically choose timers depending on init frequency but i'm really not sure about that.

Any help is welcome

robert-hh, 2021-12-23:
PWM has been changed to support different frequencies. Try the latest daily build.

kthod861, 2021-12-23:
Awesome ! , using latest nightly solved it.
Thanks a lot

IhorNehrutsa, 2022-01-14:
@kthod861 please close the issue if all right

closed 2022-01-14 by kthod861