micropython/micropython #8109: esp32-c3 port used wrong rmt channel for NeoPixel driver

innovak, 2021-12-23:
esp32-c3 only has channel 0 and channel 1 can use for TX, 2 and 3 are rx only,but in modesp32.h  #define MICROPY_HW_ESP32_RMT_CHANNEL_BITSTREAM (RMT_CHANNEL_MAX - 1)  will use rx only channel for neopixel lib.

Walkline80, 2021-12-27:
I have tried to modify `ports/esp32/modesp32.h`

```c
// Reserve the last channel for machine.bitstream.
#if CONFIG_IDF_TARGET_ESP32C3
    #define MICROPY_HW_ESP32_RMT_CHANNEL_BITSTREAM (0)
#else
    #define MICROPY_HW_ESP32_RMT_CHANNEL_BITSTREAM (RMT_CHANNEL_MAX - 1)
#endif
#endif // MICROPY_INCLUDED_ESP32_MODESP32_H
```

And now, my esp32c3 board can drive led strip via NexPixel

dpgeorge, 2022-02-01:
In 1f04a9a1fcd4172f7c721e05a4515c9c8bb0e8c9 the default channel has been changed to `SOC_RMT_TX_CANDIDATES_PER_GROUP - 1`, which should work on all known ESP32 variants (it will be 1 on ESP32-C3).

If this still doesn't work then please try running `esp32.RMT.bitstream_channel(0)` to select channel 0, and report here your findings.

andypiper, 2022-02-02:
Aha! `esp32.RMT.bitstream_channel(0)` works perfectly here - previously I was seeing the first pixel always set to bright green when I didn't set it at all (and a mixture of green+whatever I asked for, otherwise); now, it is individually controllable. Going to run through the whole board to check, but this is looking like the answer...

jonnor, 2024-09-08:
There seems to be a fix available.

closed 2024-09-08 by jonnor