micropython/micropython #8112: esp32/machine_sdcard.c: Invalid pin numbers are allowwed.

IhorNehrutsa, 2021-12-24:
https://github.com/micropython/micropython/blob/7566d107d5acc82a45df0dd5c5803bf5d2564043/ports/esp32/machine_sdcard.c#L72-L79

This function does not check int value.
```
    return MP_OBJ_SMALL_INT_VALUE(arg);
```
may return invalid pin numbers like 12345

nickovs, 2021-12-25:
Thanks for the feedback. I think that this should be OK since if you pass an invalid pin ID as an argument then it gets picked up shortly afterwards. The only code that uses that function is the macro [`SET_CONFIG_PIN`](https://github.com/micropython/micropython/blob/7566d107d5acc82a45df0dd5c5803bf5d2564043/ports/esp32/machine_sdcard.c#L81-L83). This macro is only used in two blocks, [here](https://github.com/micropython/micropython/blob/7566d107d5acc82a45df0dd5c5803bf5d2564043/ports/esp32/machine_sdcard.c#L236-L244) and [here](https://github.com/micropython/micropython/blob/7566d107d5acc82a45df0dd5c5803bf5d2564043/ports/esp32/machine_sdcard.c#L254-L265), both of which ultimately pass the pin numbers to the functions that initialise the SPI or MMC interfaces _and check the errors_, so you should end up with a suitable exception if an invalid number is passed. Furthermore, the range of legitimate values depends on which signal is being configured and which sort of interface is being used, so this specific function is not in a position to range check the values.

jonnor, 2024-09-08:
Based on previous comment, I believe this is a wont-fix.

closed 2025-02-04 by IhorNehrutsa