micropython/micropython #8114: SAMD - XIAO

pacmac, 2021-12-25:
I have also reported this on the micropythong forum.

If I load the circuitpython firmware then neither of these issues exist.

1) Unable to Mount Mass Storage Device
After copying the micropython firmware.uf2 file to the device, it reboots as expected but never mounts the mass storage volume.

If I copy the circuitpython .uf2 file to the device, it reboots and mounts as a mass storage device and works great.

2) Floating Point not enabled:
```
>>> 4.1
Traceback (most recent call last):
  File "<stdin>", line 1
SyntaxError: decimal numbers not supported
>>>
```

pacmac, 2021-12-27:
I can enable floating point with:
```
#define MICROPY_FLOAT_IMPL              (MICROPY_FLOAT_IMPL_FLOAT)
#define MICROPY_PY_BUILTINS_COMPLEX     (0)
#define MICROPY_PY_MATH                 (0)
#define MICROPY_PY_CMATH                (0)
```

Is it possible to enable I2C and mass storage support ?

pacmac, 2021-12-30:
I tried adding the following but this did not enable I2C after a rebuild:

```
#define MICROPY_PY_MACHINE_I2C          (1)
```

How can I enable I2C and mass storage (as on the circuitpython version)

Thanks

dpgeorge, 2021-12-30:
The samd port does not yet have hardware I2C support.  But it should be relatively easy to enable software I2C.

It also doesn't have USB mass storage.

pacmac, 2021-12-30:
Do you mean that the samd device hardware does not support mass storage or that the port does not yet support it ?

The circuitpython port supports mass storage and I2C, I looked at the configs from their github to copy into the micropython port but it does not seem compatible.

I am trying to use this device to replace an ESP32 device, all that is needed is I2C support.

dpgeorge, 2021-12-30:
> Do you mean that the samd device hardware does not support mass storage or that the port does not yet support it ?

The port does not yet support it.

To get software I2C requires an implementation of an accurate microsecond clock, and the `mp_hal_delay_us_fast()` function.

robert-hh, 2022-12-15:
Hardware I2C, SPI, UART, etc. support is available now.

closed 2022-12-15 by robert-hh