micropython/micropython #8115: drivers/display/ssd1306.py  ```writevto``` does not exits and self_write_list define but not used

PaulskPt, 2021-12-26:
By search for some info on framebuffer I passed across the file ```.../drivers/display/ssd1306.py```
I would like to bring under your attention the following:

Line 117:         ```self.write_list = [b"\x40", None]  # Co=0, D/C#=1 ```
```self.write_list``` is defined here but never used

Line: 127   ```self.i2c.writevto(self.addr, self.write_list)```
```writevto``` does not exist". Did you mean ```writeto``` ?

robert-hh, 2021-12-26:
Which board and firmware version do you use? The writevto() method was added to MicroPython quite a while ago.
b.t.w.: self.write_list is used in lines 126 and 127.

PaulskPt, 2021-12-26:
Hi Robert. Thank you for a swift reply. OK about the writevto() method. I was not aware of that. About self.write_list being used in lines 126 and 127. I missed that! Oops, I guess I created an issue unnecessary. Excuse me for bothering.

PaulskPt, 2021-12-26:
Hi Robert. Thank you for a swift reply. OK about the writevto() method. I was not aware of that. About self.write_list . I missed that! Oops, I guess I created an issue unnecessary. Excuse me for bothering.

closed 2021-12-26 by PaulskPt