micropython/micropython #8118: Raspberry Pi Pico - Not showing in Windows Explorer

ninjaWolf8724, 2021-12-27:
I have flashed MicroPython to a Raspberry Pi Pico. I am am able to complete the flash process fine, but the device won't show in Windows Explorer, and I want to install modules, as well as a new code.py. But it shows up in Thonny just fine.
I am using MicroPython 1.17 and my host OS is Windows 11 Pro 64-bit

robert-hh, 2021-12-27:
That is the intended behavior. It only shows a drive on the PC for the firmware update. During regilart operation you have to use tools like Thonny or mpremote to transfer files between the PC and the RPi Pico.

Kyuchumimo, 2021-12-28:
This is normal behavior, it also happens in Linux. Using MicroPython firmware the board uses FS mode.
Using Adafruit CircuitPython firmware you can see the board as a mass storage device.

pavelrevak, 2022-01-04:
MicroPython support mass storage only with STM microcontrollers
for RP there is circuitpython with mass storage support.

davehylands, 2022-01-04:
> MicroPython support mass storage only with STM microcontrollers
> for RP there is circuitpython with mass storage support.

See this thread for more details: https://forum.micropython.org/viewtopic.php?f=21&t=11054&p=60773

domdfcoding, 2022-07-13:
Is there a plan to support mass storage? I noticed that Anvil's Pico W firmware (available [here](https://github.com/anvil-works/anvil-pico/releases)) enables USB mass storage (see [this Tom's Hardware guide](https://www.tomshardware.com/how-to/build-raspberry-pi-pico-w-web-app)), but I'm not sure how they achieve that.

jimmo, 2022-07-14:
> Is there a plan to support mass storage? I noticed that Anvil's Pico W firmware (available [here](https://github.com/anvil-works/anvil-pico/releases)) enables USB mass storage (see [this Tom's [...]

Mass storage can be enabled by setting the `MICROPY_HW_USB_MSC` build option. The Arduino Nano RP2040 Connect board does this for example: https://github.com/micropython/micropython/blob/master/ports/rp2/boards/ARDUINO_NANO_RP2040_CONNECT/mpconfigboard.h#L14

Mass storage requires that we use the FAT filesystem (rather than LittleFS) which is significantly worse for performance and reliability.

More background here https://github.com/micropython/micropython/pull/7402#issuecomment-1048777491

closed 2022-07-14 by jimmo