micropython/micropython #8120: Failing to build rp2040 firmware for the ARDUINO_NANO_RP2040_CONNECT board

teuler, 2021-12-27:
Hi
I am trying to build the rp2 firmware for the ARDUINO_NANO_RP2040_CONNECT board. The build process runs fine until the very end, when it hits an "cannot find entry symbol _entry_point" error:
```
make -j 8 BOARD=ARDUINO_NANO_RP2040_CONNECT 
```
results in:
```
[ 96%] Generating frozen_content.c
[ 96%] Building C object CMakeFiles/firmware.dir/frozen_content.c.obj
[ 97%] Linking C executable firmware.elf
/usr/lib/gcc/arm-none-eabi/9.2.1/../../../arm-none-eabi/bin/ld: warning: cannot find entry symbol _entry_point; defaulting to 0000000010000100
   text    data     bss     dec     hex filename
   8644     104      28    8776    2248 /home/tom/pico/micropython/ports/rp2/build-ARDUINO_NANO_RP2040_CONNECT/firmware.elf
[100%] Built target firmware
```
The resulting UF2 file is 18 kB in size. The default build process targeting the Pico works fine.
Any hints?
Thanks!

P.S. Windows 10, Linux subsystem, Ubuntu 20.04

teuler, 2021-12-28:
... maybe I should explain why I want to rebuild the firmware. I would like to add the `ulab` module. For the default board (`PICO`) this works.

dpgeorge, 2021-12-29:
It builds OK for me (under Arch Linux) with arm-none-eabi-gcc v11.2.0.

Try without the "-j 8", and make sure the git submodules are all up to date.

teuler, 2021-12-29:
Thanks for the reply. The hint was the gcc version. Default under Ubuntu 20.xx seems to be 9.x. 
Forcing a newer gcc version using 
```
sudo apt install gcc-10
```
did the trick.
P.S. gcc v11 was not found (?) but does not seem to be needed.

closed 2021-12-29 by teuler