micropython/micropython #8122: machine.PWM(Pin(pin)).freq() return invalid and inconsistent frequency values

Kyuchumimo, 2021-12-28:
Development board: Raspberry Pi Pico with RP2040
Firmware: MicroPython v1.17 on 2021-09-02

Test script:
```
from machine import Pin, PWM
import time

pwm = PWM(Pin(15))
pwm.freq(100000)
pwm.duty_u16(32767)

freq = 1000000

while pwm.freq() < 4000000:
    print(freq, pwm.freq())
    pwm.freq(freq)
    freq += 100
    time.sleep(0.01)
```
As you run the code, you may notice that the frequency printed by machine.PWM(Pin(pin)).freq() bounces between values. This is what I mean by inconsistency. Additionally, there are invalid values have no effect on the actual output frequency.
For example:
Setting a frequency of 3579545 Hz will return 3584229 Hz according to machine.PWM(Pin(pin)).freq(), which is not the case, since the frequency delivered is actually 3571428 Hz.

All valid values for the PWM frequency are obtained by dividing the base clock frequency by integers.
125000000 Hz / N = frequency

IhorNehrutsa, 2021-12-29:
https://docs.micropython.org/en/latest/library/machine.PWM.html#limitations-of-pwm

robert-hh, 2021-12-29:
There is more than just a simple divider in place. And even that seems to be off by 1. To get a frequency of 1 MHz, I have to request 1.0081 MHz. That's just one of in a 125/N divider. And even then it is a average frequency with a frequency/phase jitter.

[image]
.

Kyuchumimo, 2021-12-29:
> https://docs.micropython.org/en/latest/library/machine.PWM.html#limitations-of-pwm
```
... Typically the PWM frequency is obtained by dividing some integer base frequency by an integer divider. ...
```

Is just what i say.

Applying the formulas:
125000000 Hz / 3579545 Hz = 34.9201515814 ~ 35
125000000 Hz / 35 = **3571428.57143 Hz ~ 3571428 Hz**

3571428 Hz is different from the 3584229 Hz that is printed by .freq(). Something is wrong with this function.

robert-hh, 2021-12-30:
Trying to set the frequency 3.579545 MHz, dutyu16=32768 (50%) I get even worse figures. Actually the resulting signal alternates between 3.37649 MHz and 4.47424 MHz. That's what I have seen before: The divider seems to be off by one.
SO I agree: something is wrong with the PWM. The other not-so-nice behavior is, that the duty rate changes when the frequency is changed. But that should be easy to fix.

robert-hh, 2021-12-30:
With a small fix in the code, I get 1000000MHz when set to 1 MHz, and
3584238 when set to 3579545, while p.freq() reports 3584229, which is pretty precise.
Due to the fractional prescaler, the frequency alternates between ~3569350 and ~3678690

dpgeorge, 2022-01-06:
The issue with off-by-1 frequency setting/getting is fixed by 9e56e630caeec9077cd95ac33b702a7b2187300e

So I think this issue can be closed.  If there are still problems, please reopen with details.

closed 2022-01-06 by dpgeorge

pdg137, 2022-11-22:
Hello,
I noticed some inconsistent PWM frequencies around 1770 Hz and wonder if it's related to this issue. I plan to investigate more but wanted to put this here in case the answer is obvious to someone:

```python
MPY: soft reboot
MicroPython v1.19.1 on 2022-06-18; Raspberry Pi Pico with RP2040
Type "help()" for more information.
>>> from machine import Pin, PWM
>>> PWM(Pin(25)).freq(1760)
>>> PWM(Pin(25)).freq()
1838
>>> PWM(Pin(25)).freq(1770)
>>> PWM(Pin(25)).freq()
1808
>>> 
```

PWM of 1770 generates a lower frequency than PWM of 1760 (and played through a speaker it sounds lower, too 😉).  I understand that we can't make every frequency perfectly, but it seems like they should at least be monotonically increasing.

pdg137, 2022-11-22:
I think it's the same issue. Relevant code seems to be here:

https://github.com/micropython/micropython/blob/master/ports/rp2/machine_pwm.c#L116-L132

There are a few of problems:

1. Not taking advantage of the fractional division in DIV_FRAC (maybe that's intentional?)
2. Not taking advantage of the integer divider DIV_INT (is this also intentional?)
3. When increasing TOP and doing a corresponding decrease to DIV_FRAC, like this:
    ```c
    div16_top /= 2;
    top *= 2;
    ````
    the resulting frequency either stays the same or increases.  It's bad that it's biased upward but also the randomness introduced here is probably what causes the generated frequencies to not be monotonic.

I would like to propose a better algorithm, but could someone explain whether points 1 and 2 above (having DIV_INT=0 and DIV_FRAC be a multiple of 16) are intended for some reason?