micropython/micropython #8123: STM32 port does not compile with WIZNET5K and LWIP

bartagergely, 2021-12-28:
Running the following:
make MICROPY_PY_WIZNET5K=5500 MICROPY_PY_LWIP=1 BOARD=WeAct_F411CE -j4
produces an error:
```
mpnetworkport.c: In function 'pyb_lwip_poll':
mpnetworkport.c:58:5: error: implicit declaration of function 'wiznet5k_poll' [-Werror=implicit-function-declaration]
   58 |     wiznet5k_poll();
      |     ^~~~~~~~~~~~~
cc1: all warnings being treated as errors
make: *** [../../py/mkrules.mk:77: build-WeAct_F411CE/mpnetworkport.o] Error 1
```

It does not depend of the board, tried with "factory provided" STM32 board types too, but they ran into the same error.

Any suggestion is welcomed!

dpgeorge, 2021-12-29:
Thanks for the report, I can confirm the issue.

Looks like it was introduced in 4dba04a50fea01f6f8fec83d64f958f8d14e285a.  The fix would be to add back the declaration of `void wiznet5k_poll(void);` somewhere (in `mpnetwork.port.c` would get it working at least).

bartagergely, 2021-12-29:
@dpgeorge 
Thank you for the quick response. I tried to add the declaration:
```
#include "extmod/network_cyw43.h"
#include "drivers/cyw43/cyw43.h"

void wiznet5k_poll(void);

// Poll lwIP every 128ms
#define LWIP_TICK(tick) (((tick) & ~(SYSTICK_DISPATCH_NUM_SLOTS - 1) & 0x7f) == 0)
```

But now I got this failure message:
```
CC ../../lib/lwip/src/core/ipv6/mld6.c
CC ../../lib/lwip/src/core/ipv6/nd6.c
CC ../../lib/lwip/src/netif/ethernet.c
CC network_wiznet5k.c
network_wiznet5k.c: In function 'wiznet5k_make_new':
network_wiznet5k.c:272:32: error: comparison of distinct pointer types lacks a cast [-Werror]
  272 |     if (wiznet5k_obj.base.type == &mod_network_nic_type_wiznet5k) {
      |                                ^~
network_wiznet5k.c:280:28: error: assignment to 'const mp_obj_type_t *' {aka 'const struct _mp_obj_type_t *'} from incompatible pointer type 'const struct _mod_network_nic_type_t *' [-Werror=incompatible-pointer-types]
  280 |     wiznet5k_obj.base.type = &mod_network_nic_type_wiznet5k;
      |                            ^
network_wiznet5k.c: At top level:
network_wiznet5k.c:458:21: error: conflicting types for 'mod_network_nic_type_wiznet5k'
  458 | const mp_obj_type_t mod_network_nic_type_wiznet5k = {
      |                     ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~
In file included from ../../py/mpconfig.h:62,
                 from ../../py/mpstate.h:31,
                 from ../../py/runtime.h:29,
                 from network_wiznet5k.c:30:
./mpconfigport.h:256:45: note: previous declaration of 'mod_network_nic_type_wiznet5k' was here
  256 | extern const struct _mod_network_nic_type_t mod_network_nic_type_wiznet5k;
      |                                             ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~
cc1: all warnings being treated as errors
make: *** [../../py/mkrules.mk:77: build-WeAct_F411CE/network_wiznet5k.o] Error 1
```

iabdalkader, 2021-12-29:
There's actually more than one issue with that module, I sent a fix here: #8126

bartagergely, 2021-12-29:
@iabdalkader @dpgeorge thank you for the ultra fast response. Now it is working as expected:
```
>>> import network
>>> nic = network.WIZNET5K(pyb.SPI(2), pyb.Pin.board.PB12, pyb.Pin.board.PB10)
>>> nic.active()
False
>>> nic.active(True)
>>> nic.ifconfig('dhcp')
>>> nic.ifconfig()
('192.168.62.119', '255.255.255.0', '192.168.62.1', '192.168.62.1')
```

dpgeorge, 2021-12-30:
Thanks for confirming the fix.

Fixed by 8af9dbbde1fca07371aa13a8276c8e764edac28c

closed 2021-12-30 by dpgeorge