micropython/micropython #8129: Document values of CAN bus Constants

keelung-yang, 2021-12-31:
Hi,
Since CAN.send() have no error code returned, I've to check CAN.state() and got 4. But what 4 stand for? Then I've to search code to figure out this error code.
So, it's very convenient if values of CAN bus Constants can be documented clearly.
[image]

iabdalkader, 2022-02-16:
> But what 4 stand for? Then I've to search code to figure out this error code.

You don't need to know the value of the error symbol to know which error it is, you could just test using the symbols:

```Python
err = CAN.state()
if err == CAN.BUS_OFF:
    print("bus off")
elif err == CAN.ERROR_PASSIVE:
    print("error passive")
```

jonnor, 2024-09-08:
The documentation is found in the docs/ directory, and contributions in form of merge requests are welcomed!