micropython/micropython #8133: Random SHA1 digests in ESP32 S2

unaiur, 2022-01-03:
I get a different SHA1 digest every time I run this piece of code using the REPL:

```
>>> webkey='dlz94Y64esOVcG/1+TUdtg=='
>>> d = hashlib.sha1(webkey)
>>> d.update(b"258EAFA5-E914-47DA-95CA-C5AB0DC85B11")
>>> respkey = d.digest()
>>> binascii.b2a_base64(respkey)[:-1]
b'wK7rH6Gbv1vVlCptCUc2KxE3adQ='
>>> webkey=b'dlz94Y64esOVcG/1+TUdtg=='
>>> d = hashlib.sha1(webkey)
>>> d.update(b"258EAFA5-E914-47DA-95CA-C5AB0DC85B11")
>>> respkey = d.digest()
>>> binascii.b2a_base64(respkey)[:-1]
b'OKH4LEOZweHZKCerLXu/xPVCdio='
>>> d = hashlib.sha1(webkey)
>>> d.update(b"258EAFA5-E914-47DA-95CA-C5AB0DC85B11")
>>> respkey = d.digest()
>>> binascii.b2a_base64(respkey)[:-1]
b'6zKTQ0bhuGDNRiDRicxN0u5WTxo='
```

This code is used in WEBREPL here: 

https://github.com/micropython/micropython/blob/4d9e657f0ee881f4a41093ab89ec91d03613744d/extmod/webrepl/websocket_helper.py#L44-L47

Of course, WebREPL is not working at all.

unaiur, 2022-01-03:
A more simple example:
```
>>> binascii.hexlify(hashlib.sha1('mango').digest())
b'07d4f0247b2e607d6d1a833e00cfd42f1e74778c'
>>> binascii.hexlify(hashlib.sha1('mango').digest())
b'51f8c326df0283b468a0beaa1f7077355331341e'
>>> binascii.hexlify(hashlib.sha1('mango').digest())
b'e631b8d9854bc6d114b272ad89f7907602407e2b'
>>> binascii.hexlify(hashlib.sha1('mango').digest())
b'd481fe9a7f62e455aa250ef2847eac62a6d32e4e'
>>> binascii.hexlify(hashlib.sha1('mango').digest())
b'f748f35561519a372042bd20eff49c90317d054f'
```

unaiur, 2022-01-03:
Seems to be related with https://github.com/espressif/esp-idf/issues/7944, which got resolved 6 weeks ago.
We need to wait for a new SDK release.

dpgeorge, 2022-01-07:
Interesting bug!  A workaround would be to use a build without SPIRAM enabled.  You can also try building with the very latest IDF that includes the fix.

unaiur, 2022-03-16:
This bug is also the root cause of #7726

dpgeorge, 2022-03-17:
I can confirm this is fixed on the latest ESP32-S2 builds, when using IDF 4.4.0.  So I will close the issue.

closed 2022-03-17 by dpgeorge