micropython/micropython #8134: Micropython Library Build?

duaneellissd, 2022-01-03:
Hi - I sort of need to build MicroPython into a library rather then the application, that's my thinking any way.

Here's why:  The number one issue is my final stuff needs to be buildable under "odd-ball" versions of Eclipse from various chip vendors - otherwise the debugger does not work and other things don't work well.

An example of ODD BALL Eclipse that comes from MicroSemi - and the Eclipse that comes from Xilinx, you could also say ST Microsystems STM CUBE is another good example.

My general idea is this:
    Building MicroPython on Windows (cross to an ARM target) - will not work, the Makefiles assume too much Linux stuff.
The result of that build is a static library

   I can then export that static library (and some [a few]) header files to the primary application.

The assumption here is: The "primary app" - is not Micro Python, instead - MicroPython is embedded as the secondary application inside a larger application.

This would or could - just another directory in the examples folder.

I would think long term (if other maintainers are interested) the existing examples could some day shift to: "build the library" then "build the app" which links the library.  Don't want to suggest one "installs" the cross built microPython library

Q: Does this sound like a reasonable idea?

dpgeorge, 2022-01-07:
You can see `examples/embedding/` which takes this approach of building `libmicropython.a`.

See also #5964.

jonnor, 2024-09-08:
This has been answered above.

closed 2024-09-08 by jonnor