micropython/micropython #8135: ESP32-s2 i2c - writeto with stop=False cause reboot

pavelrevak, 2022-01-03:
On ESP32-S2 and latest build v1.17-231-g0892ebe09 (2021-12-14) from https://micropython.org/download/ESP32_S2_WROVER/

attempt to i2c_write with `stop=False` cause restart:

```Python
import machine
>>> i = machine.I2C(0, scl=machine.Pin(9), sda=machine.Pin(8), freq=100000)
>>> i.scan()
[119]
>>> i.writeto(119, b'\xaa')
1
>>> i.writeto(119, b'\xaa', True)
1
>>> i.writeto(119, b'\xaa', False)
---- RESTARTED ----
```
same issue is with command: `readfrom_mem` where is also used "repeat-start"

pavelrevak, 2022-01-08:
tested also with latest firmware 2022-01-08, but still same issue

dpgeorge, 2022-01-23:
I can reproduce this issue.  My guess is that the IDF I2C API for the S2 requires a STOP condition on all transactions.  If that's the case then we'll need to rework the I2C implementation for this chip.

dpgeorge, 2022-06-01:
It looks like this is actually a bug in the IDF, see issue https://github.com/espressif/esp-idf/issues/8348

And as of IDF 4.4.1 it has been fixed, see https://github.com/espressif/esp-idf/pull/8353

dpgeorge, 2022-06-02:
Also fixed by 6bda80d81147217a1d830b99b93d2e35d372e8f9

closed 2022-06-02 by dpgeorge