micropython/micropython #8137: Raspberry Pi Pico (RP2040) stops working in long sessions

Kyuchumimo, 2022-01-04:
Tested on 2 Raspberry Pi Pico boards with MicroPython firmware v1.17 to verify behavior

https://user-images.githubusercontent.com/74131798/147994118-0dadc26c-b746-4b1e-b447-f5850afca14c.mp4

https://user-images.githubusercontent.com/74131798/147994564-cd9c0672-8e64-4937-8593-9d417975e77f.mp4

peterhinch, 2022-01-04:
It will be hard for anyone to investigate this unless you can provide a way to replicate the problem with a small, complete test case. 

FWIW applications such as [this one](https://github.com/peterhinch/micropython-monitor/blob/master/ANALYSER.md) have run for long periods on the Pico without issue. In my experience it is stable.

Kyuchumimo, 2022-01-04:
@peterhinch Ok, here is the code for you to check for yourself, it is based on the SN76489AN-MicroPython project. Fortunately, you don't need the actual hardware to run the code.
The code requires that you have the following 2 libraries:
https://github.com/Kyuchumimo/SN76489AN-MicroPython
https://github.com/mcauser/micropython-74hc595
You also require a .vgm file found in the SN76489AN-MicroPython repository

```py
from machine import Pin, PWM
import time
import music76489

#####################################
clock = PWM(Pin(10))
clock.freq(3676470)
clock.duty_u16(2**16//2)

music = music76489.Music76489()

music.load_vgm("boss_battle.vgm")
#####################################
#####################################
led = Pin(25)
led.init(led.OUT, value=0)
#####################################
try:
    while True:
        delta = time.ticks_us()
        #####################################
        print("{:02d}:{:02d}".format(time.ticks_ms()//60000,time.ticks_ms()//1000%60), time.ticks_ms())
        #####################################
        music.tick()
        #####################################
        time.sleep_us(16666-(time.ticks_us()-delta))
except:
    led.value(1)
```

Kyuchumimo, 2022-01-05:
Here is another example that you can try as well. In the same way, you do not need to have the hardware to execute the code, although if you want to know what happens,  it is recommended to have the screen or replace the lines fbuf.text() with print().

The following library is required to execute the code: https://github.com/mcauser/micropython-pcd8544

```py
from machine import Pin, SPI
import time
import pcd8544

#####################################
spi = SPI(0,sck=Pin(18),mosi=Pin(19))
spi.init(baudrate=2000000, polarity=0, phase=0)
cs = Pin(21)
dc = Pin(20)
rst = Pin(22)
bl = Pin(17, Pin.OUT, value=1)
lcd = pcd8544.PCD8544(spi, cs, dc, rst)

import framebuf
buffer = bytearray((pcd8544.HEIGHT // 8) * pcd8544.WIDTH)
fbuf = framebuf.FrameBuffer(buffer, pcd8544.WIDTH, pcd8544.HEIGHT, framebuf.MONO_VLSB)

# bg0 - col major msb
bg0 = bytearray(b'\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x80\x40\x40\x40\x40\x40\x80\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x80\x40\x20\x20\x40\x20\x10\x10\x10\x10\x20\xc0\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x80\xc0\xe0\xe0\xe0\xe0\xe0\xc0\x80\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x18\x24\x22\x22\x22\x24\x22\x21\x20\x20\x20\x20\x10\x20\x10\x21\x12\x0c\x00\x00\x00\x00\x00\x00\x00\x00\x00\x01\x02\x02\x02\x02\x02\x02\x01\x02\x01\x02\x01\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x07\x0f\x1f\x1f\x1f\x1f\x1f\x0f\x07\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00')
bg0_fbuf = framebuf.FrameBuffer(bg0, 84, 16, framebuf.MONO_VLSB)

# bg1 - col major msb
bg1 = bytearray(b'\xa0\xc0\xe0\xf0\xe8\xf0\xe8\xd4\xa8\xd0\xe8\xf0\xf8\xfc\xfc\xfe\xfe\xfe\xfe\xfc\xfc\xf8\xf0\xe8\xd4\xe8\xf0\xf0\xf0\xf8\xf4\xe8\xf0\xf8\xfc\xfe\xfe\xfe\xfc\xf8\xf8\xf0\xe0\xf0\xe8\xf0\xe8\xd4\xa8\xd0\xe8\xf0\xf8\xfc\xfc\xfe\xfe\xfe\xfe\xfc\xfc\xf8\xf0\xe8\xd4\xe8\xf0\xf0\xf0\xf8\xf4\xe8\xf0\xf8\xfc\xfe\xfe\xfe\xfc\xf8\xf8\xf0\xe8\xd4')
bg1_fbuf = framebuf.FrameBuffer(bg1, 84, 8, framebuf.MONO_VLSB)

# bg2 - col major msb
bg2 = bytearray(b'\x4a\x4a\x4a\x4a\x4a\x4a\x4a\x4a\x4a\x4a\x4a\x4a\x4a\x4a\x4a\x4a\x4a\x4a\x4a\x4a\x4a\x4a\x4a\x4a\x4a\x4a\x4a\x4a\x4a\x4a\x4a\x4a\x4a\x4a\x4a\x4a\x4a\x4a\x4a\x4a\x4a\x4a\x4a\x4a\x4a\x4a\x4a\x4a\x4a\x4a\x4a\x4a\x4a\x4a\x4a\x4a\x4a\x4a\x4a\x4a\x4a\x4a\x4a\x4a\x4a\x4a\x4a\x4a\x4a\x4a\x4a\x4a\x4a\x4a\x4a\x4a\x4a\x4a\x4a\x4a\x4a\x4a\x4a\x4a')
bg2_fbuf = framebuf.FrameBuffer(bg2, 84, 8, framebuf.MONO_VLSB)

# bg3 - col major msb
bg3 = bytearray(b'\x00\xf0\xf0\xfc\xde\xbe\x1e\xfe\x7c\xe0\xe0\xcc\x1e\xee\x9c\x0c\x00\x00\xf0\xf0\xfc\xde\xbe\x1e\xfe\x7c\xe0\xe0\xcc\x1e\xee\x9c\x0c\x00\xf0\xf0\xfc\xde\xbe\x1e\xfe\x7c\xe0\xe0\xc0\x0c\x1e\xee\x9c\x0c\x00\xf0\xf0\xfc\xde\xbe\x1e\xfe\x7c\xe0\xe0\xcc\x1e\xee\x9c\x0c\x00\x00\xf0\xf0\xfc\xde\xbe\x1e\xfe\x7c\xe0\xe0\xc0\x0c\x1e\xee\x9c\x0c\x10\x08\x11\x41\x43\x63\x7f\x7c\x63\x61\x01\x08\x04\x28\x10\x20\x08\x04\x08\x01\x41\x43\x63\x7f\x7c\x63\x61\x11\x08\x10\x20\x10\x30\x08\x10\x01\x41\x43\x63\x7f\x7c\x63\x61\x01\x10\x08\x30\x10\x20\x00\x10\x08\x11\x41\x43\x63\x7f\x7c\x63\x71\x09\x10\x20\x10\x20\x00\x10\x08\x10\x01\x41\x43\x63\x7f\x7c\x63\x61\x11\x08\x10\x20\x10\x20\x00')
bg3_fbuf = framebuf.FrameBuffer(bg3, 84, 16, framebuf.MONO_VLSB)

#####################################
#####################################
led = Pin(25)
led.init(led.OUT, value=0)
#####################################
#####################################
while True:
    delta = time.ticks_us()
    #####################################
    fbuf.fill(0)
    #BG0
    fbuf.blit(bg0_fbuf, 0-((time.ticks_us()//16666)//3%84), 0, 0)
    fbuf.blit(bg0_fbuf, 84-((time.ticks_us()//16666)//3%84), 0, 0)
    #BG1
    fbuf.blit(bg1_fbuf, 0-((time.ticks_us()//16666)//2%84), 16, 0)
    fbuf.blit(bg1_fbuf, 84-((time.ticks_us()//16666)//2%84), 16, 0)
    #BG2
    fbuf.blit(bg2_fbuf, 0, 24, 0)
    #BG3
    fbuf.blit(bg3_fbuf, 0-((time.ticks_us()//16666)%84), 32, 0)
    fbuf.blit(bg3_fbuf, 84-(time.ticks_us()//16666)%84, 32, 0)

    fbuf.text("{:02d}:{:02d}".format(time.ticks_ms()//60000,time.ticks_ms()//1000%60), 0, 0, 1)
    fbuf.text(str(time.ticks_ms()),0 , 8, 1)
    lcd.data(buffer)
    #####################################
    time.sleep_us(16666-(time.ticks_us()-delta))
```

peterhinch, 2022-01-06:
This is incorrect because it doesn't allow for rollover:
```python
time.sleep_us(16666-(time.ticks_us()-delta))
```
This applies in other places in your code too: it is invalid to use + or - operators on ticks values. Please read [this doc](https://docs.micropython.org/en/latest/library/time.html#time.ticks_diff).

Aside from rollover you may also need to consider the case where the time difference is negative.

Kyuchumimo, 2022-01-24:
time.ticks_diff() fixes hangups on long sessions. I will close the issue.

closed 2022-01-24 by Kyuchumimo