micropython/micropython #8139: Transmission errors things while transfering bytearray with pyboard.py

labodezao, 2022-01-04:
While transferring  a bytearray test, I exeperiement that the length of this one is not constant and the bytearray is not the same during the transfer beetween microython and Cpython on windows with usb cable :

I have a Cpython program (the same than the beginings) with grab data from micropython serial :

On the micropython side a simple function returning basically a bytearray ( length 2424 when I ask micropython prompt)

```
 def Grab_Full_Mes(self) -> None:
        """Mesure de la pression température et remplissage d'un array de valeurs"""
         return bytearray(b'\xb5\xfe\x967\xca\xfc#<\x03\xec\xa3<\xba\xdb\xf5<\xc5\xe6#=\x94\xddL=J\xd2u=]k\x8f=<\xdb\xa3=\x07_\xb8=\x87\xdc\xcc=\xc7I\xe1=\x16\xc3\xf5=\xb8\x1e\x05>Id\x0f>c\x9a\x19>f\xd9#>\x87\x15.>CY8>\xdd\x94B>)\xcfL>c\rW>\xb6Ma>\xb1\x86k>`\xc9u>%\x03\x80>\xd1 \x85>:>\x8a>N_\x8f>=}\x94>\xf6\x9b\x99>\xcf\xba\x9e>f\xd9\xa3>\x12\xf7\xa8>\xbe\x14\xae>\xbd6\xb3>\xdaU\xb8>zr\xbd>>\x92\xc2>\x9e\xb1\xc7>)\xcf\xcc>\xd5\xec\xd1>\x81\n\xd7>\x80,\xdc>ZK\xe1>\xcfi\xe6>\xa9\x88\xeb>\xfc\xa6\xf0>;\xc6\xf5>\x15\xe5\xfa>e\x00\x00?\xc1\x8f\x02?: \x05?y\xae\x07?q=\n?\xa3\xce\x0c?\x9a]\x0f?\x81\xec\x11?F{\x14?=\n\x17?1\x9a\x19?\x8d)\x1c?\x17\xba\x1e?0I!?k\xd8#?Qg&?oJ)?\x83\x85+?\x87\x15.?\xa0\xa40?U33?e\xc45?LS8?C\xe2:?\x9fq=?\xd6\x01@?S\x91B?m E?S\xafG?\x9f>J?\xde\xccL?\x00qO?)\xedQ?\xab{T?\x91\nW?\xd8\x9aY?\xd0)\\?\xa6\xb8^?\x84Ia?\x8c\xd8c?\xcbff?\xd3\xf5h?\x7f\x86k?f\x15n?\xf9\xa3p?\xf84s?\xd2\xc2u?\xe3Sx?\xce\xe1z?\xb5p}?\x00\x00\x00\x00/\xd3\x7fG/\xd3\x7fG/\xd3\x7fG/\xd3\x7fG/\xd3\x7fG/\xd3\x7fG/\xd3\x7fG/\xd3\x7fG/\xd3\x7fG/\xd3\x7fG/\xd3\x7fG/\xd3\x7fG/\xd3\x7fG/\xd3\x7fG/\xd3\x7fG/\xd3\x7fG/\xd3\x7fG/\xd3\x7fG/\xd3\x7fG/\xd3\x7fG/\xd3\x7fG/\xd3\x7fG/\xd3\x7fG/\xd3\x7fG/\xd3\x7fG/\xd3\x7fG/\xd3\x7fG/\xd3\x7fG/\xd3\x7fG/\xd3\x7fG/\xd3\x7fG/\xd3\x7fG/\xd3\x7fG/\xd3\x7fG/\xd3\x7fG/\xd3\x7fG/\xd3\x7fG/\xd3\x7fG/\xd3\x7fG/\xd3\x7fG/\xd3\x7fG/\xd3\x7fG/\xd3\x7fG/\xd3\x7fG/\xd3\x7fG/\xd3\x7fG/\xd3\x7fG/\xd3\x7fG/\xd3\x7fG/\xd3\x7fG/\xd3\x7fG/\xd3\x7fG/\xd3\x7fG/\xd3\x7fG/\xd3\x7fG/\xd3\x7fG/\xd3\x7fG/\xd3\x7fG/\xd3\x7fG/\xd3\x7fG/\xd3\x7fG/\xd3\x7fG/\xd3\x7fG/\xd3\x7fG/\xd3\x7fG/\xd3\x7fG/\xd3\x7fG/\xd3\x7fG/\xd3\x7fG/\xd3\x7fG/\xd3\x7fG/\xd3\x7fG/\xd3\x7fG/\xd3\x7fG/\xd3\x7fG/\xd3\x7fG/\xd3\x7fG/\xd3\x7fG/\xd3\x7fG/\xd3\x7fG/\xd3\x7fG/\xd3\x7fG/\xd3\x7fG/\xd3\x7fG/\xd3\x7fG/\xd3\x7fG/\xd3\x7fG/\xd3\x7fG/\xd3\x7fG/\xd3\x7fG/\xd3\x7fG/\xd3\x7fG/\xd3\x7fG/\xd3\x7fG/\xd3\x7fG/\xd3\x7fG/\xd3\x7fG/\xd3\x7fG/\xd3\x7fG/\xd3\x7fG\x00\x00\x00\x00\xb5\xfe\x967\xca\xfc#<\x03\xec\xa3<\xba\xdb\xf5<\xc5\xe6#=\x94\xddL=J\xd2u=]k\x8f=<\xdb\xa3=\x07_\xb8=\x87\xdc\xcc=\xc7I\xe1=\x16\xc3\xf5=\xb8\x1e\x05>Id\x0f>c\x9a\x19>f\xd9#>\x87\x15.>CY8>\xdd\x94B>)\xcfL>c\rW>\xb6Ma>\xb1\x86k>`\xc9u>%\x03\x80>\xd1 \x85>:>\x8a>N_\x8f>=}\x94>\xf6\x9b\x99>\xcf\xba\x9e>f\xd9\xa3>\x12\xf7\xa8>\xbe\x14\xae>\xbd6\xb3>\xdaU\xb8>zr\xbd>>\x92\xc2>\x9e\xb1\xc7>)\xcf\xcc>\xd5\xec\xd1>\x81\n\xd7>\x80,\xdc>ZK\xe1>\xcfi\xe6>\xa9\x88\xeb>\xfc\xa6\xf0>;\xc6\xf5>\x15\xe5\xfa>e\x00\x00?\xc1\x8f\x02?: \x05?y\xae\x07?q=\n?\xa3\xce\x0c?\x9a]\x0f?\x81\xec\x11?F{\x14?=\n\x17?1\x9a\x19?\x8d)\x1c?\x17\xba\x1e?0I!?k\xd8#?Qg&?oJ)?\x83\x85+?\x87\x15.?\xa0\xa40?U33?e\xc45?LS8?C\xe2:?\x9fq=?\xd6\x01@?S\x91B?m E?S\xafG?\x9f>J?\xde\xccL?\x00qO?)\xedQ?\xab{T?\x91\nW?\xd8\x9aY?\xd0)\\?\xa6\xb8^?\x84Ia?\x8c\xd8c?\xcbff?\xd3\xf5h?\x7f\x86k?f\x15n?\xf9\xa3p?\xf84s?\xd2\xc2u?\xe3Sx?\xce\xe1z?\xb5p}?\x00\x00\x00\x00\xd7\xa3\xc2A\xd7\xa3\xc2A\xd7\xa3\xc2A\xd7\xa3\xc2A\xd7\xa3\xc2A\xd7\xa3\xc2A\xd7\xa3\xc2A\xd7\xa3\xc2A\xd7\xa3\xc2A\xd7\xa3\xc2A\xd7\xa3\xc2A\xd7\xa3\xc2A\xd7\xa3\xc2A\xd7\xa3\xc2A\xd7\xa3\xc2A\xd7\xa3\xc2A\xd7\xa3\xc2A\xd7\xa3\xc2A\xd7\xa3\xc2A\xd7\xa3\xc2A\xd7\xa3\xc2A\xd7\xa3\xc2A\xd7\xa3\xc2A\xd7\xa3\xc2A\xd7\xa3\xc2A\xd7\xa3\xc2A\xd7\xa3\xc2A\xd7\xa3\xc2A\xd7\xa3\xc2A\xd7\xa3\xc2A\xd7\xa3\xc2A\xd7\xa3\xc2A\xd7\xa3\xc2A\xd7\xa3\xc2A\xd7\xa3\xc2A\xd7\xa3\xc2A\xd7\xa3\xc2A\xd7\xa3\xc2A\xd7\xa3\xc2A\xd7\xa3\xc2A\xd7\xa3\xc2A\xd7\xa3\xc2A\xd7\xa3\xc2A\xd7\xa3\xc2A\xd7\xa3\xc2A\xd7\xa3\xc2A\xd7\xa3\xc2A\xd7\xa3\xc2A\xd7\xa3\xc2A\xd7\xa3\xc2A\xd7\xa3\xc2A\xd7\xa3\xc2A\xd7\xa3\xc2A\xd7\xa3\xc2A\xd7\xa3\xc2A\xd7\xa3\xc2A\xd7\xa3\xc2A\xd7\xa3\xc2A\xd7\xa3\xc2A\xd7\xa3\xc2A\xd7\xa3\xc2A\xd7\xa3\xc2A\xd7\xa3\xc2A\xd7\xa3\xc2A\xd7\xa3\xc2A\xd7\xa3\xc2A\xd7\xa3\xc2A\xd7\xa3\xc2A\xd7\xa3\xc2A\xd7\xa3\xc2A\xd7\xa3\xc2A\xd7\xa3\xc2A\xd7\xa3\xc2A\xd7\xa3\xc2A\xd7\xa3\xc2A\xd7\xa3\xc2A\xd7\xa3\xc2A\xd7\xa3\xc2A\xd7\xa3\xc2A\xd7\xa3\xc2A\xd7\xa3\xc2A\xd7\xa3\xc2A\xd7\xa3\xc2A\xd7\xa3\xc2A\xd7\xa3\xc2A\xd7\xa3\xc2A\xd7\xa3\xc2A\xd7\xa3\xc2A\xd7\xa3\xc2A\xd7\xa3\xc2A\xd7\xa3\xc2A\xd7\xa3\xc2A\xd7\xa3\xc2A\xd7\xa3\xc2A\xd7\xa3\xc2A\xd7\xa3\xc2A\xd7\xa3\xc2A\xd7\xa3\xc2A\xd7\xa3\xc2A\xd7\xa3\xc2A\x00\x00\x00\x00\xb5\xfe\x967\xca\xfc#<\x03\xec\xa3<\xba\xdb\xf5<\xc5\xe6#=\x94\xddL=J\xd2u=]k\x8f=<\xdb\xa3=\x07_\xb8=\x87\xdc\xcc=\xc7I\xe1=\x16\xc3\xf5=\xb8\x1e\x05>Id\x0f>c\x9a\x19>f\xd9#>\x87\x15.>CY8>\xdd\x94B>)\xcfL>c\rW>\xb6Ma>\xb1\x86k>`\xc9u>%\x03\x80>\xd1 \x85>:>\x8a>N_\x8f>=}\x94>\xf6\x9b\x99>\xcf\xba\x9e>f\xd9\xa3>\x12\xf7\xa8>\xbe\x14\xae>\xbd6\xb3>\xdaU\xb8>zr\xbd>>\x92\xc2>\x9e\xb1\xc7>)\xcf\xcc>\xd5\xec\xd1>\x81\n\xd7>\x80,\xdc>ZK\xe1>\xcfi\xe6>\xa9\x88\xeb>\xfc\xa6\xf0>;\xc6\xf5>\x15\xe5\xfa>e\x00\x00?\xc1\x8f\x02?: \x05?y\xae\x07?q=\n?\xa3\xce\x0c?\x9a]\x0f?\x81\xec\x11?F{\x14?=\n\x17?1\x9a\x19?\x8d)\x1c?\x17\xba\x1e?0I!?k\xd8#?Qg&?oJ)?\x83\x85+?\x87\x15.?\xa0\xa40?U33?e\xc45?LS8?C\xe2:?\x9fq=?\xd6\x01@?S\x91B?m E?S\xafG?\x9f>J?\xde\xccL?\x00qO?)\xedQ?\xab{T?\x91\nW?\xd8\x9aY?\xd0)\\?\xa6\xb8^?\x84Ia?\x8c\xd8c?\xcbff?\xd3\xf5h?\x7f\x86k?f\x15n?\xf9\xa3p?\xf84s?\xd2\xc2u?\xe3Sx?\xce\xe1z?\xb5p}?\x00\x00\x00\x00\x00\x00\x00\x00\xa1\x0e\xea<\xa1\x0ej=\xa1\x0ej=\xa1\x0e\xea=\xa1\x0ej=\xf9\x8a\xaf=\xa1\x0e\xea<\xa1\x0e\xea=\xa1\x0e\xea<\x00\x00\x00\x00\x00\x00\x00\x00\xa1\x0e\xea<\xa1\x0e\xea\xbc\xa1\x0e\xea<\xf9\x8a\xaf=\xa1\x0e\xea<\xf9\x8a\xaf=\xa1\x0e\xea<\xf9\x8a\xaf=\xa1\x0e\xea<\xa1\x0ej=\xf9\x8a\xaf=\xa1\x0ej=\xa1\x0e\xea<\xa1\x0ej=\xa1\x0ej=\xa1\x0ej=\xa1\x0e\xea<\xa1\x0e\xea\xbc\x00\x00\x00\x00\xa1\x0ej=\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\xa1\x0ej=\xa1\x0e\xea\xbc\xa1\x0e\xea<\xa1\x0ej=\xa1\x0e\xea=\x00\x00\x00\x00\xa1\x0ej\xbd\x00\x00\x00\x00\xa1\x0e\xea\xbc\xa1\x0e\xea<\xa1\x0e\xea<\xa1\x0e\xea<\x00\x00\x00\x00\xa1\x0e\xea<\xa1\x0e\xea<\xa1\x0ej=\xf9\x8a\xaf=\xa1\x0e\xea\xbc\x00\x00\x00\x00\xa1\x0ej=\x00\x00\x00\x00\xa1\x0e\xea<\x00\x00\x00\x00\xa1\x0ej=\xa1\x0ej=\xa1\x0e\xea<\xa1\x0e\xea<\xa1\x0e\xea<\xa1\x0e\xea<\x00\x00\x00\x00\xf9\x8a\xaf=\xf9\x8a\xaf=\xa1\x0e\xea\xbc\xa1\x0e\xea<\xa1\x0e\xea\xbc\xa1\x0e\xea<\xa1\x0e\xea<\xa1\x0e\xea<\xa1\x0e\xea<\xa1\x0ej=%I\x12>\xa1\x0e\xea\xbc\xa1\x0ej=\xa1\x0e\xea\xbc%I\x12>\xf9\x8a\xaf=\xa1\x0e\xea<\xa1\x0e\xea<\x00\x00\x00\x00\xa1\x0e\xea\xbc\xf9\x8a\xaf=\xf9\x8a\xaf=\xf9\x8a\xaf=\xf9\x8a\xaf=\xa1\x0ej=\xa1\x0ej=\xa1\x0ej=\xa1\x0ej=\xa1\x0e\xea<\xa1\x0ej=\xa1\x0e\xea\xbc\xa1\x0e\xea\xbc\xa1\x0ej=\xa1\x0e\xea<\x00\x00\x00\x00')

```

In Cpython side a function to grab the byte array :

```
pyb = pyboard.Pyboard('COM4')
pyb.enter_raw_repl()

print(pyb.eval("len(mes.Grab_Full_Mes())")) # to looks on the array length before transfer : results b'2424'
Mes_bytes = pyb.eval("print(Grab_Full_Mes())") # grab the data though rshell pyboard.py file
Mes_bytestes_2 = pyb.exec("Grab_Full_Mes()")  # test if that wors better with eval ....  but not ! 
print("length bytes:" ,len(Mes_bytes)) # check the leng of the reicived datas .... and here a big problem : length bytes: 6544
Mes_Press =np.frombuffer(Mes_bytes, dtype=np.float32) # works if I copy the original bytearray and not working after transfer ....
Mes_Press.reshape(6,int(Mes_Press.size/6))
print(Mes_Press)

```

So I don't get the same bytearray length after transfer, and the length seems fluctating ...

I hope you could help me, I spend all the day on this ...

The original bytearray : (with the right length : 2424 )
```
`bytearray(b'\xb5\xfe\x967\xca\xfc#<\x03\xec\xa3<\xba\xdb\xf5<\xc5\xe6#=\x94\xddL=J\xd2u=]k\x8f=<\xdb\xa3=\x07_\xb8=\x87\xdc\xcc=\xc7I\xe1=\x16\xc3\xf5=\xb8\x1e\x05>Id\x0f>c\x9a\x19>f\xd9#>\x87\x15.>CY8>\xdd\x94B>)\xcfL>c\rW>\xb6Ma>\xb1\x86k>`\xc9u>%\x03\x80>\xd1 \x85>:>\x8a>N_\x8f>=}\x94>\xf6\x9b\x99>\xcf\xba\x9e>f\xd9\xa3>\x12\xf7\xa8>\xbe\x14\xae>\xbd6\xb3>\xdaU\xb8>zr\xbd>>\x92\xc2>\x9e\xb1\xc7>)\xcf\xcc>\xd5\xec\xd1>\x81\n\xd7>\x80,\xdc>ZK\xe1>\xcfi\xe6>\xa9\x88\xeb>\xfc\xa6\xf0>;\xc6\xf5>\x15\xe5\xfa>e\x00\x00?\xc1\x8f\x0 [... cut ...] ea<\\xf9\\x8a\\xaf=\\xa1\\x0e\\xea\\xbc\\xa1\\x0ej=\\xa1\\x0ej=\\xa1\\x0e\\xea\\xbc\\x00\\x00\\x00\\x00\\xa1\\x0e\\xea<\\xa1\\x0ej=\\xa1\\x0e\\xea<\\xa1\\x0ej=\\x00\\x00\\x00\\x00\\xf9\\x8a\\xaf=\\xa1\\x0ej=\\xa1\\x0ej=\\xa1\\x0ej=\\xa1\\x0e\\xea<\\xa1\\x0e\\xea\\xbc\\x00\\x00\\x00\\x00\\xf9\\x8a/>\\x00\\x00\\x00\\x00\\xa1\\x0e\\xea<\\xa1\\x0ej\\xbd\\xa1\\x0ej\\xbd\\x00\\x00\\x00\\x00\\x00\\x00\\x00\\x00\\xa1\\x0e\\xea<\\xf9\\x8a\\xaf=\\xa1\\x0e\\xea<\\xa1\\x0e\\xea<\\x00\\x00\\x00\\x00\')\r\n
`

dpgeorge, 2022-01-06:
You need to evaluate the returned data, like this (on the PC):
```python
mes_expr = pyb.exec("print(Grab_Full_Mes(0))")
mes = eval(str(mes_expr, "ascii"))
print(len(mes))
```

You may also be having issues with flow control and/or a broken Windows USB serial driver.  In that case you may need to break up the data into smaller chunks and send them separately.

labodezao, 2022-01-06:
Hi, Thanks a lot for your answear ! 
Oh my god, this is working :) 
👍 
❤️

closed 2022-01-06 by labodezao