micropython/micropython #8143: Build broken?

snhirsch, 2022-01-06:
I'm unable to build on a Ubuntu 18.04 system. Following the instructions in the SDK docs (and some alternatives from forums), I still end up here:

```/net/src/platforms/raspberry_pi/rp2040/micropython/ports/rp2$ make
[ -d build-PICO ] || cmake -S . -B build-PICO -DPICO_BUILD_DOCS=0 -DMICROPY_BOARD=PICO
CMake Error: The source directory "/net/src/platforms/raspberry_pi/rp2040/micropython/ports/rp2/build-PICO" does not exist.```

If I create the missing directory it dies with an error about missing target.

snhirsch, 2022-01-06:
More information:  Acting on a hunch, I tried building in /tmp on the local filesystem (rather than a fairly deep path in an NFS mounted directory).  Lo and behold, it worked properly!  I suspect you are running out of environment space somewhere when propagating long file paths.  Please let me know if you need any more information to debug this?

dpgeorge, 2022-01-07:
I think you just need to remove `build-PICO` completely and retry.

snhirsch, 2022-01-07:
Then why is it sensitive to the build directory path?  I used the same script to set it up in both cases with pull coming fresh from GitHub.  In the NFS directory it fails.  In /tmp it works.

dpgeorge, 2022-01-07:
Did you try building with a completely fresh clone of this repo?  If so then it could be an issue with the NFS mount, eg the directory is created but when it's stat'd it doesn't yet show up.

Can you try cloning into the top level of the NFS volume?  Or at least only 1 directory deep?

snhirsch, 2022-01-07:
And, of course, it's now working fine in the deep directory...  If I tried this once I must have tried it a half-dozen times with no prior success.  My apologies for the noise - I guess we can close this issue.

closed 2022-01-07 by snhirsch