micropython/micropython #8144: RFC: overriding mpconfig(port).h macros

stinos, 2022-01-06:
Introduction of variants came along with wrapping a bunch of definitions in mpconfigport.h with `#ifndef MICROPY_SOME_OPTION` because otherwise these couldn't be defined in mpconfigvariant.h anymore. Likewise this also means that a variant cannot define anything which is already defined in mpconfigport.h and is not yet wrapped. Which is not ideal.

A simple solution is allowing to include another file *after* everything in mpconfigport.h (or even mpconfgh.h). In such file users can then `#undef` and `#define` again at will.

So how about adding

```
#ifdef MP_POSTCONFIGFILE
#include MP_POSTCONFIGFILE
#endif
```

at the end of mpconfig.h ? (name is chosen for consistency with MP_CONFIGFILE at the top though that doesn't follow naming conventions).

Alternatively this could go at the end of each mpconfigport.h but I'm not sure that is needed.

stinos, 2022-01-13:
@dpgeorge ping :) I've given it some more thought and it seems the next logical step to making variants more useful.

stinos, 2022-01-14:
> Going this route, I think it's best to put that near the top of py/mpconfig.h

Ok, that should be safer.

> It's not clear which should go where 

Basic strategy would be to put everything in mpconfigvariant.h, and if the compiler complains about duplicate symbols you know they have to be moved. I admit that doesn't sound super nice and clean, but my reasoning is also that people who know how variants work are also likely to know the implications of using 2 files.

The only other option I see is indeed wrapping pretty much everything in each mpconfigport.h in `#ifdef` just like mpconfig.h. But that's a huged fiee, and should really be done consistently otherwise it's no use.

stinos, 2022-05-02:
Kinda bumped into this again with https://github.com/micropython/micropython/commit/3c20ddb41adc177442b68a8fdff221c80e0b0b5e: I'm typically building the unix port with `CFLAGS_EXTRA=-DMICROPY_MODULE_BUILTIN_INIT=1` but that leads to multiple definitions now.

I can work around by making my own variant (but then it might be required in some cases to keep it up to date with the standard one, which is now harder because mpconfigvariant went from like 2 lines to 20 lines), or possibly create my own variant which includes unix/standard/mpconfigvariant.h then removes what is not needed, or create a PR to wrap just that one option I personally need which seems a tad silly, or a PR to just wrap everything for unix/windows ports and be done with it?

dpgeorge, 2022-05-03:
> Kinda bumped into this again with [3c20ddb](https://github.com/micropython/micropython/commit/3c20ddb41adc177442b68a8fdff221c80e0b0b5e): I'm typically building the unix port with `CFLAGS_EXTRA=-DMICROPY_MODULE_BUILTIN_INIT=1` but that leads to multiple definitions now.

I think this is slightly different from the original issue/request/discussion: now you want to be able to override things defined by the variant, right?  And in this case adding #ifndef... around everything in `mpconfigport.h` won't help, you need them in `mpconfigvariant.h`

In this case I think you just need to define your own variant.

> possibly create my own variant which includes unix/standard/mpconfigvariant.h then removes what is not needed

Yes that is a good way to do it.  That's how `stm32/boards/PYBD_SF6` works (based on `PYBD_SF2`).

> or create a PR to wrap just that one option I personally need

Do you mean to wrap the option in `standard/mpconfigvariant.h`?

-----

Ultimately we want to clean up the mpconfig*.h's even more so they are as small as possible, and everything is set by a feature level or just left as the default.  Then the remaining options can all have #ifndef's around them.

But I'm not sure it makes sense to also #ifndef the variant's config files?

stinos, 2022-05-03:
> I think this is slightly different from the original issue/request/discussion: now you want to be able to override things defined by the variant, right?

Yes.

> Ultimately we want to clean up the mpconfig*.h's even more so they are as small as possible,

Sounds ok.

> But I'm not sure it makes sense to also #ifndef the variant's config files?

Normally I'd say no, but for the default variant it's perhaps different?

It would be convenient if you can just type `make CFLAGS_EXTRA='-DMICROPY_...=1'` to enable/disable any feature at will, instead of having to figure out for which features this works, and then having to either modify the define in the code itself and/or writing your own variant. At least I often use it like that and it's also the easiest way to instruct others on the forum etc. Happens quite often there are questions like 'feature X is missing' and just enabling it at build time on the command line is far easier to explain and test than the other options.

dpgeorge, 2022-05-03:
> It would be convenient if you can just type `make CFLAGS_EXTRA='-DMICROPY_...=1'` to enable/disable any feature at will, instead of having to figure out for which features this works, and then [...]

I tend to agree with this, at least for the unix port.  Well, maybe it even makes sense for bare-metal boards, so users can easily rebuild them with some options enabled.  But it's going to be messy to put #ifndef's around every option in all `mpconfigboard.h` files....

jonnor, 2024-09-08:
Is this issue still relevant? If so, what is the proposed way forward?

stinos, 2024-09-09:
It is still relevant imo, because it's unavoidable for micropython that at one point someone runs on a specific port/board and would like to use this one particular feature which isn't enabled for that port/board. But not all features, because of code size concerns.

To summarize the issue is that if you want to change one of the many options (the PP macros in mpconfig.h and similar) micropython has in a simple way (for example being able to to simply drop a `#define MP_FEATURE_X (1)` in your own variant if you have one already or with `make CFLAGS_EXTRA=-DMP_FEATURE_X=1`), depending on the feature might not be possible to do that without either modifying code in the micropython repository itself or by being forced to define your own variant i.e. write code to deal with it. Both options are also not ideal for beginners.

The problem is that all of the fixes each come with their own disadvantages and as such it's hard to come up with a final conclusion :)

So I'm not sure what the policy should be for issues like this: it's not fixed, but the options to fix have been discussed in enough detail I think, so it remains inconclusive. Should the be Open or Closed then?