micropython/micropython #8147: Thingy:52 nrf port, what board name to use to compile port

alexisc67, 2022-01-07:
Hi,

I created this topic in the forum with no much luck, maybe someone here can help me.
https://forum.micropython.org/viewtopic.php?f=12&t=11704

I am compiling a port for the Nordic Seminconductors Thingy:52. I see in the documentation that nrf port has been tested for Thingy:52. It is listed in tested hardware. However there is no target board code for the Thingy:52 specifically.
https://github.com/micropython/micropyt ... /ports/nrf

I would like to know how this board what tested and if another target board name was used for it, for example pca10040. The Thingy:52 is pca20020.

Thanks,

dpgeorge, 2022-01-23:
I don't think there was ever a board definition for Thingy:52.  And I don't know who tested it, maybe @tralamazza?

tralamazza, 2022-01-24:
Nope not me, but maybe @glennrub

glennrub, 2022-01-25:
Yeah, i did test _Thingy:52_ on the @tralamazza fork. However this was even before "nrf" was moved to "ports/nrf". I still have the branch on my old laptop, so I should be able to bring it up again. If i recall correctly, i did not bring it upstream due to lack of on-board debugger and lack of USB-UART. The USB connector is only there to provide power, not UART. Hence, there is a need for the board to use Bluetooth REPL/FTDI chip or similar connected on the free pins inside to get something interactive. I'll set up a branch aligned with the new folder layout.

glennrub, 2022-01-25:
> I am compiling a port for the Nordic Seminconductors Thingy:52.

@alexisc67, i've pushed a branch with the board definition i used, with a few mods to get it up to date ([glennrub - thingy52](https://github.com/glennrub/micropython/tree/thingy52)). In general it is lacking quite a lot of python scripts to act as a full featured thingy52, but could be used as a base. Did you get any further on your attempt to support the thingy52?

alexisc67, 2022-01-28:
Hi, I did not do more since my last message. I will have a look at the branch you pushed.

jonnor, 2024-09-08:
Anyone willing to provide a MR with tested board definitions for Thingy52?