micropython/micropython #8148: esp32 s3 deepsleep wake

stephanelsmith, 2022-01-07:
Migrating to S3, one curious issue is I'm not able to wake from deep sleep.  Timer wake nor the ext0/ext1 modes.  I have no problem with D4 or S2 versions.  I'm using the UART console on S2/S3, so I do not get disconnected on deepsleep.

import machine
machine.deepsleep(1000)

Purusing modmachine.c where looks like the important bits are, I don't see anything jumping out at me in 'machine_sleep_helper' that would make a difference on a per sku basis.

I'm on micropython master branch.  What esp-idf branch are S3 folks using?  I'm on a October commit b86fe0c66ca17d5907ae26ecdffc9d4eb79cec07 which is working for me.  Master esp-idf is giving my compilation errors.  Maybe a more recent version?

stephanelsmith, 2022-01-23:
Not sure if I'm the only one seeing this or no one else has S3 yet (my boards have the sample silicon.)  

Moving to S2 until S3 is more available.

UnexpectedMaker, 2022-01-29:
Working fine here on production S3 silicon. Made a video about ti actually a few days ago showing deep sleep being used in MP and measuring the current.
https://youtu.be/bIm1ofpL6FY

Which "sample silicon" do you have? Marlin?

stephanelsmith, 2022-01-29:
I'm not sure which sample silicon, I will ask my esp rep.  I received 5 samples of ESP32-S3FN8, 5 ESP32-S3, and 1 devkit, but no idea which version.  Very valid question that I need to go back and figure out.

Thank you for clues on the build.  I will try again.  I am relieved you got it.  I'll revisit.

I'll keep an eye on your store.  I have a bin full of tinypicos/tinys2/feathers2.  Looking forward to adding tinys3 and pros3 to the mix.  Use these all the time!  Very versatile boards!

stephanelsmith, 2022-02-01:
I am confirming on v4.4-rc1, deepsleep is working correctly on s3.  Thank you for the pointers.

closed 2022-02-01 by stephanelsmith