micropython/micropython #8157: Feature - call a file when REPL is logged into

pacmac, 2022-01-11:
Would it be possible to add a feature whereby a file can be called (login.py ??) when a user initially accesses the REPL ?

At the moment, I have to login, and then perform a soft reboot to get the boot.py / main.py files triggered.

It would be good if this could be triggered automagically upon login :-)

pacmac, 2022-01-16:
When you login to the REPL you always get the prompt header:
```
MicroPython v1.17 on 2021-12-30; Raspberry Pi Pico with RP2040
Type "help()" for more information.
```

Is it possible to use this hook to trigger a file similar to what happens in linux using ```.profile``` or ```.bashrc```

I have developed a console app and it would be nice if each time a use logs in, I could execute a script

davehylands, 2022-01-16:
There is no concept of logging in on Micropython. boot.py and main.py are the scripts that run when the microcontroller boots up (or is soft-reset).

If by logging in you mean connect on the serial port - there is no general way for micropython to dectect that a user has connected via the serial port.

pacmac, 2022-01-16:
Hmm, then howcome this gets printed each time you login:

```
MicroPython v1.17 on 2021-12-30; Raspberry Pi Pico with RP2040
Type "help()" for more information.
```

Carglglz, 2022-01-16:
> Hmm, then howcome this gets printed each time you login / access the REPL:
> 
> ```
> MicroPython v1.17 on 2021-12-30; Raspberry Pi Pico with RP2040
> Type "help()" for more information.
> ```

This is the client sending a control character ``0x02`` at the beginning, this is also sent by `CTRL-B` (see [character-sets](https://www.ibm.com/docs/en/xl-fortran-aix/16.1.0?topic=appendix-ascii-ebcdic-character-sets))

>At the moment, I have to login, and then perform a soft reboot to get the boot.py / main.py files triggered.
>It would be good if this could be triggered automagically upon login :-)

The soft reboot is also triggered by a control character ``0x04`` (`CTRL-D`), and I believe some terminal clients have an option to choose which control characters are sent at the beginning. See ``--initstring`` in [picocom](https://www.mankier.com/1/picocom)

This for example does a soft reset:

```
$ picocom /dev/tty.usbserial-016418E3 -b 115200 -t "$(echo -ne '\x04\r')"
```
I don't know if this is what you were looking for, but I hope it helps 👍🏼

pacmac, 2022-01-17:
@Carglglz - thanks, do you know if it is possible to use screen and pass a command to it, I tried this but no luck:

```
screen -m "machine.soft_reset()" $DEVICE
```

machine is already imported with boot.py

Carglglz, 2022-01-17:
Check this [How to create a screen executing given command?](https://stackoverflow.com/questions/7049252/how-to-create-a-screen-executing-given-command)

I've just tested that this works on a detached screen
```
screen -r -X stuff "$(echo -ne '\x04\r')"
```
But I'm not sure about create a screen and send a command...

You could always try a python script with serial lib, to login-reset and then enter screen. 🤔
Something as simple as:
```$ python reset_device.py && screen $ DEVICE```
Then maybe make an alias in `.profile` or even a shell function...

pacmac, 2022-01-17:
Thanks, after posting the previous, I also discovered the ```-X stuff``` method, but like you was unable to send the command to an attached screen.

So I downloaded and installed (osx brew) picocom and that works well.

On a similar not, do you know if micropython can send ```^A^Q``` to the console to quit it ??

I tried ```sys.stdout.write('\x11');sys.stdout.write('\x01')``` but that just prints '1 1' to the console.

Carglglz, 2022-01-17:
> On a similar not, do you know if micropython can send ^A^Q to the console to quit it ??
>
> I tried sys.stdout.write('\x11');sys.stdout.write('\x01') but that just prints '1 1' to the console.

No I don't think it is possible to do that, maybe in HID mode, but I'm not sure ... 🤷🏼

robert-hh, 2022-01-17:
`a=sys.stdout.write(b"\x01\x11")` actually writes Ctrl-A Ctrl-Q to the stdout, but that does not get to the input channel of your console app.