micropython/micropython #8163: ports/esp8266/README.md could be much clearer about toolchains

ChrisMacGregor, 2022-01-12:
Currently, the MicroPython ESP8266 README refers to https://github.com/pfalcon/esp-open-sdk, which has been effectively unmaintained for several years, and which now needs several patches in order to build on recent Linux distro versions (such as Ubuntu 21.10, anything with bash 4.x or 5.x, recent gcc versions, Python 3.9, etc.).  This presents a significant barrier to entry, particularly for people who just want to be able to run MicroPython on their ESP8266, and don't have the background to debug toolchain build errors (such as the friend of mine who asked me to help him with this).

It seems like it would be helpful for the README.md to also refer to places from which a working ESP8266 toolchain might be more easily obtained, such as:

https://github.com/esp-open-sdk/esp-open-sdk - recent, but intended to be aggressively updated and maintained, per https://github.com/pfalcon/esp-open-sdk/issues/387#issue-1068410106 by @igagis and other comments on that issue

https://github.com/someburner/esp-open-sdk - per https://github.com/pfalcon/esp-open-sdk/issues/387#issuecomment-983709721 by @someburner, less aggressively updated and maintained (but maybe better if you need SDK 2.x?)

https://github.com/ChrisMacGregor/esp-open-sdk/tree/builds-on-Ubuntu-21.10 - fork of pfalcon/esp-open-sdk with minimal tweaks to build on Ubuntu 21.10, and no other changes

https://github.com/ChrisMacGregor/esp-open-sdk/releases/tag/builds-on-Ubuntu-21.10 - tarball of ESP8266 toolchain built for Ubuntu 21.10 with the aforementioned minimal tweaks

The README also recommends using a Docker image, but provides no information whatsoever on how or where one might obtain such a Docker image; I believe this may discourage all but the most determined potential MicroPython-on-ESP8266 user, which presumably is not the intended effect.  Perhaps the bold "recommended" might be removed until the README provides adequate guidance for those wishing to follow the recommendation?

gscofano, 2022-01-15:
I do agree that the reference to the project https://github.com/pfalcon/esp-open-sdk seems problematic. Although it appears to be a good toolchain, it's no longer maintened, as you mentioned. I tried to build it myself and had no success. This was my first project on Micropython and it got me discouraged to use it. I'll try to use https://github.com/esp-open-sdk/esp-open-sdk, thank you very much for this tip.

The readme file does mention a link for the docker image, larsks/esp-open-sdk. It didn't work for me, though, when I tried to build this port in my Fedora machine. Inside the ports/ESP8266 directory, I had the following output:

`[user@localhost esp8266]$ docker run --rm -v $HOME:$HOME -u $UID -w $PWD larsks/esp-open-sdk make`
`make: stat: Makefile: Permission denied`
`make: *** No targets specified and no makefile found.  Stop.`

Perhaps the readme file could mention a little bit about this?

joewez, 2022-01-16:
I agree with this assessment.  I have been building the ESP8266 port for years and the one thing I dreaded was losing the precious ESP toolchain.  For such a key feature for using MicroPython to it's fullest, the attention it has gotten has been poor.  I have made pleas on the forums, but have received very little interest on the issue.

I did however come up with a solution that would satisfy me... [https://forum.micropython.org/viewtopic.php?f=16&t=11241#p63829](https://forum.micropython.org/viewtopic.php?f=16&t=11241#p63829)

stinos, 2022-01-16:
> losing the precious ESP toolchain

tools/ci.sh knows how to build all ports (almost) from scratch so it includes getting toolchains which actually work because they're used daily. Uses https://github.com/jepler/esp-open-sdk btw, so none of the ones mentioned earlier.

gscofano, 2022-01-16:
> tools/ci.sh knows how to build all ports (almost) from scratch so it includes getting toolchains which actually work because they're used daily. 

I just tried it. It exited just after I hited enter.

gscofano, 2022-01-16:
`[gscofano@localhost tools]$ chmod +x ci.sh `
`[gscofano@localhost tools]$ ./ci.sh `
`[gscofano@localhost tools]$ `

ChrisMacGregor, 2022-01-17:
https://github.com/jepler/esp-open-sdk is basically identical to the pfalcon one; it adds travis integration, and is missing one commit from pfalcon.

And in any case, ci.sh doesn't use that repo to build the tools - it [just fetches a prebuilt toolchain from the jepler repo](https://github.com/micropython/micropython/blob/da4b38e7562dfa451917f9d7f344a7f26de8c7bd/tools/ci.sh#L135).

That's probably another option to add to the README here, since for many people wanting to use MicroPython, there's no benefit to building the ESP8266 toolchain themselves anyway.  But for those folks who need or want to build the toolchain, neither ci.sh nor https://github.com/jepler/esp-open-sdk seem helpful.

The suggestion from @joewez is interesting.  It will take a bit of work to implement, though, and at least in the meantime, a quick and easy way to reduce the barriers for new MicroPython-on-ESP8266 users would be to tweak the README to direct them to options that work, rather than options that don't.

I can submit a pull request, if someone who can merge it says they want one (and either specifies what the change should be, or is open to any reasonable suggestion).

gscofano, 2022-01-19:
>> https://github.com/jepler/esp-open-sdk is basically identical to the pfalcon one; it adds travis integration, and is missing one commit from pfalcon.

I tried to build it with:

`$ git clone --recursive https://github.com/jepler/esp-open-sdk.git`
`$ cd esp-open-sdk/`
`$ make`

... and I got no success on Fedora 34. It prints:

`checking for bash >= 3.1... no`
`configure: error: could not find bash >= 3.1`
`make[1]: *** [../Makefile:147: _ct-ng] Error 1`

I tried https://github.com/esp-open-sdk/esp-open-sdk/ and its build succeded. I'm still having problems to build the Micropython esp8266 port, though. What I tried:

`$ git clone --recursive https://github.com/micropython/micropython.git`
`$ cd micropython/`
`$ cd mpy-cross/`
`$ make`
`$ cd ../ports/esp8266/`
`$ make`

It prints:

[...]
`CC ../../lib/libm/wf_tgamma.c`
`CC ../../drivers/bus/softspi.c`
`CC ../../drivers/dht/dht.c`
`LINK build-GENERIC/firmware.elf`
`/home/user/Documentos/dev/esp-open-sdk/sdk/xtensa-lx106-elf/xtensa-lx106-elf/sysroot/usr/lib/libmain.a(mem_manager.o): In function malloc':`
`(.text+0x478): multiple definition of malloc'`
`build-GENERIC/posix_helpers.o:posix_helpers.c:(.text.malloc+0x8): first defined here`
`/home/user/Documentos/dev/esp-open-sdk/sdk/xtensa-lx106-elf/xtensa-lx106-elf/sysroot/usr/lib/libmain.a(mem_manager.o): In function free':`
`(.text+0x494): multiple definition of free'`
`build-GENERIC/posix_helpers.o:posix_helpers.c:(.text.free+0x4): first defined here`
`/home/user/Documentos/dev/esp-open-sdk/sdk/xtensa-lx106-elf/xtensa-lx106-elf/sysroot/usr/lib/libmain.a(mem_manager.o): In function calloc':`
`(.text+0x4c4): multiple definition of calloc'`
`build-GENERIC/posix_helpers.o:posix_helpers.c:(.text.calloc+0x4): first defined here`
`/home/user/Documentos/dev/esp-open-sdk/sdk/xtensa-lx106-elf/xtensa-lx106-elf/sysroot/usr/lib/libmain.a(mem_manager.o): In function realloc':`
`(.text+0x4dc): multiple definition of realloc'`
`build-GENERIC/posix_helpers.o:posix_helpers.c:(.text.realloc+0x8): first defined here`
/home/user/Documentos/dev/esp-open-sdk/sdk/xtensa-lx106-elf/xtensa-lx106-elf/sysroot/usr/lib/libmain.a(app_main.o): In function `system_phy_freq_trace_enable':`
`(.irom0.text+0x6e4): undefined reference to user_pre_init'`
/home/user/Documentos/dev/esp-open-sdk/sdk/xtensa-lx106-elf/xtensa-lx106-elf/sysroot/usr/lib/libmain.a(app_main.o): In function `flash_data_check':`
`(.irom0.text+0x780): undefined reference to user_pre_init'`
`make: *** [Makefile:223: build-GENERIC/firmware.elf] Error 1`

Is this comment in the right place, should I raise a new issue, should I make a post on the forum or am I completely off track?

ChrisMacGregor, 2022-01-19:
> > > https://github.com/jepler/esp-open-sdk is basically identical to the pfalcon one; it adds travis integration, and is missing one commit from pfalcon.
> 
> I tried to build it with:
> [...]
> ... and I got no success on Fedora 34. It prints:
> [...]

That makes sense; since it's pretty much the same, I'd expect it to fail the same way.  My point (which I failed to state explicitly) was that I saw no reason to think it would be any improvement over the pfalcon one.

> I tried https://github.com/esp-open-sdk/esp-open-sdk/ and its build succeded. I'm still having problems to build the Micropython esp8266 port, though. What I tried:
>
> `$ git clone --recursive [...]

I can't help you there, since I used my repo with the minimal changes from the pfalcon one; using that, the micropython build succeeded (up to the point where it needed esptool.py, and the ancient version of that from the pfalcon repo failed).

> Is this comment in the right place, should I raise a new issue, should I make a post on the forum or am I completely off track?

To be clear, I speak with zero authority here - I'm just a guy whose friend wanted help getting MicroPython built for ESP8266.  I have no official role with MicroPython, ESP8266, toolchains, or anything else; I don't even use Python (micro or otherwise).

That said, I would guess that for link error you're now seeing, you'll get more help from the forums and/or a new issue (with appropriately descriptive title) specific to this link error.

igagis, 2022-01-19:
I'm the maintainer of https://github.com/esp-open-sdk/esp-open-sdk/ fork.

I have no idea of MicroPython and I don't use Python either in any form.
I don't know about all the errors you get, but for the following one I have some idea:

> /libmain.a(app_main.o): In function system_phy_freq_trace_enable':
> (.irom0.text+0x6e4): undefined reference to user_pre_init' [...]

it is due to that between NONEOS ESP SDK version `2.x.x` and version `3.0.0` there is a backwards incompatible change. Since version `3.0.0` it is required that application define a `user_pre_init()` function and it is required that inside of that function a partition table is initialized. This was not required in SDK versions `2.x.x`. See [here for more info](https://piers.rocks/esp8266/espressif/sdk/2018/11/02/esp8266-sdk-v3.0-user_pre_init.html).

pfalcon's SDK was using ESP sdk version `2.x.x`, while my fork is updated to SDK version `3.0.0`, hence these linker errors.

PsuFan, 2022-01-28:
@ChrisMacGregor well written.

How do we get some actual action on this? I struggle to see why everyone is maintaining MicroPython and completely overlooking the 8266 toolchain. Here is what I have, but I am not the most knowledgeable in this area to be creating this.

```
For Ubuntu Server 18.04

sudo apt-get install make unrar-free autoconf automake libtool gcc g++ gperf \
    flex bison texinfo gawk ncurses-dev libexpat-dev python-dev python python-serial \
    sed git unzip bash help2man wget bzip2 libtool-bin

git clone --recursive https://github.com/pfalcon/esp-open-sdk.git

make -C esp-open-sdk

Install missing tarballs to /home/pi/esp-open-sdk/crosstool-NG/.build/tarballs
From http://mirror.opencompute.org/onie/crosstool-NG/ (currently isl-0.14 & expat-2.1.0)

make -C esp-open-sdk (again)

Add export PATH="/home/pi/esp-open-sdk/xtensa-lx106-elf/bin:$PATH"
to .bashrc file (DO NOT USE ~) or you can use .profile another cmd

git clone -b v1.18 --recursive https://github.com/micropython/micropython.git

make -C micropython/ports/esp8266 submodules

make -C micropython/mpy-cross

make -C micropython/ports/esp8266

ChrisMacGregor, 2022-01-28:
> @ChrisMacGregor well written.

Thanks!

> How do we get some actual action on this?

If that question was directed to me (and I'm hoping it wasn't), I have no idea.

> I struggle to see why everyone is maintaining MicroPython and completely overlooking the 8266 toolchain. Here is what I have, but I am not the most knowledgeable in this area to be creating this.

What you have mostly matches what I did for Ubuntu 21.10.  The differences are:

- Used python3 (including python-is-python3, to make python3 the default).  Note that there's no python-serial package for Ubuntu 21.10, but there is a python3-serial.
- I also installed build-essential, libtool, and libtool-bin; I don't know if those were needed for this.
- I used my repo, which has the fixes for getting isl-0.14 and expat-2.1.0, plus other fixes needed to build on Ubuntu 21.10.  That eliminated the need to run the second esp-open-sdk build.
- I used a tarball of micropython-1.17, but presumably 1.18 also works.
- Adding the toolchain path to PATH in your .profile (better than .bashrc) is helpful IFF you want to have it automatically available in future shell sessions.
- Either way, though, you'll need to export it immediately (which remains in effect until the end of the current shell session) or else it won't affect the subsequent "make" commands.
- If you are using python 3 (and maybe even with python 2), I'm told you may want to remove the esptool.py from the toolchain bin dir, and get a current one using pip.

Here is a version of what you wrote, tweaked to follow what I did:

```
For Ubuntu Server 21.10

sudo apt-get install make unrar-free autoconf automake libtool gcc g++ gperf \
    flex bison texinfo gawk ncurses-dev libexpat-dev python3-dev python3 python3-serial \
    sed git unzip bash help2man wget bzip2 libtool-bin

sudo apt install python-is-python3 build-essential libtool libtool-bin

git clone --recursive https://github.com/ChrisMacGregor/esp-open-sdk.git

make -C esp-open-sdk

# You will need to get a current esptool.py from somewhere, perhaps using pip.
rm esp-open-sdk/xtensa-lx106-elf/bin/esptool.py

export PATH="$(pwd)/esp-open-sdk/xtensa-lx106-elf/bin:$PATH"

# The above adjusts the path for the subsequent commands.
# To apply it to future shells by default, also do the following
# (and be sure to include the backslashes as shown!):
echo export PATH=\"$(pwd)/esp-open-sdk/xtensa-lx106-elf/bin:\$PATH\" >> ~/.profile

git clone -b v1.18 --recursive https://github.com/micropython/micropython.git

make -C micropython/ports/esp8266 submodules

make -C micropython/mpy-cross

make -C micropython/ports/esp8266
```

abishekmuthian, 2022-03-07:
I appreciate the effort of all contributors here, There seems to be a disconnect regarding building micropython for esp8266 here and in the forum.

I'm summarizing what I've learnt so far regarding this subject so those who are new to micropython can benefit.

As detailed by others, Much of the build issues stems from the toolchain i.e. esp-open-sdk:

1. Docker builds stated the in the [forum by user jimmo](https://forum.micropython.org/viewtopic.php?t=7136#p40603) seem to be the simplest way, Provided the docker quirks are understood(Building in home dir on linux should eliminate the privilege issues)

```
$ cd src/github.com/micropython/micropython
$ git submodule update --init
$ cd ports/esp8266
$ docker run --rm -v $HOME:$HOME -u $UID -w $PWD larsks/esp-open-sdk make -j $(nproc)
```
There are at least half-dozen docker builds for esp-open-sdk, Each with their own features but larsks' build seems to be maintained and has a wrapper for easy invocation.

2. [ChrisMacGregor](https://github.com/ChrisMacGregor)'s pre-built [esp-open-sdk](https://github.com/ChrisMacGregor/esp-open-sdk/releases/tag/builds-on-Ubuntu-21.10) fork likely works fine on Ubuntu 21.0 but I was unable to build it on Ubuntu 20.04 LTS due to glibc version mismatch and on Arch Linux due to Guile version mismatch issue(https://github.com/pfalcon/esp-open-sdk/issues/254).

3. [igagis](https://github.com/igagis)'s fork is likely the better long term solution as it uses updated NONEOS ESP SDK, But cannot be used for building micropython due to backwards incompatibility issue (https://github.com/micropython/micropython/issues/8163#issuecomment-1016883709).

To fix this, `user_pre_init()` needs to be added to the code. There was a failed attempt by [pacmac](https://github.com/pacmac) several years back(https://github.com/micropython/micropython/issues/4295#issuecomment-438525506) to do so.

I request [dpgeorge](https://github.com/dpgeorge) to consider adding `user_pre_init()` so that we can use latest esp-open-sdk and avoid aforementioned issues.

P.S. Due to the current semiconductor shortage and shipping issues ESP8266/NodeMCU boards continues to be the most affordable, accessible, featured platform to use micropython and so fixing the build issues can be a huge benefit to many.

bai-yi-bai, 2022-04-04:
@ChrisMacGregor, thank you for consolidating all this information in this issue. I wish I had found it sooner. I would strongly urge you to update the readme.md on your repo to include the instructions from your last post. It will be a good stopgap until a permanent solution is found. I suggest inserting a line (after the rm command), `sudo pip3 install esptool` to address the "You will need to get a current esptool.py from somewhere, perhaps using pip."

I agree with everyone here, it's not ideal that toolchain isn't being maintained. Micropython on Espressif devices is too important to let the SDK whither away. The momentum of Micropython continues to grow. I am here because of several prominent youtubers who made videos over the past two years showing off the potential of micropython. Compiling custom firmware/freezing bytecode is a key tenet of Damian George's ("Writing fast and efficient MicroPython"|https://www.youtube.com/watch?v=hHec4qL00x0&ab_channel=PyConAU). Students and teachers won't have the time, patience, or expertise to get this working. Six weeks ago I encountered my first situation where the standard firmware images on micropython.org became the limiting factor of my code. Since that time, I made several attempts to try to create an environment which can produce firmware files for ESP8266s on Arch Linux, Windows, and Windows Cygwin to no avail.

- @igagis, thank you for creating your fork, however today I encountered an issue the expat library getting updated, breaking the tar download, so I created an (issue|https://github.com/esp-open-sdk/esp-open-sdk/issues/14).
- I also cannot get @someburner's build to work correctly, when trying make, it shows this error. 
```
xtensa-lx106-elf-ld: cannot find -llwip_open
make: *** [Makefile:221: build-GENERIC/firmware.elf] Error 1
```
- @joewez Kudos to you for making your youtube video [Build Your Own Custom MicroPython Firmware for the ESP8266 on Windows ](https://www.youtube.com/watch?v=jG7WBY_vmpE&ab_channel=JoeWezensky).  Perhaps you could edit the description of your youtube video to point to this issue.
- And thank you for suggesting the idea of a web-server which can build firmware with specific modules. I think this would have great value... but I wonder if long term micropython.org could host more images with different libraries. For example, I've been trying to compile my firmware/bytecode for the ESP8266 w/[Peter Hinch's aync-based replacement MQTT library](https://github.com/peterhinch/micropython-mqtt). According to
[MicroPython Asynchronous MQTT|ESP8266 Limitations](https://github.com/peterhinch/micropython-mqtt):
`The module is too large to compile on the ESP8266 and should be precompiled or preferably frozen as bytecode. On the reference board with uasyncio and mqtt_as frozen, the demo script range_ex reports 21.8K of free RAM while running.`  I'm not aware of any other libraries which require freezing as bytecode/firmware.

peterhinch, 2022-04-04:
Perhaps I'm being naive but it seems to me that these problems would go away if we had a set of official build instructions that actually worked, including installing a reasonably recent SDK. This also applies to ESP32 - a platform for which my attempts to build have all failed.

I'm agnostic on whether the official build process should be local, in a container or in the cloud. So long as its readily accessible.

Re `mqtt_as` I considered publishing ESP8266 binaries but I don't think this is the right solution. It should be easy for users to build their own, freezing their own choice of modules. With the limited RAM of ESP8266 the more you can freeze, the better.

joewez, 2022-04-04:
Thanks for the mention @bai-yi-bai !  I did script out a newer revision of the instructions here [https://wezensky.no-ip.org/fwm/doku.php?id=buildingthemicropythonfirmware](url) but have not had time to finish formatting it.  I will do an updated video soon.

My other idea is a simple web page, where the ESP_SDK could be maintained as it changes... so all you would need to do is choose your frozen modules. and build a BIN file from that.  I really thought this would be useful, but have had very little interest.  Using frozen modules is almost essential when working on larger applications, so I am not sure why we don't fix the process in some way.

jimmo, 2022-05-02:
> The README also recommends using a Docker image, but provides no information whatsoever on how or where one might obtain such a Docker image; I believe this may discourage all but the most [...]

@ChrisMacGregor I wrote this section of the README (and some of the forum posts that have been referenced above).

There is nothing for the user to obtain, the command is entirely self-contained and Docker will handle fetching the image for you. I realise of course that this is not at all obvious if you're not already familiar with docker, so I will update the README to make this more clear.

But in the meantime, for example, if you have docker installed and run

`docker run --rm -v $HOME:$HOME -u $UID -w $PWD larsks/esp-open-sdk xtensa-lx106-elf-gcc --version`

it will fetch the `larsks/esp-open-sdk` image automatically (or used a cached version), then run the command (in this case, `xtensa-lx106-elf-gcc --version` inside a temporary container).

So in this case, once cd'ed into the right directory, prefixing this to the `make` command will build firmware using Docker.

`docker run --rm -v $HOME:$HOME -u $UID -w $PWD larsks/esp-open-sdk make`

jimmo, 2022-08-26:
See #[9080](https://github.com/micropython/micropython/pull/9080) to change the instructions to no longer use [pfalcon/esp-open-sdk](https://github.com/pfalcon/esp-open-sdk) and improve the docker instructions slightly.

closed 2022-08-26 by jimmo

dpgeorge, 2022-08-26:
README updated in 09879f99ca3a6c0e48353eb2aaa1c71ed237f126