micropython/micropython #8167: RP2: Hard I2C is intolerant of clock stretching.

peterhinch, 2022-01-12:
This arose from [this forum thread](https://forum.micropython.org/viewtopic.php?f=21&t=11745). This [driver](https://github.com/micropython-IMU/micropython-bno055) for the Bosch BNO055 IMU has been tested on Pyboards, ESP32 and ESP8266 but has proved problematic on the Pico. Investigation showed that the IMU indulges in clock stretching for varying periods. The longest I have observed is 500μs.
### Hard I2C
Hard I2C, invoked as below, fails almost immediately with an EIO error.
```python
i2c = machine.I2C(0, sda=machine.Pin(16), scl=machine.Pin(17))
```
By contrast, hard I2C works on Pyboards and on ESP32.
### Soft I2C
On RP2 soft I2C must be invoked with a higher timeout value:
```python
i2c = machine.SoftI2C(sda=machine.Pin(16), scl=machine.Pin(17), timeout=100_000)
```
Subsidiary question. On ESP32 and ESP8266, soft I2C works with the default timeout. This now seems surprising.

Should the default timeout be increased?

dpgeorge, 2022-01-12:
> Hard I2C, invoked as below, fails almost immediately with an EIO error

Interesting!  It would be good to see if the rp2040 datasheet, or pico-sdk, says anything about clock stretching.

> Subsidiary question. On ESP32 and ESP8266, soft I2C works with the default timeout. This now seems surprising.

Yes that's surprising, because they share the same code with rp2.  I'm guessing it's the implementation of the delay, `mp_hal_delay_us_fast()`, that could be different, probably more accurate on rp2 (ie faster).

peterhinch, 2022-01-13:
Scrub my comments about SoftI2C on ESP32 and ESP8266. When I developed the driver soft I2C was specified with id==-1 and the I2C constructor did not have a timeout arg. The problem has emerged since the timeout was introduced in the SoftI2C constructor. IOW SoftI2C is consistent across all three platforms.

The I2C section of the RP2040 datasheet seems to have no reference to clock stretching. However it is discussed in their PIO implementation of I2C.

dpgeorge, 2022-01-14:
> The problem has emerged since the timeout was introduced in the SoftI2C constructor.

Maybe the default timeout needs to be increased?

> The I2C section of the RP2040 datasheet seems to have no reference to clock stretching

It does mention stretching when the RP2040 is in controller (master) mode.  But I also couldn't see any reference to peripheral clock stretching support.  I'd be surprised if it doesn't support it though.

This comment at CircuitPython seems to indicate that the BMO005 does work with the Pico: https://github.com/adafruit/circuitpython/issues/4082#issuecomment-776961617

peterhinch, 2022-01-14:
In my testing I used the latest release build:
```python
MicroPython v1.17 on 2021-09-02; Raspberry Pi Pico with RP2040
```
I tried reducing the hard I2C clock speed but it still failed with EIO.
```python
i2c = machine.I2C(0, sda=machine.Pin(16), scl=machine.Pin(17), freq=100_000)
```

dhalbert, 2022-01-14:
@peterhinch Does it fail at `machine.I2C()` or later? Could you show some code that fails?

peterhinch, 2022-01-14:
It fails after running for a little while. [This is the script](https://github.com/micropython-IMU/micropython-bno055/blob/master/bno055_test.py) using the I2C invocation in [line 14](https://github.com/micropython-IMU/micropython-bno055/blob/8f59f7c8852ef89d76313bd4d10184ad85434b66/bno055_test.py#L14). With SoftI2C it runs indefintely.

It typically runs for a few seconds, outputting a few complete sets of readings, before the `EIO`. The exact duration varies between runs.

peterhinch, 2022-01-15:
To detail my test conditions, I'm using the [Adafruit 2472 breakout](https://www.adafruit.com/product/2472) powered from the 3.3V output of the Pico. I added 1KΩ pullups on SCL and SDA to 3.3V. Connection is by short jumper leads.

Waveforms on SCL and SDA look fine.

Relying on the 10KΩ pullups on the breakout board led to slow risetimes causing data errors at 400KHz. Hence the added 1KΩ resistors.

dhalbert, 2022-01-15:
I adapted your test script to CircuitPython. I'm running this on an Adafruit QT Py RP2040, with CircuitPython 7.1.0, using the STEMMA connectors, connected to a https://www.adafruit.com/product/4646, which is basically the same as a 2472, but with the added connectors. It runs indefinitely both at 100kHz and 400kHz.
```py
import time
import board
import busio
import adafruit_bno055
from bno055 import *

i2c = busio.I2C(scl=board.SCL1, sda=board.SDA1)  # default clock is 100kHz
#i2c = busio.I2C(scl=board.SCL1, sda=board.SDA1, frequency=400000)

imu = adafruit_bno055.BNO055_I2C(i2c)

calibrated = False
while True:
    time.sleep(1)
    if not calibrated:
        calibrated = imu.calibrated
        print('Calibration required: sys {} gyro {} accel {} mag {}'.format(*imu.calibration_status))
    print('Temperature {}°C'.format(imu.temperature))
    print('Mag       x {:5.0f}    y {:5.0f}     z {:5.0f}'.format(*imu.magnetic))
    print('Gyro      x {:5.0f}    y {:5.0f}     z {:5.0f}'.format(*imu.gyro))
    print('Accel     x {:5.1f}    y {:5.1f}     z {:5.1f}'.format(*imu.acceleration))
    print('Lin acc.  x {:5.1f}    y {:5.1f}     z {:5.1f}'.format(*imu.linear_acceleration))
    print('Gravity   x {:5.1f}    y {:5.1f}     z {:5.1f}'.format(*imu.gravity))
    print('Heading     {:4.0f} roll {:4.0f} pitch {:4.0f}'.format(*imu.euler))
```
[moving around to show different values]
```
Calibration required: sys 0 gyro 3 accel 1 mag 0
Temperature 25°C
Mag       x   -10    y    45     z    40
Gyro      x     0    y     0     z     0
Accel     x  -3.5    y  -4.4     z  -8.0
Lin acc.  x   0.0    y  -0.0     z   0.0
Gravity   x  -3.5    y  -4.3     z  -8.0
Heading      258 roll  -21 pitch  152
Calibration required: sys 0 gyro 3 accel 1 mag 0
Temperature 25°C
Mag       x    -3    y    27     z    22
Gyro      x     9    y     2     z     3
Accel     x  -6.9    y -14.0     z -13.6
Lin acc.  x  -3.1    y  -7.5     z  -7.3
Gravity   x  -3.6    y  -6.4     z  -6.5
Heading       52 roll  -19 pitch  135
Calibration required: sys 0 gyro 3 accel 1 mag 2
Temperature 25°C
Mag       x   -19    y    26     z    21
Gyro      x    -0    y     4     z     1
Accel     x   4.7    y  -1.3     z  -6.2
Lin acc.  x  -2.7    y   2.0     z  -0.6
Gravity   x   7.4    y  -3.6     z  -5.3
Heading      239 roll   49 pitch  146
[...]
```

dhalbert, 2022-01-15:
A simple experiment to try would be to use the bitbangio I2C only when doing zero-length writes, instead of also one- and two-byte writes. @dpgeorge and I discussed that here:
- #6954

Change https://github.com/micropython/micropython/blob/cf258c898eab664615220683d0f9e0507383b925/ports/rp2/machine_i2c.c#L130
to
```c
 if (len == 0) {
 ```

peterhinch, 2022-01-16:
That has fixed it! It ran for 15 minutes until I stopped it. Previously it never managed 10 seconds.

peterhinch, 2022-01-17:
@dpgeorge 
> Maybe the default timeout needs to be increased?

Why not choose the default value so that a t/o will only occur in the event of a hardware failure?

@dhalbert 
While I like the logic of the patch I'm curious as to why it is needed. Why does bitbangio fail for transfers of length 1 or 2?

dpgeorge, 2022-01-17:
> Why not choose the default value so that a t/o will only occur in the event of a hardware failure?

That would mean increasing it to a large value.  But too large means blocking for a while before you realise there's a failure, which could impact other things (eg in an asyncio based application).

That said, bumping it up so the BNO055 works out-of-the-box does seem like a sensible thing to do.

> While I like the logic of the patch I'm curious as to why it is needed. Why does bitbangio fail for transfers of length 1 or 2?

It could be because of the turn around of the GPIO configuration, changing it from alt-function (controlled by the I2C peripheral) to standard GPIO then back again.  This may put some transient signals on the lines that confuse the sensor.

dhalbert, 2022-01-17:
There could be clock-stretching needed for one- or two-byte writes on a particular (badly-behaved?) sensor, and that stretching would be limited to `.us_timeout = 255,`.

In fact we saw a very similar timeout problem with zero-byte writes for a new implementation of our seesaw protocol on an ATTiny. 255 was not quite long enough for it to respond. So we lengthened the zero-bytes timeout (to something too long -- it could be made shorter):
- https://github.com/adafruit/circuitpython/pull/5656

> This may put some transient signals on the lines that confuse the sensor.

I do not remember that I saw transients when I was debugging this  -- I think just saw hangups. I think I would I have gone to some effort to fix the transients if I had seen them. But I don't have any leftover screenshots.

The reason for the bitbang transaction is only because the I2C hardware does not handle zero-byte writes, so it may as well be used only in that case. Handling 1 and 2-byte writes that way was never really necessary, per https://github.com/raspberrypi/pico-sdk/issues/238 and #6954. We found a problem with the bitbang 1- and 2-byte writes when debugging the TCS34725, a sensor that is even more badly behaved than the BNO055.

Here is the corresponding fix in CircuitPython to do only zero-byte writes with bitbang:
- https://github.com/adafruit/circuitpython/pull/4378

dpgeorge, 2022-01-18:
> There could be clock-stretching needed for one- or two-byte writes on a particular (badly-behaved?) sensor, and that stretching would be limited to `.us_timeout = 255,`.

Ah, yes, that's probably it!

@peterhinch can you please try changing that 255 to, say, 100000 and test (line 135 of `ports/rp2/machine_i2c.c`)?

peterhinch, 2022-01-18:
It failed in about 5 seconds with EIO. Code is
```C
        if (len <= 2) {
            // Workaround issue with hardware I2C not accepting short writes.
            mp_machine_soft_i2c_obj_t soft_i2c = {
                .base = { &mp_machine_soft_i2c_type },
                .us_delay = 500000 / self->freq + 1,
                .us_timeout = 100000,
```

dpgeorge, 2022-01-18:
Thanks for testing.

That means that so far we have:
- pure bit-bang I2C works (with timeout=100_000us)
- bit-bang for len=0 (timeout=255us), and HW I2C for len>=1 works
- bit-bang for len<=2 (timeout=255us, 100_000us) and HW I2C for len>=3 does not work

What are the sequence of I2C operations that are going out to the BNO055?

peterhinch, 2022-01-18:
Reviewing the code reveals two places where single byte reads are performed (the rest are  N-byte, where N > 2). I'll reduce the test script to only doing single byte reads. On the assumption that this fails (with the above firmware) I should be able to capture a LA trace of what is going on.

It may be tomorrow before I get there.

peterhinch, 2022-01-18:
I reduced the test script to this:
```python
import machine
import time
from bno055 import *

i2c = machine.I2C(0, sda=machine.Pin(16), scl=machine.Pin(17)) 
trig = machine.Pin(18, machine.Pin.OUT)
trig(0)

imu = BNO055(i2c)
count = 0
while True:
    time.sleep(0.1)
    try:
        print('Temperature {}°C'.format(imu.temperature()))
        count += 1
        print(count)
    except:
        trig(1)
        trig(0)
        print('FAIL')
        break
```
The call to `imu.temperature()` reduces to
```python
self._i2c.readfrom_mem_into(self.address, memaddr, buf)
```
which unpacks to:
```python
self._i2c.readfrom_mem_into(0x28, 0x34, bytearray(1))
```
As expected this fails. A normal transfer looks like this:  
[image]
and a bad one:  
[image]
Zooming in on the second transfer - good:  
[image]
bad:  
[image]
If you need to examine the data more closely I can publish the capture file. With the Saleae software (a [free download](https://www.saleae.com/downloads/)) the capture can be studied in detail.

dpgeorge, 2022-01-19:
Thanks for the traces.  For the "good", is that using `machine.SoftI2C`, or is it a case of `machine.I2C` that succeeds (I'm assuming the latter)?  What value for `.us_timeout` is it using, 255 or did you change the C code to 100000?

The `readfrom_mem_into` is a 1-byte write followed by a 1-byte read.  In the current firmware, when using `machine.I2C`, that will use software I2C for the 1-byte write, then hardware I2C for the 1-byte read.

The "bad" is failing in the 1-byte write.  It looks like the BNO055 stretches the clock for about 350us before it responds with ACK for the 0x28 address, and then the rp2040 sends 0x50 instead of 0x34... and I think the BMO NACKs that value, leading to an `OSError`.

peterhinch, 2022-01-19:
The firmware is [as above](https://github.com/micropython/micropython/issues/8167#issuecomment-1015262666) in all cases. The test script runs for a few seconds before failing. The "good" examples are successful transfers which took place before failure. The "bad" is when failure finally occurs.

After it failed I then rebooted and did another run to see if the "bad" transfer remained similar to the first. It was similar, notably the value 0x50.

Is it relevant that `0x50 == 0x28 << 1`?

peterhinch, 2022-01-19:
I've examined the capture some more. After the address is sent, the clock stretching normally goes to a maximum of about 120μs. Prior to the failure the stretching is unusually long - 370μs on one run and 454μs on the other. This seems to confuse the bit banging driver.

The long delay seems to provoke the RP2 into re-sending the 0x28 address. The protocol analyser incorrectly interprets this as data, hence the observation of a value of 0x50 (because of the way the I2C address is left shifted to provide for the R/W bit). If the bit pattern is examined it looks like a repetition of address 0x28.

I haven't figured out how to interpret the transitions after the ACK, but these trigger the EIO exception as evidenced by the trigger pulse on pin 18.

dpgeorge, 2022-01-20:
What's happening with the 0x50 is that the software I2C driver is doing a 0-byte write in order to release the bus cleanly after the timeout.

This is the sequence of calls:
- `.readmem_from_into(0x28, 0x34, bytearray(1))` at the Python level
- `machine_i2c_readfrom_mem_into`
- `read_mem`
- `mp_machine_i2c_writeto(self, 0x28, "\x34", 1, false)` - the `false` means do not send a stop condition at the end (because there will be a repeat start and read following it)
- this fails with ETIMEDOUT in `mp_hal_i2c_scl_release` while waiting for the ACK from the address (I'm guessing, by looking at the traces above)
- because of the above failure there's a call to do a 0-byte write to cleanly release the bus: `mp_machine_i2c_writeto(self, 0x28, NULL, 0, true)` - the `true` means send a stop

I can understand all the above (and the traces) if `us_timeout=255` because the BNO055 is stretching the ACK beyond that timeout value.  I don't understand why it would still fail in this way if `us_timeout=100000`.

@peterhinch are you certain that you tested it with `us_timeout=100000` as per https://github.com/micropython/micropython/issues/8167#issuecomment-1015262666?  Or even just `us_timeout=1000` should work.

dpgeorge, 2022-01-20:
In 7b0a42374e1ae6a4ab3f756090c6f3b66e82acb5 I made some of the I2C errors more specific.  It may now return ETIMEDOUT instead of EIO.

peterhinch, 2022-01-20:
Apologies - you are right. My recompile-load sequence must have gone adrift somehow.

Recompiling to ensure I have [this code](https://github.com/micropython/micropython/issues/8167#issuecomment-1015262666) the Python scripts run without error. (Both the stripped down script and the original Python test script.)

dpgeorge, 2022-01-20:
Ok, very good, thanks for confirming!

So I'd say there needs to be 3 changes:
1. default timeout for SoftI2C increased, maybe to 1000us?
2. `len <= 2` replaced with `len == 0` in rp2 short-write workaround
3. timeout for rp2 short-write workaround also increased

peterhinch, 2022-01-20:
Yes, I agree with that.

I think the SoftI2C default timeout should be increased substantially.

I take your point about blocking, but advanced users who care about this can override. A default which is too short leads to confusing, rare, errors. Further, while I have run the BNO055 for two hours with a SoftI2C timeout of 1ms, and I haven't measured clock stretching longer than 500μs, there is no way of knowing the worst-case. My tests were done with the IMU stationary. A timeout at 60,000 feet caused by the shock when the rocket first stage detaches would be embarrassing. Lastly, in a tested system, a true timeout (infinite clock stretching) should only occur if the attached hardware actually fails.

The blocking issue can always be highlighted in the docs.

dpgeorge, 2022-01-21:
Fixed by a707fe50b085ec83722106609f6fd219faf9f030

dpgeorge, 2022-01-21:
I increased the default SoftI2C timeout to 50ms in 7d71ae25edc4b405dd03222f044bc7cf3b77f956.  50ms matches the existing timeout for stm32 hardware I2C.

> Lastly, in a tested system, a true timeout (infinite clock stretching) should only occur if the attached hardware actually fails.

Yes, but it than case you'd want the software to timeout as early as possible to catch the hardware failure, and then maybe do something about it :smile: 

I think 50ms is a fair value, it's 100x the 500us seen on the BNO055.  And for a broken I2C bus with SCL permanently low, an I2C.scan() takes about 5 seconds.

Users can do "non blocking" I2C scans by explicitly scanning each device with a 0-byte write, if needed.

closed 2022-01-21 by dpgeorge